robotology/mesh-support #344: Confirming skin version number of our iCub

pattacini, 2016-12-21:
@gsaponaro commented on [Thu Dec 15 2016](https://github.com/robotology/QA/issues/174)

Hi, having read [here](http://wiki.icub.org/wiki/Tactile_sensors_(aka_Skin)) that there exist different versions of the iCub skin, we'd like to know which particular type is mounted on `iCubLisboa01`. For reference, this skin was installed during summer 2014.

/cc @randaz81 @maggia80 @vicentepedro https://github.com/robotology/QA/issues/39

---

@pattacini commented on [Thu Dec 15 2016](https://github.com/robotology/QA/issues/174#issuecomment-267414097)

cc @robotology/icub-support-team

pattacini, 2016-12-21:
Hi @gsaponaro 

Could you take some pictures of the wrist in particular and post them here, so that @julijenv will be able to give you a certain answer?

gsaponaro, 2016-12-21:
Thank you, here it is:

[image]

julijenv, 2016-12-21:
Hi @gsaponaro , could you please take a picture of the palm as well.
The skin part of the upper forearm is V1. and the other part of the forearm never changed sicne the beginning so it is V1, wait for a picture to tell you which version of the palm skin u havfe

gsaponaro, 2016-12-21:
Sure. I removed part of the glue (but not all) to lift part of the palm cover, is the picture below clear enough?

[image]

(Top number says P23.)

julijenv, 2016-12-21:
palm V2

gsaponaro, 2016-12-21:
Great, thanks.

closed 2016-12-21 by gsaponaro