robotology/mesh-support #345: Parts for AEPS plug required for remote controlling EA power supplies

frank-foerster, 2016-12-22:
As agreed yesterday, here's an issue/request to document and track the parts needed to build
the AEPS plug plus Diode needed in order for the on/off switch to work together with the EA power supplies.

I wrote down the following parts needed in order to build the plug. (I couldn't verify the IIT codes as I couldn't find the IIT parts list, but I verified the RS codes.)

1. _part_: Omron G6S-2-Y 12DC   _RS number_: 292-7610    IITCode: 4016
2. _part_: D-sub connector   _RS number_: 528-3660    IITCode: 4017
3. _part_: hood/shell   _RS number_: 484-795
4. _part_: PCB board: _RS number_: none as custom made

Then some resistors are evidently needed, I jotted down 0603 10k resistors. In the diagram in the document '_AEPS_01_DSN.pdf_', I found one such 10 k resistor, 2 6K2 ones, and one 3K3.
When searching RS UK for this, the following seemed the most plausible to me as "non electronician":
[Surface mount resistor 0603 10k](http://uk.rs-online.com/web/p/surface-mount-fixed-resistors/2132418/)

The closest I could find at RS when searching for 0603 3K2 where 
[these 3kOhm resistors](http://uk.rs-online.com/web/p/surface-mount-fixed-resistors/6790295/)

For the 6k2 ones RS gives me [this](http://uk.rs-online.com/web/p/surface-mount-fixed-resistors/8048906/?sra=pmpn)

Could you please verify whether these are what we'll need?

For the **diode**, you wanted to look up, what precisely the part number was. 

In terms of the **PCB board** a colleague of mine was rather pessimistic that our university has facilities to manufacture one, but I'll have to verify this with some guys in electrical engineering department.
If we don't have the facilities, it would probably easiest if you guys could manufacture one for us.

One related question: For the time being, what is the procedure to power the robot, given that we don't have the plug + diode yet? 

Instead of using the "icub switches", do I just leave those switched to 'on' and to the real switching for both CPU and motors via the "output on/off" switch of the two EA power supplies? 
And do I leave the two cables that eventually will connect to the AEPS plug just "dangling" / non-connected?

Many thanks for your help.

frank-foerster, 2016-12-22:
A further question not directly related to the AEPS plug: 
I just tried to operate the robot by running both power supplies in local mode. I can startup the CPU with the 12V power supply, if the iCub CPU switch is turned on to start with and switching it on by turning the `output` of the 12V supply to `on`. 
However, as soon as I switch on the 48 V power supply, both of them switch themself off and I get an `OV` reading in the 12V supply - so some overvoltage appears to occur. 
If I leave the iCub motor switch switched on the small power supply only start up for one second and then immediately switches itself off. 

I used the same presets as the ones I saw in some other issue I saw reported in here (and which I can't find at the moment), namely

**12V supply**:
voltage: 13.5V
undervoltage: 11.5V
overvoltage: 14.5V
current: 20A
power: 500W

**48V supply**
voltage: 38.8V
undervoltage: 24V
overvoltage: 48V
current: 60A
power: 1.5kW

Both negative poles are connected, but the cables that are ought to go in the AEPS plug are left disconnected / dangling.

Any idea, what could cause the over voltage?

julijenv, 2016-12-22:
Hi @frank-foerster ,
for the 3 resistor you found out, 2 are ok the first (10k and 3.3k) price and qty but the 6.2k i would recommend you to find another one because it is a reel of 5000 pieces and you just need 1, i guess you can find another rs code that can suit 0603 and 6.2k for less than 1 pound but afterwards it is totally up to you.
Hope you found the main component - Omron G6S-2-Y 12DC.
I think you forget to find the component:
PCB connector, DB, 15 contacts, 2 rows, male, p.2.77mm, 90°, w to b, THD, long termination
RS	528-3660				

Anyhow, I was wondering if it couldnt be faster to you to ask for AEPS board module to us and get an invoice and pay for it, it will ease the process and maybe speed it up?
let me know,

frank-foerster, 2016-12-22:
Hi @julijenv 
yes, I think that might be the best option.

frank-foerster, 2017-01-03:
As agreed on Skype, we'll wait for your quote for the AEPS board/plug and diode and will use the old power supplies for the time being. 
Note to others who run may try the same: We couldn't get a "static" power supply setup without the AEPS board to work. Upon trying to switch on the power supplies for the motors first, the 12V power supply still received/generated and overvoltage (OV symbol in display) as described above.
Closing this issue for now.

closed 2017-01-03 by frank-foerster