robotology/mesh-support #346: fingertip came loose

frank-foerster, 2017-01-04:
### Description of the failure
Our iCub's fingertip came loose today as depicted on the image.

### Detailed conditions and logs of the failure

My simple question: can we just fix it by glueing it back to area that I marked with the red oval
using superglue, or do I need to use some special glue for this.
Or is there another solution not requiring any glue.

[image]

Many thanks.

julijenv, 2017-01-04:
Hi @frank-foerster ,
we usually glue this plastic bit of the fingertipskin on the pcb thanks to a repair extreme glue from pattex:
here is one example the price and where to get it in UK:
https://www.amazon.co.uk/Pattex-Repair-Extreme-Power-Glue/dp/B000WL3KR2
LEt us know if you cant find it

frank-foerster, 2017-01-04:
Thanks, that was what I wanted to know.

closed 2017-01-04 by frank-foerster