robotology/mesh-support #350: Strange limit cycle occuring on joints l_hip_roll and r_hip_roll when they are put in home using the home button in yarpmotorgui

traversaro, 2017-01-09:
### Description of the failure
On startup, the joint `r_hip_roll` goes to its home position, but it slightly oscillates around its home position. If you move it to any position (even the same position of the home) the oscillation stops. If you then "home" with the "home" button in the yarpmotorgui either the `r_hip_roll` and `l_hip_roll`, then the oscillation happen again.

traversaro, 2017-01-09:
Apparently this limit cycle is occurring when `l_hip_roll` and `r_hip_roll` position setpoint is 10.198 , but not always (but this is indipendent from how the set point is set). Decreasing the integral gain of the position loop of a factor 14 remove the issue, but it is not clear if this make sense, as it could decrease the performance of the position contorl in other situations.

ale-git, 2017-01-09:
Hi @traversaro, we experienced the same problem on iCubNancy01: the effect of the gravity leads to instability in some configurations with standard PID values. We solved the problem by adding a small derivative component ( Kd = Kp/5) to the leg roll position control.

traversaro, 2017-01-09:
Great. 👍

traversaro, 2017-01-10:
Setting Kd = Kp/5 does not solve the problem, apparently it just increases the frequency of the limit cycle.

iron76, 2017-01-10:
Usually these limit cycles are caused by the integral term. Can you check if Ki is set to zero?

traversaro, 2017-01-10:
> Can you check if Ki is set to zero?

From https://github.com/robotology/icub-support/issues/350#issuecomment-271234599 : 
> Decreasing the integral gain of the position loop of a factor 14 remove the issue, but it is not clear if this make sense, as it could decrease the performance of the position contorl in other situations.

traversaro, 2017-01-10:
In a nutshell: the integral term is not zero in all the joints of the leg, so this may suggest that setting the integral part to zero would be a workaround shadowing a deeper problem in the system.

Furthermore it would be better to evalute the effect in term of static error before setting the integral term to zero. In particular it would make sense to run the appropriate `icub-tests`.

iron76, 2017-01-10:
In the past, tuning the PID controller was an alchemy that required experience and time. Specifically, it was @randaz81, @maggia80 and me responsible for doing this tuning. I suspect we still rely on hand-tuning. Again in the past, I tested more advanced tuning techniques (e.g. adaptive control) and in my experience they work much better but coding them requires a dedicated person.

traversaro, 2017-01-10:
cc @ale-git @randaz81 @davidetome

traversaro, 2017-01-18:
cc @fjandrad @S-Dafarra @gabrielenava

julijenv, 2017-03-28:
any update on this matter, @traversaro , @fjandrad , @randaz81 ???? or can we close it?

closed 2017-08-08 by julijenv