robotology/mesh-support #353: Sometimes during startup, the motor LED in the backpack becomes red and blinks

traversaro, 2017-01-10:
### Description of the failure
Sometimes during startup, the motor led in the backpack becomes red and blinks.

### Detailed conditions and logs of the failure
It happened once today and once yesterday. Unfortunately I do not have logs of what happened, but apparently when the motor LED in the backpack becomes red and blinks the motor power is not erogated.

traversaro, 2017-01-10:
To stop the LED motor blinking and actually turn it to green, you have to keep it pressed for 1 or 2 seconds (not sure).

francesco-romano, 2017-01-10:
On the iCubGenova02 this happened with spikes in the power supply. 
cc @maggia80

traversaro, 2017-01-11:
It happened again once today.

nunoguedelha, 2017-01-12:
It happened again once this morning

MrAndrea, 2017-01-12:
Hi @traversaro and @francesco-romano 
when the motor led blinks red, it means that an overcurrent occurred. (btw: the CPU led has same behaviour). If this seldom happens, just press for few seconds the button and forget it. But if it often happens there is a problem to understand, so in this case call us:  it could be a problem with the power supply or, worse, problems on boards or cables.

nunoguedelha, 2017-01-12:
The issue is occurring very often now, at least 1 over 2 yarprobotinterface run.

nunoguedelha, 2017-01-12:
@MrAndrea, the issue does not occur if we don't activate the motor control board for head joints 2 to 5, i.e. the neck yaw and the eyes.
So for now, as a work around, I deactivated both head motor control boards, so all head joints (through the local configuration files), since anyway, the neck chain cannot be fully controlled.
@traversaro, I will add this workaround to the initial description of the problem.

nunoguedelha, 2017-01-13:
Fixed by @MrAndrea by increasing the over-current limit to 6 amps on the robot main power board. The robot interface now starts with all joints activated (after removing the workaround) and without the over-current failure.
I suggest we keep using the robot one or two days before closing the issue, for checking this doesn't occur anymore.

closed 2017-01-18 by julijenv

traversaro, 2017-02-14:
Something similar happened today on iCubGenova02, three weeks after the installation of the new head. @gabrielenava is monitoring the situation and will open a new issue if necessary.

MrAndrea, 2017-02-27:
@arrenglover and @francesco-romano, I have updated the firmware on the backpack in order to increase the over-current limit in the same way of the green iCub. Try and let me know if the issue is fixed.

arrenglover, 2017-02-27:
During experiments I noticed the purple robot would no longer fault during startup of yarprobotinterface. However when all joints were under control, and I also run gaze tracking software to move the head, the firmware limit in the backpack would again be triggered. To run gaze tracking (e.g. something like the red-ball demo) I needed to idle other joints such that the total current was less.