robotology/mesh-support #354: icub-head is not a known hosts for the terminal

traversaro, 2017-01-10:
### Description of the failure
Typing:
~~~
ssh icub-head
~~~
in the robot terminal results in a password being asked.
~~~
iCub OS version 7.0-16.06.28
icub@icub-head's password: 
~~~

## Details 
I think we use a standard setup for setting up these things, but I do not know what it is. It is using `.ssh/authorized_keys` ? There is some documentation on this?

 cc @mbrunettini

mbrunettini, 2017-01-10:
No, it is not a matter of known hosts.
If you add a new pc to a robot environment and you want to SSH to the icub-head without password, you must copy the SSH public key to the destination machine. The procedure is standard ( see http://askubuntu.com/questions/46930/how-can-i-set-up-password-less-ssh-login)
The instructions are also on iCub wiki here : http://wiki.icub.org/wiki/The_Linux_on_the_pc104#SSH_passwordless_login

traversaro, 2017-01-10:
Great, thanks. Opening the issue until I actually implement this changes in the robot.

traversaro, 2017-01-11:
Done.

closed 2017-01-11 by traversaro