robotology/mesh-support #355: Some skin patches are not working

claudiofantacci, 2017-01-12:
### Description of the failure
Running Skin_Gui_All on Blacky, all skin patches are correctly shown, but some of them are not working.

### Detailed conditions and logs of the failure
- Right forearm skin is not working.
- Right hand skin (including fingertips) is not working.
- Left forearm skin is not working.
- Torso lower-right part of the skin is not working.

---
There is also (an old) issue #28 about Blacky skin, but I can't tell whether they are related or not.

cc @pattacini

towardthesea, 2017-01-12:
After restarting all the machines as previous suggestion of @julijenv, all skin patches of `right forearm` and `right hand` work normally, while the ones in `left forearm` doesn't response. 
Cheers,
Phuong

claudiofantacci, 2017-01-12:
What about the torso skin?

towardthesea, 2017-01-13:
As @julijenv said, the problem maybe due to the power supply cable, i.e too small to provide enough power to the `forearm`, the ending part of the load chain. 
Currently, he fixed it by bypassing the power supply to skin part of `right arm`, so there is no skin activation for that `right arm`.

Keep this issue on until it is totally fixed!

towardthesea, 2017-01-19:
@julijenv solved it! Close now!

closed 2017-01-19 by towardthesea