robotology/mesh-support #356: F/T sensors saturation during the Yoga++ demo

gabrielenava, 2017-01-12:
During the Yoga++ demo, F/T sensors often saturate as reported by yarplogger (see example below) :

` <#STRING_START#> from BOARD 10.0.1.6 (EB6), src CAN2, adr 13, time 1542s 728m 963u: (code 0x01000002, par16 0x0003 par64 0x00000000000003e4) -> HW: strain values saturation. In p16 there is the channel involved, in p64 there are the saturation counts for the last sec of both types (lower: LS32bits, upper MS32bits + .
<#STRING_END#> `

In order to debug this problem, we run the Yoga++ demo three times, and we saved the data from yarplogger. The purpose is to identify which sensor saturates the most and change it.

In this issue, we will keep trace of the logger info collected during the test:

[LOG_REPORTS.zip](https://github.com/robotology/icub-support/files/702152/LOG_REPORTS.zip)

gabrielenava, 2017-01-12:
cc @traversaro @DanielePucci

iron76, 2017-01-13:
For the records, we are going to replace the sensor with a new prototype of the serie FT45_M1_E1-5 (mechanics v1, electronics v1.5, gain 5X). The prototype is SN163-FT45_M1_E1-5.

gabrielenava, 2017-01-17:
Yesterday, we substituted sensor EB6 (left leg) that was the one which saturates the most. After substituting it, we performed the yoga++ demo four times, and EB6 saturates only once. Results of the test are attached here:

[NEW_LOG_REPORTS.zip](https://github.com/robotology/icub-support/files/710454/NEW_LOG_REPORTS.zip)

gabrielenava, 2017-01-17:
cc @fjandrad

traversaro, 2017-01-27:
@gabrielenava @fjandrad You reported all this information in the relevant redmine issue? If this is the case, we can close this issue.

gabrielenava, 2017-01-27:
I didn't report the info, but looking in redmine I think @fjandrad did. Am I right?

traversaro, 2017-02-02:
@fjandrad ?

fjandrad, 2017-02-06:
yes we opened the redmine issue where the lgos where also posted.
We can close it

closed 2017-02-06 by fjandrad