robotology/mesh-support #357: Right hand cable broken

amaroyo, 2017-01-13:
Dear @julijenv, 

We are missing you in Erzelli! :D 

The right hand thumb cable is broken - as the yarpmotorgui is not working I cannot tell you which joint, but I send you the photo attached. 

It is strange that it's broken because since before Christmas nobody has touch the robot and it was working. Maybe some of the cables are touching..? (as you may see in the photo, the cable is very long, it broke at the beginning!)

We are planning to run some experiments in the coming days.. is it possible for you to come?

Thanks a lot!
Alex.

[image]

amaroyo, 2017-01-20:
@julijenv 

Left hand thumb cable broken. You can find which one is it! The thumb is totally blocked, even in idle position. I havent used the left hand at all, only at calibration.

[image]

julijenv, 2017-01-20:
Hi @amaroyo ,
it is the proximal joint (j9).

julijenv, 2017-01-30:
closing issue,
It has been rewired both thumb proximal joint (j9) and mounted the new dorso cover.
material used for the repair:
IITCODE Qta Description
1815 x 5m : carlstahl 0.63mm
2391 x 5m : carlstahl 0.45-0.61mm

closed 2017-01-30 by julijenv