robotology/mesh-support #358: IMU failure: unknown write response: 0x00 0x00

nunoguedelha, 2017-01-13:
### Description of the failure
Sometimes, at startup, we get a failure message from the IMU.

### Detailed conditions and logs of the failure
Startup with configuration `icub_all_inertials.xml`, with both legs accelerometers activated. But this also occurred in the past before any accelerometer was activated. Now that we don't get the over-current issue anymore, we will check how frequently the IMU failure occurs.
```
121388	28015,157395	ERROR	Bosch BNO055 IMU - Received unknown write response: 0x00 0x00. This should not happen!!
121389	28015,461074	ERROR	Bosch BNO055 IMU - Received error response (0xEE) - code 0x01
121392	28015,481362	ERROR	Bosch BNO055 IMU - Received unknown read response: 0x07 0xbb. This should not happen!!
121394	28015,683764	ERROR	FAILED writing  Set config NDOF_MODE
121396	28015,683777	ERROR	Bosch BNO055 IMU - Received error response (0xEE) - code 0x03
121398	28015,693887	ERROR	FAILED writing  Set config NDOF_MODE
121400	28015,693900	ERROR	Bosch BNO055 IMU - Received error response (0xEE) - code 0x03
121402	28015,704010	ERROR	FAILED writing  Set config NDOF_MODE
121403	28015,704016	ERROR	Bosch BNO055 IMU - Received error response (0xEE) - code 0x03
121405	28015,714100	ERROR	FAILED writing  Set config NDOF_MODE
121408	28015,714147	ERROR	BoschIMU: set config NDOF_MODE failed
```

nunoguedelha, 2017-03-15:
This IMU failure is happening systematically when the robot interface starts.

nunoguedelha, 2017-03-15:
@valegagge could this be related to the new config format at the firmware level you've worked on some time ago?

barbalberto, 2017-03-15:
Tha bosh imu is a usb device, it is not related to our custom firmware (no EMS, no can).
Those errors simply means that the USB communication somehow is not working, maybe some problem on the cable or electrical disturbance or similar.

nunoguedelha, 2017-03-15:
Thanks @barbalberto. For the record, I'm having reasonable measurements, although more noisy than expected.
So I should ask @julijenv directly to take a look at this?

barbalberto, 2017-03-16:
Can you try to run the IMU without the motors, to see it there is some 
kind of interference or the problem persist?

nunoguedelha, 2017-03-21:
I made some tests:
- removing all the motors configurations --> no IMU error occurred
- added back head motors --> still no error
- added back the other motors part by part (just in case some unexpected interference could occur, since everything goes through the head CPU) --> no error occurred

I don't understand what make the error disappear. I have to stress it a little bit more.

barbalberto, 2017-03-22:
hihihi, it is always like this. The falous EisenBUG.
Please update this issue when you find new information. I've updated the driver with more meaningful (I hope) error messages.

nunoguedelha, 2017-03-22:
Thanks 👍 . I'l let you know.

julijenv, 2017-03-28:
hi @nunoguedelha , any update on this matter?

nunoguedelha, 2017-10-26:
The error occured again today, but with some differences:
```
962	3,734475	ERROR	Bosch BNO055 IMU - Received unknown response message. 
            If this error happens more than once in a row, please check serial communication is working fine and it isn't affected by electrical disturbance.
966	4,027630	ERROR	Bosch BNO055 IMU - Inertial sensor didn't understand the command. 
            If this error happens more than once in a row, please check serial communication is working fine and it isn't affected by electrical disturbance.
969	4,057912	ERROR	Bosch BNO055 IMU - Received unknown response message. 
            If this error happens more than once in a row, please check serial communication is working fine and it isn't affected by electrical disturbance.
971	4,270321	ERROR	FAILED writing  Set config NDOF_MODE
973	4,280419	ERROR	Bosch BNO055 IMU - Inertial sensor didn't understand the command. 
            If this error happens more than once in a row, please check serial communication is working fine and it isn't affected by electrical disturbance.
977	4,320950	ERROR	FAILED writing  Set config NDOF_MODE
979	4,331078	ERROR	Bosch BNO055 IMU - Inertial sensor didn't understand the command. 
            If this error happens more than once in a row, please check serial communication is working fine and it isn't affected by electrical disturbance.
982	4,361462	ERROR	FAILED writing  Set config NDOF_MODE
983	4,381681	ERROR	Bosch BNO055 IMU - Inertial sensor didn't understand the command. 
            If this error happens more than once in a row, please check serial communication is working fine and it isn't affected by electrical disturbance.
987	4,412097	ERROR	FAILED writing  Set config NDOF_MODE
988	4,412105	ERROR	BoschIMU: set config NDOF_MODE failed
```

Occurred after a full power on of motors and head CPU.

nunoguedelha, 2017-10-27:
Apparently this issue occurs the first time we launch the robot interface after a head CPU power OFF/ON cycle. After that, if the robot interface is rebooted while keeping the head CPU always ON, we never observe the issue.

closed 2018-01-16 by julijenv