robotology/mesh-support #362: VGA cable needed

Tobias-Fischer, 2017-01-17:
Hi @julijenv,
It would be great if you could send us a VGA cable for debugging the PC104, as it seems we need one to debug the new icub image (see https://github.com/robotology/QA/issues/177).
Thanks a lot,
Tobi

julijenv, 2017-01-18:
Hi @Tobias-Fischer ,
unfortunately we dont have any european project going on together, so I guess you will have to create your own.
you could find the way to create the VGA cable to PC104 by looking to:
#183 and #299 
if too lazy to look for it I recopy the link ;)
https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/doc/howTo/Connect_VGA_to_PC104.pdf
Let me know,

pattacini, 2017-01-18:
@julijenv WYSIWYD is considered already over?

julijenv, 2017-01-18:
@Tobias-Fischer @pattacini ,
WYSIWYD is over, but in our great generosity, we will send you one VGA cable to PC, that was around, consider this as a favor and a lucky day for you that I have it, or else you would have had to make it :D:D:D:D.
Please give us a shipping address, the personn to receive it and his phone number.

Tobias-Fischer, 2017-01-18:
Hi @julijenv, thanks a lot, that's highly appreciated :)! Glad you have a spare cable around.
This is the shipping address:

Tobias Fischer
Level 1 Stores
Electrical Engineering Department
Imperial College
London, SW7 2BT
United Kingdom

Delivery Contact: Hisham Maarki - +44 020 7594 6162

Thanks a lot :)
Tobi

Tobias-Fischer, 2017-01-19:
I have no idea how you've done it, but the cable arrived already. You're a star! Thanks a lot :)

closed 2017-01-19 by Tobias-Fischer

pattacini, 2017-01-19:
Julien is Julien 😉

julijenv, 2017-01-19:
hope it works, :D:D:D:D