robotology/mesh-support #363: Motor control boards are not reachable at startup

nunoguedelha, 2017-01-18:
### Description of the failure
Startup fails because motor control boards are not reachable.

### Detailed conditions and logs of the failure
We first get detection error for boards EB5, leftArm-jxxx, rightArm-jxxx of IPs 10.0.1.5, 10.0.1.24, 10.0.1.25, then for the rest of the daisy chained boards:
```
2237	220,970981	ERROR	EthResource::verifyBoardPresence() DID NOT have replies from BOARD EB5 with IP 10.0.1.5  even after  20  attempts and 10.0112 seconds: CANNOT PROCEED ANY FURTHER
2239	220,971032	ERROR	EthResource::verifyEPprotocol() cannot verify BOARD EB5 with IP 10.0.1.5 : cannot proceed any further
2324	222,757758	ERROR	HasSpeedEncoder  parameter not found
2390	232,777780	ERROR	EthResource::verifyBoardPresence() DID NOT have replies from BOARD leftArm-j4_7 with IP 10.0.1.24  even after  20  attempts and 10.0131 seconds: CANNOT PROCEED ANY FURTHER
2392	232,777828	ERROR	EthResource::verifyEPprotocol() cannot verify BOARD leftArm-j4_7 with IP 10.0.1.24 : cannot proceed any further
2405	234,300333	ERROR	HasSpeedEncoder  parameter not found
2472	244,316156	ERROR	EthResource::verifyBoardPresence() DID NOT have replies from BOARD leftArm-j8_11 with IP 10.0.1.25  even after  20  attempts and 10.0127 seconds: CANNOT PROCEED ANY FURTHER
2473	244,316174	ERROR	EthResource::verifyEPprotocol() cannot verify BOARD leftArm-j8_11 with IP 10.0.1.25 : cannot proceed any further
2487	245,832981	ERROR	HasSpeedEncoder  parameter not found
2553	255,851669	ERROR	EthResource::verifyBoardPresence() DID NOT have replies from BOARD leftArm-j12_15 with IP 10.0.1.26  even after  20  attempts and 10.0131 seconds: CANNOT PROCEED ANY FURTHER
2555	255,851681	ERROR	EthResource::verifyEPprotocol() cannot verify BOARD leftArm-j12_15 with IP 10.0.1.26 : cannot proceed any further
```

nunoguedelha, 2017-01-18:
cc @valegagge

julijenv, 2017-01-18:
the issue was due to a bad wiring. I t has been solve by @spada1978 .

closed 2017-01-18 by julijenv