robotology/mesh-support #366: Map of the fingertip taxels - iCubLisboa01

vicentepedro, 2017-01-18:
Hi!

We found a difference between the map of the fingertip taxels found in a recent paper:

> A New Design of a Fingertip for the iCub Hand
> Nawid Jamali, Marco Maggiali, Francesco Giovannini, Giorgio Metta and Lorenzo Natale

[image]

compared to ours:
[image]

acquired from iCubSkinGui and reading from the `/icub/skin/right_hand_comp` port.

Can it be a configuration problem on the software side? Which files should we change?

We have the skin palm V2 (as mentioned here #344)

Thanks in advance,

\cc @gsaponaro @lorejam @NunoDuarte @massimoregoli

NunoDuarte, 2017-01-29:
Hi @nawidj ,
@pattacini told us that you may help with this issue.

Thanks in advance
@NunoDuarte, @vicentepedro

vicentepedro, 2017-01-31:
@lornat75 this was the issue with the finger taxels that I was talking to you at yesterday's dinner.

Maybe we can check this during the WinterSchool, if we have time.

Regards,
Pedro

vicentepedro, 2017-02-21:
Just a brief update on this issue.

We remapped (in a hard-coded approach) the tactile sensors in the fingertips according to the article cited above. And the module we were using to get the external forces is working fine now.
However, we need a more permanent solution to this issue. In order to use every module in the iCub repo that exploits tactile sensing.

Regards,
Pedro

julijenv, 2017-03-02:
Hi @vicentepedro ,
For the rime bien please try to flash the new FW of the MTB Id 14 on can line 7 and 8 to see if it solves your issue.
@maggia80 , any idea on how to change those features on the GUI?!?!?

pattacini, 2017-03-02:
I think the problem @vicentepedro is referring to is not related to the `SkinGUI`, but more on the data streamed out by the port, because he reported on a module of theirs that started working over after the hack they applied...

julijenv, 2017-03-02:
Hi @vicentepedro ,
after talking with M.Maggiali, we found out that your fingertip (right one, we supposed) are the last version of it. Actually there is a difference in between the right and left pcb taxels position, the one quoted in the article is a left fingertip, but your test have been done on a right one, oculd you confirm that?

vicentepedro, 2017-03-08:
Hi, @julijenv. 

Thanks for your answer.

We double-check the map of the left and right pcb taxels and you are right. They are different!

So, I think you can close the issue since "remapping" from left taxels to right taxels was the right solution to this problem. 

Just a couple of last questions:

1) Why are they different? There is a reason for it? 
2) These maps of the fingertips taxels are common to all the iCubs? Or it depends on the hand version? (e.g. if we share our code with someone else, it will work there?)

maggia80, 2017-06-13:
Hi @vicentepedro ,

1) Left and Right are different because of design constraints.
2) There are two different version of the fingertips. The old iCubs have still the V1, however most of them have already the V2.

vicentepedro, 2017-06-20:
Hi @maggia80!

Thanks for your answer. 

Just a last question (I promise 😄 ), where can I find these maps of the fingertip taxels for V1, V2, left and right fingers?

I want to be able to adapt/help if someone with a different robot wants to use our code developed for iCub-Lisboa01.

Thanks

vicentepedro, 2018-01-16:
Hi @julijenv 

Sorry for re-opening this issue.

But my last question:

_Just a last question (I promise :smile: ), where can I find these maps of the fingertip taxels for V1, V2, left and right fingers?_

was not answered,  yet. Maybe @maggia80 knows the answer...

If you want, I can open a new issue here at icub-support (or on a different repository) regarding this question since is not just a problem/question about iCubLisboa01, but about the tactile sensors in general.

Best,
Pedro

maggia80, 2018-04-09:
Hi all,

sorry for the very late reply. Here there is the picture of the version 2 of the pads in the fingertips:

[image]

maggia80, 2018-04-09:
https://github.com/robotology/icub-main/blob/master/app/skinGui/conf/positions/left_hand_V2_1.txt
and 
https://github.com/robotology/icub-main/blob/master/app/skinGui/conf/positions/right_hand_V2_1.txt

for the viewer

vicentepedro, 2018-04-10:
Thanks @maggia80 for the reply!

I will close the issue.

closed 2018-04-10 by vicentepedro