robotology/mesh-support #369: Right arm board error

fjandrad, 2017-01-25:
:

### Description of the failure
356500	164454,014305	ERROR	 from BOARD 10.0.1.28 (rightArm-j8_11), src LOCAL, adr 0, time 543s 557m 85u: (code 0x04000002, par16 0x0000 par64 0x0000000000000000) -> DEBUG: tag02 + mais timeout

356603	164455,864600	ERROR	unable to complete calibration: joint 2 in 'hw_fault status' inside doneRaw() function

### Detailed conditions and logs of the failure

Upon starting yarprobotinterface I get the previously mentioned error messages.

marcoaccame, 2017-01-30:
Hi @fjandrad, can you please post the complete log?

julijenv, 2017-02-02:
Hi @fjandrad ,
any news?

fjandrad, 2017-02-06:
Sorry for the late response. It has not happened again you can close it.

nunoguedelha, 2017-03-12:
Closing this issue since it hasn't been reproduced for more than a month.

closed 2017-03-12 by nunoguedelha