robotology/mesh-support #370: Torso oscillations

giuliavezzani, 2017-01-25:
### Description of the failure
Torso of `iCubDarmstadt01` (**now in Genova**) oscillates in the `(0,0,0)` position

### Detailed conditions and logs of the failure
If we set different torso joints value, the oscillations descrease.
However if we move the robot torso, the oscillations start again.

randaz81, 2017-01-25:
ki pos too high?

giuliavezzani, 2017-01-25:
thanks @randaz81 for the help!
It seems that setting **`ki=0`**  for **`torso join no.2`**  solves the problem!
We will test tomorrow if this change is enough even while we normally use the robot.

giuliavezzani, 2017-01-30:
Today torso **seems to be stable** even **without decreasing** the `ki` value for torso joint no.2

giuliavezzani, 2017-01-30:
We tested how torso behaves while moving during `depth2kin` and with manual movements from `yarpmotorgui` and **`ki=0.0`** for **torso joint no. 2**. It works well, so I modified the configuration file for the torso in the .local.
Thanks @randaz81!

closed 2017-01-30 by giuliavezzani