robotology/mesh-support #387: Almost all the joints of the arms return 0.0 as the joint position

traversaro, 2017-02-08:
All the joints of the arms after `*_prosup` (included) return 0.0 as their encoder position, even if the position is different. 
Note that the low level position control seems to be working fine, so the encoder actually read the correct value, but that value is not correctly reported on the YARP network. 

cc @pattacini

AlexisBillier, 2017-02-08:
(-35.656215 30.152051 0.027466 50.086792 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) [ok] (0.0 0.065918 -0.032959 0.219727 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) [ok] (0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) [ok] (-3520.011646 8726.121158 930.297071 514.721716 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) [ok] (0.0 10.986355 5.493177 21.97271 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) [ok] (0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) [ok] (-2.269714 0.763788 -0.037122 0.843933 -0.023286 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) [ok] (-422.0 -393.0 110.0 460.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) [ok] (7565168 7565168 7565168 7565168 7565168 7565168 7565168 7565168 7565168 7565168 7565168 7565168 7565168 7565168 7565168 7565168) [ok] (1718187123 1718187123 1718187123 1718187123 1718187123 1718187123 1718187123 1718187123 1718187123 1718187123 1718187123 1718187123 1718187123 1718187123 1718187123 1718187123) [ok]

traversaro, 2017-02-08:
Thanks @AlexBiller for the excerpt from one message on the `/icub/left_arm/stateExt:o` port. 
Apparently the problem is not related only to encoders, but to all quantities, except for the torque that is provided by `wholeBodyDynamics`.

traversaro, 2017-02-08:
The branch used in the robot are:
* yarp : `devel` 
* icub-main : [`python_camera_compatib`](https://github.com/robotology/icub-main/pull/415)

On iCubGenova01 and iCubGenova04 with the recentish `devel` the issue is not present.

julijenv, 2017-03-28:
hi @traversaro , @pattacini ,
any update on this matter?