robotology/mesh-support #390: Can't connect to PC104

emilio-cartoni, 2017-02-09:
### Description of the failure
We can no longer connect to PC104.
It happened suddenly after a reboot of the whole iCub.
When on the laptop we click on the shortcut "PC104" to connect to the board, the terminal just opens and then after a few seconds it closes (as it does if you try to use it when the PC104 has not finished booting).

### Detailed conditions and logs of the failure
When we power on the head, we can see lights in the head of the iCub and also we can hear the fan.
The ethernet cable from the iCub seems to be powered as we can see a green light on the router when we connect to it.
Trying to ssh or ping the iCub (10.0.0.2) fails.

Given the above data, it seems that the PC104 is alive but it doesn't boot properly.

julijenv, 2017-02-09:
Hi icubRome02 users,
could you explain me more in detail what happen exactely when you say "reboot of the whole iCub"?
Did you move the robot from its earlier position or the power supplies/server??? or did update the laptop?
have you a way to plug the pc104 to a screen to see if it attempt to boot it some way?
https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/doc/howTo/Connect_VGA_to_PC104.pdf
Let me know

emilio-cartoni, 2017-02-10:
@julijenv 

By "reboot of the whole iCub" we mean that we stopped all applications/modules running, shut down pc104, powered off all iCub (head+motors) and then powered it on again.

We did not move the robot or change the power supplies.
We did not change anything on the laptop either (which has been updated recently when we sent our iCub to IIT last summer).

We do not have the materials to plug the pc104 to a VGA at the moment but we'll be able to do it on Monday.

hu-yue, 2017-02-10:
@emilio-cartoni did you try to reboot the USB of pc104 from an external laptop? 
It happened to me some times that I could not access PC104 and this was because it could not boot, by removing the USB and boot it from a laptop you can check if there are problems with the boot, if you are able to boot from the laptop then you can try to put it back in the robot and try to boot the PC104 again.

emilio-cartoni, 2017-02-10:
@hu-yue 
Thanks for the suggestion!
However, I'm a bit worried that the  OS on the USB stick would detect a different hardware (not being on the PC104 anymore) and change some settings that it would have to change back when placed into the PC104 again.
While this might work without problems, isn't there the chance to incur into some more trouble?

hu-yue, 2017-02-10:
@emilio-cartoni I don't think this should happen. At least it didn't happen to me. But if you're worried then it's better to wait for a reply from the iit support =)

julijenv, 2017-02-11:
@emilio-cartoni and thx btbw @hu-yue ,
please proceed as I suggest if you can connect the pc104 to a screen to see what's going on?
you could also try to just recreate another key and see if it starts this one or neither both, that could be a clue to some problem with the pc104 instead of the key. Let me know?

emilio-cartoni, 2017-02-13:
@julijenv thanks, we have connected it to a monitor.
Indeed it's a boot error, it says: 
_"Failed to load ldlinux.c32. 
Boot failed: please change disks and press a key to continue "_

We will now try to make a new USB stick and see how it goes.

emilio-cartoni, 2017-02-13:
@julijenv OK, pc104 is booting again successfully.
We have used the latest zip/live version.
However, we have now run into another problem, probably caused by the fact that we have changed the USB OS (not sure which version we were using before).

When we launch the icub-cluster.py script the application doesn't start the pc104 node - apparently everything runs fine and no errors appear in the terminal but the on/off check box near pc104 doesn't turn on and the yarpmanager then complains too (pc104 does not exist, check yarprun is running...)

julijenv, 2017-02-13:
HI @emilio-cartoni ,
actually the pblm is the RSA key of the new pc104 kUSB key, you need to authentificate the new USB key with the laptoop or pc you use as server.
For that when yu launch the icub-cluster.py shell window keep an eye on it when you try to connect the pc104 and you will see a message on the shell asking for you to confirm by Yes or Not that the key is realiable.
let me know,

emilio-cartoni, 2017-02-13:
@julijenv 
We have already validated the new RSA key (deleted the known host line using the suggested ssh-keygen -R command, accepted the new key, etc).
We have also run the procedure to automate again logging without password (ssh-copy-id ..).

We have tried to debug the problem by running manually the _yarprun --server /pc104_ command on the pc104 and it complained about the name server not being configured.

Yarp conf said there should be a file in /home/iCub/.config/yarp/yarp.conf but no file was there.
So we just created a file there on pc104 with a line "10.0.0.1 10000 yarp" in there (is it correct?).

After this, now icub-cluster.py runs successfully.

However, we have now a different problem: when we run yarpmanager and launch yarprobotinterface it fails, saying it cannot run yarprobotinterface on pc104.

julijenv, 2017-02-13:
Hi @emilio-cartoni ,
you could have done write the command:
yarp detect
instead of:
"So we just created a file there on pc104 with a line "10.0.0.1 10000 yarp" in there (is it correct?)."
Is it correct I dont really know, but as long as it did the trick I guess it works.
Your pblm avout the yarprobotinterface is I suppose:
DID YOU COMPILE YARP AND ICUB REPOSITORIES on the pc104?????
let me know

emilio-cartoni, 2017-02-13:
Hi @julijenv ,
We did not compile anything..we assumed the OS was distributed as "ready to run").
What should we compile and which instructions should we follow?

emilio-cartoni, 2017-02-13:
@julijenv 
Thanks for the support, our iCub is now working again (after compiling as described at http://wiki.icub.org/wiki/Compilation_on_the_pc104).

A few minor things happened during the compilation, which may not affect further operation but I'll list them here:
- pc104 date is set in 2008, so it leads to ccmake complaining that files are modified in the future while compiling. I used the date command to set manually pc104 time - ccmake still made some complaints about clock skew (as I wasn't so accurate to match laptop time exactly at second level precision) but it worked. After rebooting, pc104 went back to 2008.
- at some point the instructions say to _ccmake $icub_firmware_shared_ROOT_ however _$icub_firmware_shared_ROOT_ does not exist and it's not listed in the enviroment variables to be checked by the instructions). Anyway, replacing it with _ccmake .._ worked fine.
- The instructions ask to make some checks by doing _yarpdev --list_: the command returns a list but then ends with a _segmentation fault_ (is it normal?)
- instructions ask to check that the following devices exists:
_ Device "serial", C++ class ServerInertial, is a network wrapper,
 Device "serialport", C++ class SerialDeviceDriver, wrapped by "serial"
 Device "portaudio", C++ class ..._
I did find them in the list but they were listed as "available upon request" and their location on the build directory (iirc) was printed - so the output was actually different compared to the one of the instructions (but it seems it worked out ok anyway). Does the output need to match or is "available upon request" ok too?
- At some point instructions asked to enable _ENABLE_icubmod_icubheadcalibratorV2_ but I did not find it.
- While doing _make install_ for the icub-contrib-common, make complained about being unable to update the install_manifest.txt - I just bluntly rerun the make install with sudo.

barbalberto, 2017-02-14:
> pc104 date is set in 2008

command `sudo ntpdate-debian -u` will synch the clock and save the setting for next reboot

> at some point the instructions say to ccmake $icub_firmware_shared_ROOT however $icub_firmware_shared_ROOT does not exist and it's not listed in the enviroment variables to be checked by the instructions). Anyway, replacing it with ccmake .. worked fine.

You can add it in the ~/.bashrc_iCub file, however `ccmakwe ..` it is just fine

> The instructions ask to make some checks by doing yarpdev --list: the command returns a list but then ends with a segmentation fault (is it normal?)

No, it should not.
Use `yarp plugin --all` and post the output here, please.

> I did find them in the list but they were listed as "available upon request" 

It is ok. The devices now are plugins loaded at run-time and not linked at compile time. So yarpdev tells you which version of the library will be loaded "upon request"

> At some point instructions asked to enable ENABLE_icubmod_icubheadcalibratorV2 but I did not find it.

That device has been removed some time ago, I'll update the documentation. Thanks for reporting

> While doing make install for the icub-contrib-common, make complained about being unable to update the install_manifest.txt - I just bluntly rerun the make install with sudo

Here there is something wrong. Nothing shuld ever be installed with sudo on PC104
In the installation instruction, chapter `Configuring your Robot`
it says
_"Make sure that CMAKE_INSTALL_PREFIX points to $ICUBContrib_DIR"_

In the ccmake of icub contrib package you have to set the cmake install prefix equal to $ICUBContrib_DIR, so that the installation will put everything there. This is required for repositories which uses the icub_contrib to find it.

Please uninstall it from system folder and use this configuration.

emilio-cartoni, 2017-02-15:
@barbalberto 
> command sudo ntpdate-debian -u will synch the clock and save the setting for next reboot

I tried that command but it didn't work - in the end it turned out that there was already a ntp daemon running on the pc104, but it could not access internet.
Once I changed the gateway from 10.0.0.1 to 10.0.04 (our router) in /etc/network/interfaces the date was immediately fixed.

> You can add it in the ~/.bashrc_iCub file, however ccmake .. it is just fine

Ok thanks, maybe the Wiki could be updated as well, either by using _ccmake .._ there too or by listing $icub_firmware_shared_ROOT in the needed environment variables list.

> No, it should not.
Use yarp plugin --all and post the output here, please.

Ouch. Ok, I'll put the output in a separate post for easier reading.

>It is ok. The devices now are plugins loaded at run-time and not linked at compile time. So yarpdev tells you which version of the library will be loaded "upon request"

Good to know, thanks!

> That device has been removed some time ago, I'll update the documentation. Thanks for reporting

Ok!

> Here there is something wrong. Nothing shuld ever be installed with sudo on PC104
In the installation instruction, chapter Configuring your Robot
it says
"Make sure that CMAKE_INSTALL_PREFIX points to $ICUBContrib_DIR"
In the ccmake of icub contrib package you have to set the cmake install prefix equal to $ICUBContrib_DIR, so that the installation will put everything there. This is required for repositories which uses the icub_contrib to find it.
Please uninstall it from system folder and use this configuration.

Yes, I think I set the correct directory when installing.
The _install_manifest.txt_ says:
/>usr/local/src/robot/iCubContrib/share/ICUBcontrib/plugins/ICUBcontrib.ini
/usr/local/src/robot/iCubContrib/lib/ICUBcontrib/ICUBcontribConfig.cmake
/usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/ICUBcontribHelpers.cmake
/usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/ICUBcontribOptions.cmake
/usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/templates/icubcontrib-config-install.cmake.in
/usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/templates/icubcontrib-config-uninstall.cmake.in

It suspect the problem was that I used $ROBOT/icub-contrib-common/**build** (as suggested by the instructions) while I should have used $ROBOT/icub-contrib-common/**build-pc104** instead.

I tried to make the again make install in the build directory (without sudo) and it complained about being unable to access install_manifest.txt:
>icub@pc104:/usr/local/src/robot/icub-contrib-common/build$ make install
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/src/robot/iCubContrib/share/ICUBcontrib/plugins/ICUBcontrib.ini
-- Up-to-date: /usr/local/src/robot/iCubContrib/lib/ICUBcontrib/ICUBcontribConfig.cmake
-- Up-to-date: /usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/ICUBcontribHelpers.cmake
-- Up-to-date: /usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/ICUBcontribOptions.cmake
-- Up-to-date: /usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/templates/icubcontrib-config-install.cmake.in
-- Up-to-date: /usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/templates/icubcontrib-config-uninstall.cmake.in
CMake Error at cmake_install.cmake:175 (file):
  file Internal CMake error when trying to open file:
  /usr/local/src/robot/icub-contrib-common/build/install_manifest.txt for
  writing.

Installing from build-pc104 went fine instead:
> icub@pc104:/usr/local/src/robot/icub-contrib-common/build-pc104$ make install
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/src/robot/iCubContrib/share/ICUBcontrib/plugins/ICUBcontrib.ini
-- Installing: /usr/local/src/robot/iCubContrib/lib/ICUBcontrib/ICUBcontribConfig.cmake
-- Up-to-date: /usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/ICUBcontribHelpers.cmake
-- Up-to-date: /usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/ICUBcontribOptions.cmake
-- Up-to-date: /usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/templates/icubcontrib-config-install.cmake.in
-- Up-to-date: /usr/local/src/robot/iCubContrib/share/ICUBcontrib/cmake/templates/icubcontrib-config-uninstall.cmake.in

Thanks for the help!

emilio-cartoni, 2017-02-15:
@barbalberto 
Here's the output of _yarp plugin --all_ :

> icub@pc104:~$ yarp plugin --all
> Runtime plugins found:
> 
> device SerialServoBoard
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/SerialServoBoard.ini
>   * config:         SerialServoBoard (type device) (name SerialServoBoard) (library yarp_SerialServoBoard) (part SerialServoBoard)
>     Cannot find or load shared library
> 
> carrier bayer
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/bayer.ini
>   * config:         bayer (type carrier) (name bayer) (subtype recv) (library yarp_bayer) (part bayer_carrier) (code "not applicable")
>   * library:        /usr/local/src/robot/yarp/build-pc104/lib/yarp/yarp_bayer.so
>   * system version: 5
>   * class name:     yarp::os::BayerCarrier
>   * base class:     yarp::os::Carrier
> 
> device cuda
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/cuda.ini
>   * config:         cuda (type device) (name cuda) (library yarp_cuda) (part cuda)
>     Cannot find or load shared library
> 
> device dynamixelAX12Ftdi
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/dynamixelAX12Ftdi.ini
>   * config:         dynamixelAX12Ftdi (type device) (name dynamixelAX12Ftdi) (library yarp_dynamixelAX12Ftdi) (part dynamixelAX12Ftdi)
>     Cannot find or load shared library
> 
> device fakeIMU
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/fakeIMU.ini
>   * config:         fakeIMU (type device) (name fakeIMU) (library yarp_fakeIMU) (wrapper inertial)
>     Cannot find or load shared library
> 
> device fakeLaser
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/fakeLaser.ini
>   * config:         fakeLaser (type device) (name fakeLaser) (library fakeLaser) (part fakeLaser)
>     Cannot find or load shared library
> 
> device fakeMotionControl
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/fakeMotionControl.ini
>   * config:         fakeMotionControl (type device) (name fakeMotionControl) (library yarp_fakeMotionControl) (part fakeMotionControl)
>     Cannot find or load shared library
> 
> device fakebot
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/fakebot.ini
>   * config:         fakebot (type device) (name fakebot) (library yarp_fakebot) (part fakebot)
>     Cannot find or load shared library
> 
> device ffmpeg_grabber
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/ffmpeg.ini
>   * config:         ffmpeg_grabber (type device) (name ffmpeg_grabber) (library yarp_ffmpeg) (part ffmpeg_grabber) (wrapper grabber)
>     Cannot find or load shared library
> 
> device ffmpeg_writer
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/ffmpeg.ini
>   * config:         ffmpeg_writer (type device) (name ffmpeg_writer) (library yarp_ffmpeg) (part ffmpeg_writer)
>     Cannot find or load shared library
> 
> carrier human
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/human.ini
>   * config:         human (type carrier) (name human) (library yarp_human) (part human_carrier) (code HUMANITY)
>   * library:        /usr/local/src/robot/yarp/build-pc104/lib/yarp/yarp_human.so
>   * system version: 5
>   * class name:     HumanCarrier
>   * base class:     yarp::os::Carrier
> 
> device imuBosch_BNO055
>   * ini file:       /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins/imuBosch_BNO055.ini
>   * config:         imuBosch_BNO055 (type device) (name imuBosch_BNO055) (library imuBosch_BNO055) (wrapper inertial)
[... 318 lines cut ...]
> device skinWrapper
>   * ini file:       /usr/local/src/robot/icub-main/build-pc104/share/iCub/plugins/skinWrapper.ini
>   * config:         skinWrapper (type device) (name skinWrapper) (library skinWrapper)
>   * library:        /usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so
>   * system version: 5
>   * class name:     skinWrapper
>   * base class:     yarp::dev::DeviceDriver
> 
> device skinprototype
>   * ini file:       /usr/local/src/robot/icub-main/build-pc104/share/iCub/plugins/skinprototype.ini
>   * config:         skinprototype (type device) (name skinprototype) (library skinprototype) (wrapper skinWrapper)
>   * library:        /usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinprototype.so
>   * system version: 5
>   * class name:     SkinPrototype
>   * base class:     yarp::dev::DeviceDriver
> 
> device xsensmtx
>   * ini file:       /usr/local/src/robot/icub-main/build-pc104/share/iCub/plugins/xsensmtx.ini
>   * config:         xsensmtx (type device) (name xsensmtx) (library xsensmtx)
>   * library:        /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
>   * system version: 5
>   * class name:     yarp::dev::XSensMTx
>   * base class:     yarp::dev::DeviceDriver
> Segmentation fault
>

barbalberto, 2017-02-17:
Ok, can you post me the output of
`ldd /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so` ?

`gdb --args yarpdev --device xsensmtx `
`r (to run)`

after it gets segfault
` bt` and report the output

Thanks

emilio-cartoni, 2017-02-20:
Hi @barbalberto,

here's the ldd output:

> icub@pc104:~$ ldd /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
	linux-vdso.so.1 (0x00007fff1f5ff000)
	libYARP_OS.so.1 => /usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1 (0x00007fbaed10d000)
	libYARP_dev.so.1 => /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.1 (0x00007fbaecd1a000)
	libYARP_sig.so.1 => /usr/local/src/robot/yarp/build-pc104/lib/libYARP_sig.so.1 (0x00007fbaecadb000)
	libYARP_math.so.1 => /usr/local/src/robot/yarp/build-pc104/lib/libYARP_math.so.1 (0x00007fbaec8c1000)
	libYARP_init.so.1 => /usr/local/src/robot/yarp/build-pc104/lib/libYARP_init.so.1 (0x00007fbaec6bf000)
	libYARP_name.so.1 => /usr/local/src/robot/yarp/build-pc104/lib/libYARP_name.so.1 (0x00007fbaec4b3000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fbaec198000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fbaebe97000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fbaebc80000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fbaeb8d5000)
	libACE-6.2.8.so => /usr/lib/libACE-6.2.8.so (0x00007fbaeb52e000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fbaeb329000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fbaeb121000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fbaeaf04000)
	libreadline.so.6 => /lib/x86_64-linux-gnu/libreadline.so.6 (0x00007fbaeacb9000)
	libgsl.so.0 => /usr/lib/libgsl.so.0 (0x00007fbaea88c000)
	libgslcblas.so.0 => /usr/lib/libgslcblas.so.0 (0x00007fbaea64d000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fbaed732000)
	libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5 (0x00007fbaea423000)

As for the backtrace, the _yarpdev --device xsensmtx_ command does not segfault, here is its output (no backtrace if run with gdb).
> icub@pc104:~$ yarpdev --device xsensmtx
Failed to open com port /dev/ttyUSB0
yarpdev: ***ERROR*** driver <xsensmtx> was found but could not open
yarpdev: ***ERROR*** device not available.
>Suggestions:
>+ Do "yarpdev --list" to see list of supported devices.
>+ Or append "--verbose" option to get more information.

Running _yarp plugin --all_ with gdb gives the following result:
>icub@pc104:~$ gdb --args yarp plugin --all
[Lots of output here, as posted before]
Program received signal SIGSEGV, Segmentation fault.
0x0000000000000031 in ?? ()
(gdb) bt
#0  0x0000000000000031 in ?? ()
#1  0x00007ffff6601eaf in __cxa_finalize (d=0x7fffebd6ef40) at cxa_finalize.c:56
#2  0x00007fffebb08e53 in __do_global_dtors_aux () from /usr/local/src/robot/icub-main/build-pc104/lib/iCub/embObjStrain.so
#3  0x00007fffffffdd60 in ?? ()
#4  0x00007ffff7def77a in _dl_close_worker (map=<optimized out>) at dl-close.c:272
Backtrace stopped: frame did not save the PC
(gdb)

barbalberto, 2017-02-20:
Thanks for the report, we will see what's wrong. 
Are you able to run the robotInterface or is this error blocking?

emilio-cartoni, 2017-02-20:
@barbalberto 
No, it is not blocking, thanks.
Maybe we could even split the thread and close this one, as that segfault is probably unrelated to the original event (pc104 failing to boot).

closed 2018-01-16 by julijenv