robotology/mesh-support #393: Default ubuntu desktop environment crashes using gedit

gabrielenava, 2017-02-14:
### Description of the failure
I was trying to edit a file on `icub-head` pc using gedit. I accessed `icub-head` using ssh on `icub23` pc.

gedit opened as usual, but as soon as I pressed the `minimize` button the desktop environment crashed. All open windows and terminals on `icub23` were closed and the initial log-in windows appeared.  The problem is reproducible.

traversaro, 2017-02-14:
We just tried with @gabrielenava and the issue does not occur when using Metacity . 
At the moment I recommended him to use Metacity when working on the robot terminal. 

Having said that, this is just the last example of workflow disruption caused by Unity, see https://github.com/robotology/icub-support/issues/233 and first hand experiences from @drdanz, @pattacini, @randaz81, @francesco-romano, @claudiofantacci, @giuliavezzani and @alecive .
My (personal) strong impression is that Unity is simply not a good candidate to have a simple and reliable desktop environment for use in scientific lab. 

I am not a big expert of desktop environments, but I think it would make a lot of sense to just switch by default all robots laptops to something more stable, such as Metacity or XFCE. 
I think that any  other desktop environment would be ok, as long: 
  * It does not use any kind of graphics hardware acceleration, 
  * It has "normal" (i.e. Windows 95 style) menus and taskbar. 

cc @lornat75 @mbrunettini

barbalberto, 2017-02-14:
With `icub-head` do you mean the PC104 inside the iCub's head or a laptop?

traversaro, 2017-02-14:
`icub-head` is the new hostname for the former `pc104` COM Express based robots, such as `iCubGenova02` and `iCubGenova04`.

mbrunettini, 2017-02-14:
Please try to avoid install ans use graphics tool on the PC104 (refer [https://github.com/robotology/QA/issues/184](https://github.com/robotology/QA/issues/184) )

Since the graphical interface was not installed on PC104, many graphics tool may not work properly, regardless the local desktop manager used on the client

mbrunettini, 2017-02-14:
Regarding the crash, anyway, please note that the crash has happened on a Ubuntu 14.04 client : 

1. we already found a lot of problems and crashes on Ubuntu 14.04 related to nVidia graphics drivers and not (directly) to Unity ref. [https://github.com/robotology/icub-support/issues/215](https://github.com/robotology/icub-support/issues/215)
2. icub-head has Ubuntu 16.04, while client is still 14.04

closed 2017-02-14 by mbrunettini

traversaro, 2017-02-14:
I just tried, and apparently `canLoader` and `FirmwareUpdater` are minimizing fine on Unity. The problem is then only occurs in GTK+3 applications (gedit) on Unity.