robotology/mesh-support #394: Graphic interface frozen on icub21 and icub22

Nicogene, 2017-02-15:
It happens sometimes that the ubuntu graphic interface freeze on laptop **icub21** and **icub22**.
Any tips for resolving it?

cc @claudiofantacci

claudiofantacci, 2017-02-15:
When it happens, you can try to unfreeze the PC by keeping pressed the **Windows+ALT+F1** (Windows or equivalent button) while clicking with the mouse.

This problem is quite strange, but eventually you can recover from it without rebooting the PC.

closed 2017-02-15 by Nicogene

traversaro, 2017-04-20:
This just happened to @gabrielenava @MartaLorenzini on the `icub24` laptop on the `iCubGenova02` setup, running on Ubuntu 16.04 . It is quite a serious problem, has the average user has **NO** idea on how to restore the laptop to a working state. Even in this case, we had to look up to this issue to find the solution.

claudiofantacci, 2017-04-28:
The next Ubuntu release will roll back to Gnome as default DE.
I think we will also benefit from it 😄

randaz81, 2017-11-17:
in case `Windows+ALT+F1` does not work, try `ALT+ENTER`.

traversaro, 2017-12-10:
Happen also today on iiticubws014 .

traversaro, 2018-05-14:
It happened also today.