robotology/mesh-support #398: Right shoulder broken

giuliavezzani, 2017-02-16:
### Description of the failure
Right shoulder tendon broke performing normal movements.
The robot **really** needs some **low level checks**.

pattacini, 2017-02-16:
The cable popped out because the lower limit of the shoulder yaw wasn't aligned with the mechanical stop.
@randaz81 said there's a test to validate this particular correspondence. I'm afraid it has not been run.

@lornat75

julijenv, 2017-02-16:
Surely @randaz81 and @valegagge , could say more in details from when it has been released but @davidetome confirmed me that the test on joint_limit/mechanical_limit is being tested since iCubtwente01 - S/N: 030 (out of iCub Production in Feb 2016) and so iCubDarmstadt01 - S/N: 025 isn't part of the ones tested.

pattacini, 2017-02-16:
Well... ok please extend the tests on all the robot 😉

spada1978, 2017-02-17:
replaced the broken cables 1 and 8 and made the mechanical wiring
Cable break was caused by the higher values of the limits in the program

closed 2017-02-17 by spada1978

julijenv, 2017-02-20:
@spada1978 , olease nest ime dont forget to list all the needed component you used to repair.
IITCODE Qta Description
4758.1 x 1: shoulder cable N1
4765.1 x 1 : shoulder cable N4
1896 x 1 : right pulley crimp cover