robotology/mesh-support #402: wrong IMU port frequency (iCubGenova02)

gabrielenava, 2017-02-20:
### Description of the failure
Testing the IMU port frequency on iCubGenova02, we noticed that it oscillates between 40ms and 10ms. See also the attached graph (Yaxis data rate [s]; Xaxis number of samples)

[image]

francesco-romano, 2017-02-20:
probably related to https://github.com/robotology/yarp/issues/1090

traversaro, 2017-02-20:
@gabrielenava Can you clarify what you mean by : `Xaxis number of samples` ?

traversaro, 2017-02-20:
There are some things worth noticing.

All the robots configuration files do not explicitly specify a frequency for the IMU wrapper, see  https://github.com/robotology/robots-configuration/blob/master/iCubGenova02/hardware/inertial.xml . Consequently, they rely on the default period set in the `ServerInertial` (i.e. the IMU wrapper) : 
https://github.com/robotology/yarp/blob/8212ff101e952989cef0c81949fe0966b8802030/src/libYARP_dev/src/devices/ServerInertial/ServerInertial.cpp#L283

This periods should then be `5 [ms]`, while the data suggest that even in ideal situation the period is `10 ms`. 
Interestingly the `ServerInertial` is not using the `RateThread` (it was probably written when `RateThread` did not exist) and instead is relying on some custom logic, that could be related to this: 
https://github.com/robotology/yarp/blob/8212ff101e952989cef0c81949fe0966b8802030/src/libYARP_dev/src/devices/ServerInertial/ServerInertial.cpp#L481

Furthermore, in `icub-tests` we have a test for ports streaming frequency, but apparently the inertial was not on the list of ports tested: 
https://github.com/robotology/icub-tests/blob/master/suits/contexts/icub/robinterface_stream.ini

If this test is actually run by production, it may be worth to solve this issue and add this port to the list of the ports tested.

gabrielenava, 2017-02-20:
yes, it is the time between two consecutive samples

traversaro, 2017-02-20:
@gabrielenava "Words are important!" [1]

I update the original post removing the reference to "rate", please do it yourself next time. : )

[1] https://www.youtube.com/watch?v=so51Q8w1xlE (!!)

julijenv, 2017-03-28:
Hi @gabrielenava , @traversaro ,
any update on this matter, can we close it?

gabrielenava, 2017-03-28:
No updates. The imu has not been used for control since I opened this issue, so maybe is better not to close this issue until the imu is tested (probably after next week)

julijenv, 2017-08-08:
Hi @gabrielenava ,
is it posible to retest or  tell me if this issue has been solved by changing slome config files or setups?

traversaro, 2017-08-08:
Related RTF test: https://github.com/robotology/icub-tests/tree/master/src/ports-frequency .

julijenv, 2017-08-22:
Hi @gabrielenava ,
Is it still an ongoing issue?!!? or Did u solve it?

barbalberto, 2017-08-22:
it should be fixed by https://github.com/robotology/yarp/issues/1229

traversaro, 2017-08-22:
If @gabrielenava has still the code to generate that nice graph, it can check the IMU frequency the next time he uses iCubGenova02 or iCubGenova04 .

gabrielenava, 2017-08-22:
For generating the graph, this was done during a test with [RTF](https://github.com/robotology/robot-testing). I do not know if this graph is inside the code or was Ali's personal software.

barbalberto, 2017-08-22:
In robotology/yarp#1231 @diegoferigo said it was tested on the robot, maybe he knows which test was done? Have you guys using the robot experienced any issue after the fix?

gabrielenava, 2017-08-22:
The only issue directly related to this problem was the hardware fault one (see e.g https://github.com/robotology/codyco-modules/issues/225), that did not occur again on iCubGenova04 after the fix. As far as I know, iCubGenova02 has not been used in torque mode since a while, so the effectiveness of the fix on that robot is still to be verified. 

I think there is a way to enable a debug framework for verifying the IMU frequency, but maybe @diegoferigo can provide more details about this

diegoferigo, 2017-08-22:
I'm pretty confident my PR fixed this issue as well, [my tests](https://github.com/robotology/yarp/pull/1231#issuecomment-303709839) on the robot (`iCubGenova04`) have shown a big improvement in the thread's rate that likely was the culprit of what @gabrielenava experienced last winter.

barbalberto, 2017-08-22:
Great, I close the issue then

closed 2017-08-22 by barbalberto

traversaro, 2017-08-22:
If I remember correctly, the thread rate that was improved was the wholeBodyDynamics one, as it was not blocking anymore. Anyhow, moving the mutex was not affecting the fact that reading the IMU value was taking even 100 ms, see https://github.com/robotology/yarp/issues/1229#issuecomment-303646743 . Until we check on the robot, I am afraid we cannot conclude anything about the status of this issue.

barbalberto, 2017-08-22:
To be picky, this issue is about port frequency, not about reading time of IMU  device :stuck_out_tongue_closed_eyes: (but yes, they are related)
Moving the mutex reduces the time spent in the device driver, so all the threads reading data now are faster, both wholeBodyDynamics and inertialServer. So I think it is correct to say that port frequency are fine.

_If we still have doubts about how fast data are actually acquired by the sensor_ (I agree, a double check is better), _then we shall test this particular point_ , and if we are not happy I'd open a sepated issue (just to not confuse too many stuff in the same issue).

traversaro, 2017-08-22:
You are right, port frequency is now fine, and checking the duplicates is much more tricky (for example, you cannot test with the plot present in this issue). The problem of data reading can be discussed in a separate issue.