robotology/mesh-support #407: Fail left and right arm calibration. **iCubNottingham01**

pedrombmachado, 2017-02-22:
Please **fill in the following two sections** and then **remove any comment and/or remark become unnecessary**:
iCub **iCubNottingham01**
### Description of the failure
Every time that I run the yarprobotinterface I get the following errors: 
[ERROR]Right_Arm_Calibrator : checkGoneToZeroThreshold: joint  0  Timeout while going to zero! 
[[DEBUGERROR]Right_Arm_Calibrator: checkGoneToZeroThreshold: joint: 2 curr: 0.038 des: 0.060 -> delta: 0.022 threshold: 2.000 output: 246.000 mode: pos
[ERROR]Right_Arm_Calibrator : set 1 : some axis got timeout while reaching zero position... disabling this set of axes

[ERROR]Right_Arm_Calibrator : joint  0  is idle, skipping park 
[ERROR]Right_Arm_Calibrator : joint  1  is idle, skipping park 
[ERROR]Right_Arm_Calibrator : joint  2  is idle, skipping park 
[ERROR]Right_Arm_Calibrator : joint  3  is idle, skipping park ERROR

[ERROR]Left_Arm_Calibrator : joint  1  is idle, skipping park 
[ERROR]Left_Arm_Calibrator : joint  2  is idle, skipping park 
[ERROR]Left_Arm_Calibrator : joint  3  is idle, skipping park 

Could you please help me to diagnose and solve the above errors?
Many thanks,
Pedro
NTU

### Detailed conditions and logs of the failure
iCub LIVE image for PC104 version 6.0-16.04.05
Based on jessie amd64
Build on 05-Apr-2016
icub@pc104's password: 

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Feb 22 14:37:22 2017 from 10.0.0.1
Setting up yarp and iCub env vars
icub@pc104:~$ cat .bashrc_iCub | grep Nottin
export YARP_ROBOT_NAME=iCubNottingham01
icub@pc104:~$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/yarprobotinterface.ini] (pwd)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/yarprobotinterface.ini] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/yarprobotinterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./icub_all.xml
yarp: Port /log/pc104/yarprobotinterface/3655 active at tcp://10.0.0.2:10180
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./icub_all.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./cartesian/left_arm-cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./cartesian/right_arm-cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/head-cfw2_can0-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/head-cfw2_can0-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/mechanicals/head-cfw2_can0-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./wrappers/motorControl/head-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/torso-ems5-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/torso-ems5-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/mechanicals/torso-ems5-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./wrappers/motorControl/torso-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/left_upper_arm-ems1-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/left_upper_arm-ems1-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/mechanicals/left_upper_arm-ems1-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/left_lower_arm-ems2-mc.xml 
[... 1570 lines cut ...]
[INFO]from BOARD 10.0.1.2 (EB2), src CAN1, adr 7, time 1833s 315m 244u: CAN PRINT MESSAGE[id 0] -> PWM DIS:0
[INFO]from BOARD 10.0.1.4 (EB4), src CAN1, adr 7, time 1833s 359m 359u: CAN PRINT MESSAGE[id 1] -> PWM DIS:1
[INFO]from BOARD 10.0.1.2 (EB2), src CAN1, adr 7, time 1833s 336m 244u: CAN PRINT MESSAGE[id 1] -> PWM DIS:1
[INFO]from BOARD 10.0.1.2 (EB2), src CAN1, adr 7, time 1833s 356m 244u: CAN PRINT MESSAGE[id 2] -> PWM DIS:2
[INFO]from BOARD 10.0.1.4 (EB4), src CAN1, adr 7, time 1833s 379m 362u: CAN PRINT MESSAGE[id 2] -> PWM DIS:2
[INFO]from BOARD 10.0.1.4 (EB4), src CAN1, adr 7, time 1833s 399m 362u: CAN PRINT MESSAGE[id 3] -> PWM DIS:3
[INFO]from BOARD 10.0.1.2 (EB2), src CAN1, adr 7, time 1833s 376m 244u: CAN PRINT MESSAGE[id 3] -> PWM DIS:3
[DEBUG]left_arm_calibrator finished park device left_arm_mc_wrapper 
[DEBUG]right_arm_calibrator finished park device right_arm_mc_wrapper 
[DEBUG]Torso_Calibrator : Park  completed 
[DEBUG]torso_calibrator finished park device torso_mc_wrapper 
[INFO]All actions for action level 1 of interrupt1 phase finished. 
[INFO]interrupt1 phase finished. 
[INFO]shutdown phase starting... 
[INFO]Entering action level 2 of phase shutdown 
[INFO]Stopping icub/cartesianController/left_arm
[INFO]Stopping icub/cartesianController/right_arm
[INFO]All actions for action level 2 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 2 of shutdown phase finished. 
[INFO]Entering action level 5 of phase shutdown 
[INFO]All actions for action level 5 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 5 of shutdown phase finished. 
[INFO]Closing Server Inertial...
[INFO]Server Intertial thread finished
Segmentation fault

julijenv, 2017-02-22:
Hi @pedrombmachado ,
could you please send a picture of the robot as it is before launching yarprobobtinterface.
thx in advance

pedrombmachado, 2017-02-22:
Hi @julijenv ,
Please find attached the requested pictures
[image]
[image]
[image]
I am having another issue, because I had updated the EMS.hex version as suggested in the documentation:
Discovered a board @ 10.0.1.1: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.2: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.3: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.4: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.5: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.6: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.7: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.8: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.11: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.9: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.10: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
++++++++++++++++++++++++++++++++++++++++++++++
But now I am getting:
icub@pc104:~$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/yarprobotinterface.ini] (pwd)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/yarprobotinterface.ini] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/yarprobotinterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./icub_all.xml
yarp: Port /log/pc104/yarprobotinterface/2208 active at tcp://10.0.0.2:10074
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./icub_all.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./cartesian/left_arm-cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./cartesian/right_arm-cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/head-cfw2_can0-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/head-cfw2_can0-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/mechanicals/head-cfw2_can0-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./wrappers/motorControl/head-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/torso-ems5-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/torso-ems5-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/mechanicals/torso-ems5-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./wrappers/motorControl/torso-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/left_upper_arm-ems1-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/left_upper_arm-ems1-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/mechanicals/left_upper_arm-ems1-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/left_lower_arm-ems2-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/left_lower_arm-ems2-eln.xml 
[... cut ...]
[INFO]created device <parametricCalibratorEth>. See C++ class yarp::dev::parametricCalibratorEth for documentation.
[WARNING]There was some problem opening one or more devices. Please check the log and your configuration 
[ERROR]One or more devices failed opening... see previous log messages for more info 
[INFO]Closing Server Inertial...
[INFO]Server Intertial thread finished
Segmentation fault
+++++++
I think I have to update the can fw using the canLoader but I am not sure what file to upgrade per each board. 
Many thanks
Pedro

pedrombmachado, 2017-02-22:
Hi @julijenv ,
Please find below the requested fotos.
[image]
[image]
[image]
+++++++++++++++++++++++++++++++++
I have another issue to report because I ha've done the firmware update as suggested in the documentation:
icub@pc104:~$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/yarprobotinterface.ini] (pwd)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/yarprobotinterface.ini] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/yarprobotinterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./icub_all.xml
yarp: Port /log/pc104/yarprobotinterface/2208 active at tcp://10.0.0.2:10074
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./icub_all.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./cartesian/left_arm-cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./cartesian/right_arm-cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/head-cfw2_can0-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/head-cfw2_can0-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/mechanicals/head-cfw2_can0-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./wrappers/motorControl/head-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/torso-ems5-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/torso-ems5-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/mechanicals/torso-ems5-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./wrappers/motorControl/torso-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/left_upper_arm-ems1-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/left_upper_arm-ems1-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/mechanicals/left_upper_arm-ems1-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/left_lower_arm-ems2-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/left_lower_arm-ems2-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/mechanicals/left_lower_arm-ems2-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./wrappers/motorControl/left_arm-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/right_upper_arm-ems3-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/right_upper_arm-ems3-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/mechanicals/right_upper_arm-ems3-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNottingham01/./hardware/motorControl/right_lower_arm-ems4-mc.xml 
[... cut ...]
Discovered a board @ 10.0.1.7: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.8: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.9: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.11: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.10: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.

julijenv, 2017-02-23:
Hi @pedrombmachado ,
from my visit made to the FW guys, they tolld me that you update teh AYRP repository and by conseqently you need to update the firmware on the Ethernet board (EMS4) as well.
they should give you the right one to upload on them with EthLoader soon.
Hopefully today or tomorrow.

pedrombmachado, 2017-02-23:
Hi @julijenv,
   Many thanks for the update. Do you think that the FW update will solve the initial problem? Or am I doing something wrong?

julijenv, 2017-02-23:
Hi @pedrombmachado ,
I would have recommended to wait to fix your problem be4 updating everything but now it is too late.
at this point I would say let's wait to get the FW update and we will investigate the error message.
if you want you can already add icub.support in your contact on skype we may need it to debug, thx in advancce

pedrombmachado, 2017-02-23:
Hi @julijenv,
   Many thanks in advance. I have sent a skype invitation from pmbmwtf to icub.support. Do you think a good idea to downgrade the FW version? If yes where can I find the previous version 2.20?

julijenv, 2017-02-23:
Hi @pedrombmachado ,
my colleague's gonna give you some pings on what to do next and we will talk tomorrow about it

valegagge, 2017-02-23:
Hi @pedrombmachado ,

you need to update fw to the most recent version.
In details you need to update only fw of ethernet boards, the EMS boards.
In order to accomplish this task you can refer to this mini tutorial:
https://github.com/robotology/icub-firmware-build/blob/master/ETH/howto-eth-boards-simple-fw-update.txt

The executable is ems.hex ad it is in icub-firmware-build/ETH/EMS/bin/application/.

I hope this suggestions can be helpful .

pedrombmachado, 2017-02-23:
Hi @valegagge,
   Many thanks for your reply.
  I think I have done that yesterday and after the update the following error was generated.
[ERROR]EthResource::verifyEPprotocol() detected mismatching protocol version.minor in BOARD EB9 with IP 10.0.1.9 for eoprot_endpoint_motioncontrol : FW upgrade is required
[ERROR]EthResource::verifyEPprotocol() for ep = eoprot_endpoint_motioncontrol detected: pc104.version.minor = 16 and board.version.minor = 18
  Am I right or am I missing something?

The actual version on all the boards is now:
++++++++++++++++++++++++++++++++++++
icub@pc104:~$ ethLoader
Using default port 3333
Using default port 10.0.1.104
Usage: ethLoader --port n --address xxx.xxx.xxx.xxx

(process:2329): Gtk-WARNING **: Locale not supported by C library.
Using the fallback 'C' locale.
Gtk-Message: Failed to load module "canberra-gtk-module"
Discovered a board @ 10.0.1.1: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.2: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.3: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.4: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.5: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.6: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.7: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.8: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.9: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.11: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
Discovered a board @ 10.0.1.10: eobrd_ems4 w/ eApplication v 2.57 running protocol v 2 w/ capabilities = 0x60030030.
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

valegagge, 2017-02-24:
Hi @pedrombmachado ,

did you update the repository icub-firmware-shared? and you did you compiled it?
After compiling icub-firmware-shared, you need to recompile icub-main

pedrombmachado, 2017-02-24:
Hi @valegagge ,
 No I have not update the iCub-firmware-share nor compiled any of those. Can you suggest some documentation explaining the procedures step by step?
Many thanks in advance.

valegagge, 2017-02-24:
see http://wiki.icub.org/wiki/Compilation_on_the_pc104.

Anyway, you need to:
1) cd $ROBOT_CODE/icub-firmware-shared/build-pc104
2) git pull --rebase
2) cmake ..
3) make
4) cd $ICUB_DIR
5) cmake .. (here you don't need change any configuration)
6) make

Now all should works fine.

pedrombmachado, 2017-02-24:
Hi @valegagge ,
  Many thanks. I got the following error when on step 7. 
icub@icubsrv:/usr/local/src/robot/icub-main/build$ make
[  0%] Copying manager.py to /usr/local/src/robot/icub-main/build/bin/.
[  0%] Built target python-scripts
[  0%] Built target robotSymLink
[  0%] Copying icub-cluster scripts to /usr/local/src/robot/icub-main/build/bin/.
[  0%] Built target cluster-scripts
[  1%] Built target iCubDev
[  4%] Built target ctrlLib
[  7%] Built target skinDynLib
[  9%] Built target iKin
[ 11%] Built target iDyn
[ 16%] Built target learningMachine
[ 18%] Built target perceptiveModels
[ 19%] Built target actionPrimitives
[ 20%] Built target icubmod
[ 21%] Built target cartesiancontrollerserver
[ 22%] Built target cartesiancontrollerclient
[ 23%] Built target gazecontrollerclient
[ 24%] Built target optimization
[ 25%] Built target emotionInterface
[ 26%] Built target controlBoardDumper
[ 26%] Built target simpleClient
[ 27%] Built target testStereoMatch
[ 28%] Built target fingersTuner
[ 29%] Built target imuFilter
[ 29%] Built target imageSplitter
[ 29%] Built target trajectoryPlayer
[ 31%] Built target canLoaderLib
[ 31%] Built target canLoader-console
[ 32%] Built target canLoader
[ 32%] Building CXX object src/tools/ethLoader/ethLoaderLib/CMakeFiles/ethLoaderLib.dir/EthUpdater.cpp.o
/usr/local/src/robot/icub-main/src/tools/ethLoader/ethLoaderLib/EthUpdater.cpp: In member function ‘std::__cxx11::string EthUpdater::cmdGetMoreInfo(bool, ACE_UINT32)’:
/usr/local/src/robot/icub-main/src/tools/ethLoader/ethLoaderLib/EthUpdater.cpp:169:13: error: ‘struct eOuprot_cmd_MOREINFO_t’ has no member named ‘filler’
     command.filler[0] = EOUPROT_VALUE_OF_UNUSED_BYTE;
             ^
src/tools/ethLoader/ethLoaderLib/CMakeFiles/ethLoaderLib.dir/build.make:86: recipe for target 'src/tools/ethLoader/ethLoaderLib/CMakeFiles/ethLoaderLib.dir/EthUpdater.cpp.o' failed
make[2]: *** [src/tools/ethLoader/ethLoaderLib/CMakeFiles/ethLoaderLib.dir/EthUpdater.cpp.o] Error 1
CMakeFiles/Makefile2:3055: recipe for target 'src/tools/ethLoader/ethLoaderLib/CMakeFiles/ethLoaderLib.dir/all' failed
make[1]: *** [src/tools/ethLoader/ethLoaderLib/CMakeFiles/ethLoaderLib.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

pedrombmachado, 2017-02-27:
Hi @valegagge ,
 I managed to sort the error. However the main problem is remains. What should I do next?

++++++ Make log +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
icub@icubsrv:/usr/local/src/robot/icub-main/build$ make install
[  0%] Copying manager.py to /usr/local/src/robot/icub-main/build/bin/.
[  0%] Built target python-scripts
[  0%] Copying icub-cluster scripts to /usr/local/src/robot/icub-main/build/bin/.
[  0%] Built target cluster-scripts
[  0%] Built target iCubDev
[  3%] Built target ctrlLib
[  6%] Built target skinDynLib
[  8%] Built target iKin
[ 10%] Built target iDyn
[ 15%] Built target learningMachine
[ 17%] Built target perceptiveModels
[ 18%] Built target actionPrimitives
[ 18%] Built target icubmod
[ 19%] Built target cartesiancontrollerserver
[ 20%] Built target cartesiancontrollerclient
[ 21%] Built target gazecontrollerclient
[ 22%] Built target optimization
[ 23%] Built target emotionInterface
[ 24%] Built target controlBoardDumper
[ 24%] Built target simpleClient
[ 25%] Built target testStereoMatch
[ 26%] Built target fingersTuner
[ 27%] Built target imuFilter
[ 27%] Built target imageSplitter
[ 27%] Built target trajectoryPlayer
[ 28%] Built target imageBlender
[ 29%] Built target imageCropper
[ 31%] Built target canLoaderLib
[ 32%] Built target canLoader-console
[ 32%] Built target canLoader
[ 35%] Built target ethLoaderLib
[ 35%] Built target ethLoader
[ 35%] Automatic moc for target QtICubSkinGuiPlugin
[ 35%] Built target QtICubSkinGuiPlugin_automoc
[ 39%] Built target QtICubSkinGuiPlugin
[ 39%] Automatic moc for target iCubSkinGui-qt
[ 39%] Built target iCubSkinGui-qt_automoc
[ 40%] Built target iCubSkinGui-qt
[ 40%] Automatic moc for target frameGrabberGui2-qt
[ 40%] Built target frameGrabberGui2-qt_automoc
[ 44%] Built target frameGrabberGui2-qt
[ 44%] Automatic moc for target skinManagerGui-qt
[ 44%] Built target skinManagerGui-qt_automoc
[ 47%] Built target skinManagerGui-qt
[ 51%] Built target iCubSkinGui-gtk
[ 52%] Built target frameGrabberGui2-gtk
[ 53%] Built target skinManagerGui-gtk
[ 54%] Built target joystickCtrl
[ 54%] Built target joystickCheck
[ 54%] Built target focusTool
[ 54%] Automatic moc for target iCubGui
[ 54%] Built target iCubGui_automoc
[ 58%] Built target iCubGui
[ 59%] Built target stereoCalib
[ 61%] Built target depth2kin
[ 68%] Built target iCub_SIM
[ 69%] Built target simFaceExpressions
[ 70%] Built target ctpService
[ 73%] Built target lmpredict
[ 74%] Built target lmmerge
[ 75%] Built target lmtest
[ 79%] Built target lmtransform
[ 82%] Built target lmtrain
[ 82%] Built target cartesianInterfaceControl
[ 83%] Built target positionDirectControl
[... 1031 lines cut ...]
[WARNING] from BOARD 10.0.1.10 (EB10) @ 1393s 800m 530u: CAN discovery is OK but FAKE (without any control on CAN w/ get-fw-version<> message) for 13 eobrd_mtb boards with target can protocol ver 0.0 and application ver 0.0.0. Search time was 0 ms
[DEBUG] from BOARD 10.0.1.10 (EB10), src LOCAL, adr 0, time 1393s 800m 688u: (code 0x05000005, par16 0x0002 par64 0x3f0000fe00000000) -> CFG: EOtheSKIN can be correctly configured. num of patches in p16, can mapping in p64 upper 32 bits, prot and vers in p64 lower 32 bits + .
[INFO]created device <embObjSkin>. See C++ class EmbObjSkin for documentation.
[DEBUG]HostTransceiver::prepareTransceiverConfig2() has detected TXrateOfRegulars = 3 for BOARD w/ IP 10.0.1.11 
[DEBUG]HostTransceiver::prepareTransceiverConfig2() has detected capacityofTXpacket = 768 for BOARD w/ IP 10.0.1.11 
[DEBUG]HostTransceiver::prepareTransceiverConfig2() has detected maxSizeOfROP = 256 for BOARD w/ IP 10.0.1.11 
[DEBUG]HostTransceiver::prepareTransceiverConfig2() has detected protocolToUse = STANDARD and will prepare protocol for 4 jomos for BOARD w/ IP 10.0.1.11 
[DEBUG]TheEthManager::requestResource2(): has just succesfully created a new EthResource for IP =  10.0.1.11 
[INFO] from BOARD 10.0.1.11 (EB11), src LOCAL, adr 0, time 1394s 371m 433u: (code 0x00000034, par16 0x0001 par64 0x000000000000000c) -> SYS: the board has detected a restart of the remote transceiver because the rx ropframe sequence number is 1. In par64 there is the expected number. + .
[DEBUG]EthResource::setTXrate() has succesfully set the TX rate of the transceiver of BOARD EB11 with IP 10.0.1.11 at 3 ms 
[INFO] from BOARD 10.0.1.11 (EB11) @ 1394s 407m 304u: CAN discovery has started for 13 eobrd_mtb boards on (can1map, can2map) = (0x00fe, 0x3f00) with target can protocol ver 0.0 and application ver 0.0.0.
[WARNING] from BOARD 10.0.1.11 (EB11) @ 1394s 407m 486u: CAN discovery is OK but FAKE (without any control on CAN w/ get-fw-version<> message) for 13 eobrd_mtb boards with target can protocol ver 0.0 and application ver 0.0.0. Search time was 0 ms
[DEBUG] from BOARD 10.0.1.11 (EB11), src LOCAL, adr 0, time 1394s 407m 657u: (code 0x05000005, par16 0x0002 par64 0x3f0000fe00000000) -> CFG: EOtheSKIN can be correctly configured. num of patches in p16, can mapping in p64 upper 32 bits, prot and vers in p64 lower 32 bits + .
[INFO]created device <embObjSkin>. See C++ class EmbObjSkin for documentation.
[INFO]created device <parametricCalibratorEth>. See C++ class yarp::dev::parametricCalibratorEth for documentation.
[INFO]created device <parametricCalibratorEth>. See C++ class yarp::dev::parametricCalibratorEth for documentation.
[INFO]created device <parametricCalibratorEth>. See C++ class yarp::dev::parametricCalibratorEth for documentation.
[... cut ...]
[INFO]created device <parametricCalibratorEth>. See C++ class yarp::dev::parametricCalibratorEth for documentation.
[INFO]created device <parametricCalibratorEth>. See C++ class yarp::dev::parametricCalibratorEth for documentation.
[WARNING]There was some problem opening one or more devices. Please check the log and your configuration 
[ERROR]One or more devices failed opening... see previous log messages for more info 
[INFO]Closing Server Inertial...
[INFO]Server Intertial thread finished
Segmentation fault

valegagge, 2017-02-28:
Hi @pedrombmachado ,

I would notice you that you did 2 errors:
1) you had compiled on icubsrv and not on pc104.
2) you performed make install and not only make. In the link i sent you, there is written that is important don't run make install.

These two errors made a very dirty situation!
Please, next time it would be better you read the instructions more carefully.

Now I help you to restore a clean situation.

First of all, you need to keep in mind that pc104 and icubsrv share source code, but they have different compilations because different applications run on pc104 and on icubsrv and because they have different cpu. For this reason there are two folders: build and build-pc104.

Now we are going to uninstall all.
1) on icubsrv: in folder icub-main/build run "make uninstall"  (maybe sudo before)
2)  in the same folder remove CMakeCache.txt
3) on icubsrv in folder icub-firmware-shared/build-pc104: run "sudo make uninstall"
4) in the same folder remove CMakeCache.txt

5) on pc104(ssh pc104) : if,  in folder icub-main/build-pc104, you had run make install than run "make uninstall"
 and after remove file CMakeCache.txt.
6) on pc104 (ssh pc104) : if,  in folder icub-firmware-shared/build-pc104, you had run make install than run "make uninstall" and after remove file CMakeCache.txt.

After these commands you should be in a clean situation both on icubsrv than on pc104. 

Now we are going to compile applications:

**On icubsrv**:
1)move to  icub-main/build folder
2) ccmake ../
3) Hit 'c' to configure the project.
4) Select options:
 CMAKE_BUILD_TYPE to "Release"
 ENABLE_icubmod_cartesiancontrollerclient ON
 ENABLE_icubmod_cartesiancontrollerserver ON
 ENABLE_icubmod_gazecontrollerclient      ON
5) make (and not make install!!!!)

**on pc104:**
1) move to icub-firmware-shared/build-pc104
2) ccmake ../
3) Hit 'c' to configure the project.
4) Hit g to generate.
5) make
6)move to  icub-main/build-pc104 folder
7) ccmake ../
8) Hit 'c' to configure the project.
9) Select options:
 ENABLE_icubmod_dragonfly2
 ENABLE_icubmod_skinWrapper
 ENABLE_icubmod_xsensmtx
 ENABLE_icubmod_embObjAnalogSensor
 ENABLE_icubmod_embObjMotionControl
 ENABLE_icubmod_embObjSkin
 ENABLE_icubmod_embObjVirtualAnalogSensor
 ENABLE_icubmod_parametricCalibratorEth
 ENABLE_icubmod_cartesiancontrollerclient
 ENABLE_icubmod_cartesiancontrollerserver
 ENABLE_icubmod_cfw2can
 ENABLE_icubmod_skinprototype
 ENABLE_icubmod_icubheadcalibratorV2

10) make

Now if you run yarprobotinterface all should be ok!

I hope this instructions can help you to make robot running!
If you have any problem write me!

pedrombmachado, 2017-02-28:
Hi @valegagge ,
   Many thanks for your support. Sorry because I only saw the make install warning after having done the mistake. I managed to solve the the problem on the laptop (see the log). However I am having an error on the pc104.
1) First I cannot find the ccmake option ENABLE_icubmod_icubheadcalibratorV2.
[image]
2) After pressing g I get the following errors:
[image]
++++++++++++++++++ icubsrv ++++++++++++++++++++++++
icub@icubsrv:/usr/local/src/robot/icub-main/build$ make
Scanning dependencies of target python-scripts
[  0%] Copying manager.py to /usr/local/src/robot/icub-main/build/bin/.
[  0%] Built target python-scripts
[  0%] Copying icub-cluster scripts to /usr/local/src/robot/icub-main/build/bin/.
[  0%] Built target cluster-scripts
[  0%] Built target iCubDev
[  3%] Built target ctrlLib
[  5%] Built target skinDynLib
[  6%] Built target iKin
[  8%] Built target iDyn
[ 12%] Built target learningMachine
[ 14%] Built target perceptiveModels
[ 14%] Built target actionPrimitives
[ 14%] Built target icubmod
Scanning dependencies of target cartesiancontrollerclient
[ 14%] Building CXX object src/libraries/icubmod/cartesianController/CMakeFiles/cartesiancontrollerclient.dir/yarp_plugin_icubmod_cartesiancontrollerclient.cpp.o
[ 14%] Building CXX object src/libraries/icubmod/cartesianController/CMakeFiles/cartesiancontrollerclient.dir/ClientCartesianController.cpp.o
[ 15%] Linking CXX shared module ../../../../lib/iCub/cartesiancontrollerclient.so
[ 15%] Built target cartesiancontrollerclient
Scanning dependencies of target cartesiancontrollerserver
[ 15%] Building CXX object src/libraries/icubmod/cartesianController/CMakeFiles/cartesiancontrollerserver.dir/yarp_plugin_icubmod_cartesiancontrollerserver.cpp.o
[ 15%] Building CXX object src/libraries/icubmod/cartesianController/CMakeFiles/cartesiancontrollerserver.dir/ServerCartesianController.cpp.o
[ 16%] Building CXX object src/libraries/icubmod/cartesianController/CMakeFiles/cartesiancontrollerserver.dir/SmithPredictor.cpp.o
[ 16%] Linking CXX shared module ../../../../lib/iCub/cartesiancontrollerserver.so
[ 16%] Built target cartesiancontrollerserver
Scanning dependencies of target gazecontrollerclient
[ 16%] Building CXX object src/libraries/icubmod/gazeController/CMakeFiles/gazecontrollerclient.dir/yarp_plugin_icubmod_gazecontrollerclient.cpp.o
[ 16%] Building CXX object src/libraries/icubmod/gazeController/CMakeFiles/gazecontrollerclient.dir/ClientGazeController.cpp.o
[ 17%] Linking CXX shared module ../../../../lib/iCub/gazecontrollerclient.so
[ 17%] Built target gazecontrollerclient
[ 18%] Built target optimization
[ 19%] Linking CXX executable ../../../bin/emotionInterface
[ 19%] Built target emotionInterface
[ 19%] Linking CXX executable ../../../bin/controlBoardDumper
[ 21%] Built target controlBoardDumper
[ 22%] Linking CXX executable ../../../bin/simpleClient
[ 22%] Built target simpleClient
[ 22%] Linking CXX executable ../../../bin/testStereoMatch
[ 23%] Built target testStereoMatch
[ 23%] Linking CXX executable ../../../bin/fingersTuner
[ 25%] Built target fingersTuner
[ 25%] Linking CXX executable ../../../bin/imuFilter
[ 25%] Built target imuFilter
[ 25%] Linking CXX executable ../../../bin/imageSplitter
[ 26%] Built target imageSplitter
[ 27%] Linking CXX executable ../../../bin/trajectoryPlayer
[ 27%] Built target trajectoryPlayer
[ 27%] Linking CXX executable ../../../bin/imageBlender
[ 28%] Built target imageBlender
[ 28%] Linking CXX executable ../../../bin/imageCropper
[ 29%] Built target imageCropper
[ 30%] Built target canLoaderLib
[ 31%] Linking CXX executable ../../../../bin/canLoader-console
[ 31%] Built target canLoader-console
[ 31%] Linking CXX executable ../../../../bin/canLoader
[ 32%] Built target canLoader
[ 33%] Built target ethLoaderLib
[ 33%] Linking CXX executable ../../../../bin/ethLoader
[ 34%] Built target ethLoader
[ 35%] Automatic moc for target QtICubSkinGuiPlugin
[ 35%] Built target QtICubSkinGuiPlugin_automoc
[ 35%] Linking CXX shared module ../../../../lib/qt5/qml/robotology/icub/skingui/libQtICubSkinGuiPlugin.so
[... 58 lines cut ...]
[ 85%] Built target wholeBodyDynamics
[ 86%] Linking CXX executable ../../../bin/gravityCompensator
[ 86%] Built target gravityCompensator
[ 86%] Linking CXX executable ../../../bin/skinManager
[ 87%] Built target skinManager
[ 87%] Linking CXX executable ../../../bin/objectsPropertiesCollector
[ 88%] Built target objectsPropertiesCollector
[ 88%] Linking CXX executable ../../../bin/camCalib
[ 89%] Built target camCalib
[ 89%] Linking CXX executable ../../../bin/camCalibWithPose
[ 91%] Built target camCalibWithPose
[ 91%] Linking CXX executable ../../../bin/dualCamCalib
[ 93%] Built target dualCamCalib
[ 93%] Linking CXX executable ../../../bin/actionsRenderingEngine
[ 95%] Built target actionsRenderingEngine
[ 95%] Linking CXX executable ../../../bin/motionCUT
[ 95%] Built target motionCUT
[ 95%] Linking CXX executable ../../../bin/templatePFTracker
[ 96%] Built target templatePFTracker
[ 97%] Linking CXX executable ../../../bin/lumaChroma
[ 97%] Built target lumaChroma
[ 98%] Linking CXX executable ../../../bin/iKinGazeCtrl
[ 99%] Built target iKinGazeCtrl
[ 99%] Linking CXX executable ../../../bin/iKinCartesianSolver
[100%] Built target iKinCartesianSolver

valegagge, 2017-02-28:
hi @pedrombmachado !
for the first error: i'm sorry the flag ENABLE_icubmod_icubheadcalibratorV2 doesn't exist anymore. So for this reason you could not find it.  So you can conitue ignoring it.

For the second error, you need to go to $YARP_ROOT and run "git pull --rebase".
The cmake error tell you that YARP is not updated, so you need to update it with the command above.
After updating, you need to compile it both on icubsrv and pc104. this mean:
1) on icubsrv: cd $YARP_DIR; cmake ..; make
2) on  pc104: cd $YARP_DIR; cmake ..; make.

Since icub stuff depends on yarp, you need to recompile icub both on icubsrv and pc104. Using instruction of previous post to compile icub.

marcoaccame, 2017-02-28:
Dear @pedrombmachado (cc @valegagge), in reference to https://github.com/robotology/QA/issues/190, I would like to point out the following.

The content of your post in QA is:
```
Dear all,
I am trying to upgrade the CAN FW update on iCub 2.5 (Eth base). I am able to connect to see 
CAN 1 and CAN 2.  However I am not sure where to find the FW for each one one of CAN 1 and 2. 
I have updated CAN1 with the EMS.hex. and I got the following:
[DEBUG] BUS:id type version
[DEBUG] CAN1:4 8 2.1.-1
[DEBUG] CAN1:1 8 2.1.-1
[DEBUG] CAN1:2 8 2.1.-1
[DEBUG] CAN1:3 8 2.1.-1
[DEBUG] CAN2:13 6 3.0.4
However the version is lower. I think the type is now correct, but I lost the S/N of strain 
and it now says PROTOCOL N/A.
Therefore, I may be using the wrong FW version. should I use MCP2PLUS and/or MCP4PLUS? 
If so what file is to be downloaded to CAN1 and CAN2?
Many thanks in advance.
Pedro

```

It tells me that the FOC boards in CAN1 are running the bootloader and that the STRAIN board in CAN2 is running the application. You should read this information more easily from the GUI of the canLoader.

You also wrote: `I have updated CAN1 with the EMS.hex`. 

From the above information, I suspect that you have uploaded into the FOC boards the wrong FW and now the FOC boards are stuck into the bootloader, hence cannot control the motors.  Which motors, I cannot tell because you did not write the IP address of the EMS.

Thus, **before running yarprobotinterface you should ask for support in checking the status of the CAN FW**, at least of the boards which you have attempted re-programming.

pedrombmachado, 2017-02-28:
Hi @valegagge,
   Many thanks for your help. I was able to run make without errors on the icubserv, however I got the following error on the pc104 when I attempt to run cmake.
+++++++++++++++++++ PC104 +++++++++++++++++++++++++++++++++++++++++++++++
icub@pc104:/usr/local/src/robot/icub-main/build-pc104$ cd $YARP_DIR
icub@pc104:/usr/local/src/robot/yarp/build-pc104$ cmake ..
-- YARP Version: 2.3.68 (2.3.68+198-20170228.4+git3483cd3)
-- Looking for C++ include cstddef
-- Looking for C++ include cstddef - found
-- Performing Test CXX_HAS_WHEADER_GUARD
-- Performing Test CXX_HAS_WHEADER_GUARD - Failed
-- Performing Test CXX_HAS_WIGNORED_ATTRIBUTES
-- Performing Test CXX_HAS_WIGNORED_ATTRIBUTES - Failed
-- Performing Test CXX_HAS_WNEWLINE_EOF
-- Performing Test CXX_HAS_WNEWLINE_EOF - Failed
-- Performing Test CXX_HAS_WDANGLING_ELSE
-- Performing Test CXX_HAS_WDANGLING_ELSE - Failed
-- Performing Test CXX_HAS_WGCC_COMPAT
-- Performing Test CXX_HAS_WGCC_COMPAT - Failed
-- Performing Test CXX_HAS_WMICROSOFT_EXISTS
-- Performing Test CXX_HAS_WMICROSOFT_EXISTS - Failed
-- Performing Test CXX_HAS_WSTATIC_INLINE_EXPLICIT_INSTANTIATION
-- Performing Test CXX_HAS_WSTATIC_INLINE_EXPLICIT_INSTANTIATION - Failed
-- Performing Test CXX_HAS_WMISLEADING_INDENTATION
-- Performing Test CXX_HAS_WMISLEADING_INDENTATION - Failed
-- Performing Test CXX_HAS_WC__98_COMPAT
-- Performing Test CXX_HAS_WC__98_COMPAT - Failed
-- Performing Test CXX_HAS_WC__98_COMPAT_PEDANTIC
-- Performing Test CXX_HAS_WC__98_COMPAT_PEDANTIC - Failed
-- Performing Test CXX_HAS_STD_CXX98
-- Performing Test CXX_HAS_STD_CXX98 - Success
-- Performing Test CXX_HAS_STD_CXX14
-- Performing Test CXX_HAS_STD_CXX14 - Success
-- Performing Test CXX_HAS_STD_CXX17
-- Performing Test CXX_HAS_STD_CXX17 - Failed
-- Performing Test CXX_HAS_STD_CXX1Z
-- Performing Test CXX_HAS_STD_CXX1Z - Failed
-- Performing Test CXX_HAS_WERROR
-- Performing Test CXX_HAS_WERROR - Success
-- Performing Test CXX_HAS_WFATAL_ERROR
-- Performing Test CXX_HAS_WFATAL_ERROR - Success
CMake Warning at cmake/YarpDeprecatedWarning.cmake:7 (message):
  "CREATE_SHARED_LIBRARY" is deprecated in favour of "BUILD_SHARED_LIBS".
  Updating new cache entry.
Call Stack (most recent call first):
  cmake/YarpRenamedOption.cmake:10 (yarp_deprecated_warning)
  cmake/YarpOptions.cmake:111 (yarp_renamed_option)
  CMakeLists.txt:54 (include)

CMake Warning at cmake/YarpDeprecatedWarning.cmake:7 (message):
  "YARP_CLEAN_API" is deprecated.  Removing old cache entry.
Call Stack (most recent call first):
  cmake/YarpDeprecatedOption.cmake:10 (yarp_deprecated_warning)
  cmake/YarpOptions.cmake:252 (yarp_deprecated_option)
  CMakeLists.txt:54 (include)

CMake Warning at cmake/YarpDeprecatedWarning.cmake:7 (message):
  "CREATE_YARPBUILDER" is deprecated.  Removing old cache entry.
Call Stack (most recent call first):
  cmake/YarpDeprecatedOption.cmake:10 (yarp_deprecated_warning)
  cmake/YarpFindDependencies.cmake:272 (yarp_deprecated_option)
  CMakeLists.txt:62 (include)

CMake Warning at cmake/YarpDeprecatedWarning.cmake:7 (message):
  "CREATE_YARPMANAGER_PP" is deprecated.  Removing old cache entry.
Call Stack (most recent call first):
  cmake/YarpDeprecatedOption.cmake:10 (yarp_deprecated_warning)
  cmake/YarpFindDependencies.cmake:273 (yarp_deprecated_option)
  CMakeLists.txt:62 (include)

-- Detecting required libraries
[... 965 lines cut ...]
[ 96%] Building CXX object src/yarpmotorgui/CMakeFiles/yarpmotorgui.dir/startdlg.cpp.o
[ 97%] Building CXX object src/yarpmotorgui/CMakeFiles/yarpmotorgui.dir/qrc_res.cpp.o
[ 97%] Building CXX object src/yarpmotorgui/CMakeFiles/yarpmotorgui.dir/yarpmotorgui_automoc.cpp.o
[ 97%] Linking CXX executable ../../bin/yarpmotorgui
[ 97%] Built target yarpmotorgui
Scanning dependencies of target yarpbatterygui_automoc
[ 98%] Automatic moc for target yarpbatterygui
Generating moc_display.cpp
[ 98%] Built target yarpbatterygui_automoc
[ 98%] Generating ui_display.h
[ 98%] Generating qrc_res.cpp
Scanning dependencies of target yarpbatterygui
[ 99%] Building CXX object src/yarpbatterygui/CMakeFiles/yarpbatterygui.dir/main.cpp.o
[ 99%] Building CXX object src/yarpbatterygui/CMakeFiles/yarpbatterygui.dir/display.cpp.o
[ 99%] Building CXX object src/yarpbatterygui/CMakeFiles/yarpbatterygui.dir/qrc_res.cpp.o
[ 99%] Building CXX object src/yarpbatterygui/CMakeFiles/yarpbatterygui.dir/yarpbatterygui_automoc.cpp.o
[ 99%] Linking CXX executable ../../bin/yarpbatterygui
[ 99%] Built target yarpbatterygui
[ 99%] Automatic moc for target yarplaserscannergui
[ 99%] Built target yarplaserscannergui_automoc
Scanning dependencies of target yarplaserscannergui
[ 99%] Building CXX object src/yarplaserscannergui/CMakeFiles/yarplaserscannergui.dir/main.cpp.o
[ 99%] Building CXX object src/yarplaserscannergui/CMakeFiles/yarplaserscannergui.dir/yarplaserscannergui_automoc.cpp.o
[100%] Linking CXX executable ../../bin/yarplaserscannergui
[100%] Built target yarplaserscannergui

pedrombmachado, 2017-02-28:
Hi @marcoaccame , cc @valegagge ,
I forgot to mention that the ip is 10.0.1.1. Please check the image below
[image]

pedrombmachado, 2017-02-28:
Hi @marcoaccame , cc @valegagge ,
I forgot to mention that the ip is 10.0.1.1.

valegagge, 2017-03-01:
Hi @pedrombmachado ,
in your log you can find the description of error:

> CMake Error at cmake/YarpFindDependencies.cmake:192 (message):
Optional package Eigen3 not found. Please install it or enable the option
"CREATE_LIB_MATH_USING_GSL" to build yarp.
Call Stack (most recent call first):
cmake/YarpFindDependencies.cmake:464 (check_skip_dependency)
CMakeLists.txt:62 (include)

this means you need eigen3 library: so please install libeigen3-dev and compile again.

pedrombmachado, 2017-03-01:
Hi @valegagge ,
   Many thanks for all your support. I think I am getting close.
  I have good and bad news. I was able to compile YARP on the pc104 (see the log) but now I am getting a new error when I try to compile the icub-main. It says fatal error: yarp/math/SVD.h: No such file or directory

However if I run
icub@pc104:/usr/local/src/robot/yarp/build-pc104$ locate SVD.h
/lib/live/mount/persistence/loop1/usr/include/eigen3/Eigen/src/Eigen2Support/SVD.h
/lib/live/mount/persistence/loop1/usr/include/eigen3/Eigen/src/SVD/JacobiSVD.h
/usr/include/eigen3/Eigen/src/Eigen2Support/SVD.h
/usr/include/eigen3/Eigen/src/SVD/JacobiSVD.h
/usr/local/src/robot/yarp/src/libYARP_math/include/yarp/math/SVD.h

++++ build of icub-main on pc104 ++++++
icub@pc104:/usr/local/src/robot/icub-main/build-pc104$ make
Scanning dependencies of target python-scripts
[  0%] Copying manager.py to /usr/local/src/robot/icub-main/build-pc104/bin/.
[  0%] Built target python-scripts
[  0%] Copying icub-cluster scripts to /usr/local/src/robot/icub-main/build-pc104/bin/.
[  0%] Built target cluster-scripts
[  0%] Built target iCubDev
[  1%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/math.cpp.o
/usr/local/src/robot/icub-main/src/libraries/ctrlLib/src/math.cpp:23:27: fatal error: yarp/math/SVD.h: No such file or directory
 #include <yarp/math/SVD.h>
                           ^
compilation terminated.
src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/build.make:54: recipe for target 'src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/math.cpp.o' failed
make[2]: *** [src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/math.cpp.o] Error 1
CMakeFiles/Makefile2:736: recipe for target 'src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/all' failed
make[1]: *** [src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/all] Error 2
Makefile:107: recipe for target 'all' failed
make: *** [all] Error 2

====== yarp on pc104 ============
icub@pc104:/usr/local/src/robot/yarp/build-pc104$ cmake ..
-- YARP Version: 2.3.68 (2.3.68+198-20170228.4+git3483cd3)
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Check size of short
-- Check size of short - done
-- Check size of int
^C
icub@pc104:/usr/local/src/robot/yarp/build-pc104$ cd ..
icub@pc104:/usr/local/src/robot/yarp$ git pull --rebase
remote: Counting objects: 66, done.
remote: Compressing objects: 100% (36/36), done.
remote: Total 66 (delta 29), reused 18 (delta 18), pack-reused 12
Unpacking objects: 100% (66/66), done.
From https://github.com/robotology/yarp
   3483cd3..4ee3122  master     -> origin/master
   bbdfc73..c15a467  RPLIDAR2   -> origin/RPLIDAR2
   9dbb949..768bcf9  devel      -> origin/devel
First, rewinding head to replay your work on top of it...
Fast-forwarded master to 4ee312257c5e948171fe43ff8ca0977fa7603742.
icub@pc104:/usr/local/src/robot/yarp$ cd build-pc104/
icub@pc104:/usr/local/src/robot/yarp/build-pc104$ cmake ..
-- YARP Version: 2.3.68 (2.3.68+204-20170228.10+git4ee3122)
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
[... 605 lines cut ...]
Scanning dependencies of target yarprobotinterface
[ 96%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/main.cpp.o
[ 96%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Action.cpp.o
[ 97%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/CalibratorThread.cpp.o
[ 97%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Device.cpp.o
[ 97%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Module.cpp.o
[ 98%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Param.cpp.o
[ 98%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Robot.cpp.o
[ 98%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Types.cpp.o
[ 99%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/XMLReader.cpp.o
/usr/local/src/robot/yarp/src/yarprobotinterface/XMLReader.cpp:97:13: warning: ‘std::string {anonymous}::DocTypeToString({anonymous}::RobotInterfaceDTD::DocType)’ defined but not used [-Wunused-function]
 std::string DocTypeToString(RobotInterfaceDTD::DocType doctype) {
             ^
[ 99%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/idl_generated_code/src/yarprobotinterfaceRpc.cpp.o
Linking CXX executable ../../bin/yarprobotinterface
[ 99%] Built target yarprobotinterface
Scanning dependencies of target yarpmanager-console
[ 99%] Building CXX object src/yarpmanager-console/CMakeFiles/yarpmanager-console.dir/ymanager.cpp.o
[ 99%] Building CXX object src/yarpmanager-console/CMakeFiles/yarpmanager-console.dir/yarpmanager.cpp.o
Linking CXX executable ../../bin/yarpmanager-console
[ 99%] Built target yarpmanager-console
Scanning dependencies of target yarpdatadumper
[100%] Building CXX object src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/main.cpp.o
Linking CXX executable ../../bin/yarpdatadumper
[100%] Built target yarpdatadumper

valegagge, 2017-03-01:
Did you run ccmake ... in icub-main/build-pc104 before running make?

If not could you try to run ccmake and than make?

If yes,  could you try to remove the cache file and recompiled icub-main with flags enabled that I told you some post ago? The cachefile is inside icub-main/build-pc104/CMakeCache.txt

pedrombmachado, 2017-03-01:
Hi @valegagge,
   Yes, I followed all the following steps that you posted:
on pc104:

    move to icub-firmware-shared/build-pc104

    ccmake ../

    Hit 'c' to configure the project.

    Hit g to generate.

    make
    6)move to icub-main/build-pc104 folder

    ccmake ../

    Hit 'c' to configure the project.

    Select options:
    **ENABLE_icubmod_dragonfly2**
    **ENABLE_icubmod_skinWrapper**
    **ENABLE_icubmod_xsensmtx**
    **ENABLE_icubmod_embObjAnalogSensor**
    **ENABLE_icubmod_embObjMotionControl**
    **ENABLE_icubmod_embObjSkin**
    **ENABLE_icubmod_embObjVirtualAnalogSensor**
    **ENABLE_icubmod_parametricCalibratorEth**
    **ENABLE_icubmod_cartesiancontrollerclient**
    **ENABLE_icubmod_cartesiancontrollerserver**
    **ENABLE_icubmod_cfw2can**
    **ENABLE_icubmod_skinprototype**
    **_ENABLE_icubmod_icubheadcalibratorV2 (Not available)_**

    7) make
======= CCMAKE configuration ==========
CMAKE_BUILD_TYPE                *Release                                                                                                                                                                  
 CMAKE_CXX_COMPILER              */usr/bin/c++                                                                                                                                                             
 CMAKE_C_COMPILER                */usr/bin/cc                                                                                                                                                              
 CMAKE_INSTALL_PREFIX            */usr/local                                                                                                                                                               
 ENABLE_icubmod_analogServer     *OFF                                                                                                                                                                      
 ENABLE_icubmod_bcbBattery       *OFF                                                                                                                                                                      
 ENABLE_icubmod_bmsBattery       *OFF                                                                                                                                                                      
 ENABLE_icubmod_canBusAnalogSen  *OFF                                                                                                                                                                      
 ENABLE_icubmod_canBusDoubleFTS  *OFF                                                                                                                                                                      
 ENABLE_icubmod_canBusInertialM  *OFF                                                                                                                                                                      
 ENABLE_icubmod_canBusSkin       *OFF                                                                                                                                                                      
 ENABLE_icubmod_canBusVirtualAn  *OFF                                                                                                                                                                      
 ENABLE_icubmod_canmotioncontro  *OFF                                                                                                                                                                      
 **ENABLE_icubmod_cartesiancontro  *ON                                                                                                                                                                       
 ENABLE_icubmod_cartesiancontro  *ON**                                                                                                                                                                       
 **ENABLE_icubmod_cfw2can          *ON**                                                                                                                                                                       
 **ENABLE_icubmod_dragonfly2       *ON**                                                                                                                                                                       
 ENABLE_icubmod_dragonfly2raw    *OFF                                                                                                                                                                      
 ENABLE_icubmod_ecan             *OFF                                                                                                                                                                      
 **ENABLE_icubmod_embObjAnalogSen  *ON**                                                                                                                                                                       
 ENABLE_icubmod_embObjInertials  *OFF                                                                                                                                                                      
 ENABLE_icubmod_embObjMais       *OFF                                                                                                                                                                      
[... cut ...]
 ICUB_USE_ACE                    *ON                                                                                                                                                                       
 ICUB_USE_GLUT                   *OFF                                                                                                                                                                      
 ICUB_USE_GSL                    *ON                                                                                                                                                                       
 ICUB_USE_GTK2                   *ON

pedrombmachado, 2017-03-01:
Hi @marcoaccame , cc: @valegagge 
I managed to upgrade the firmware of CAN1 ID2 by following the procedure described in 
http://wiki.icub.org/wiki/Firmware.
    1) open the CanLoader GUI.
    2) select the can bus number and the can bus device driver from the menus and click 'Connect'.
    3) select the F/T sensor (click on the checkbox on the left).
    4) select the firmware file from the file selector on the right (default location: iCub\src\firmware\build\2FOC.hex).
    5) click 'Start Download'.
As described in table Services offered by the ETH boards : for ip 10.0.1.1, CAN1.  However I noticed that I was only able to program CAN1: ID2 because the canLoader is **not listing CAN1: ID1, ID3 and ID4**
[image]
Can you please advise on how to recover CAN1: ID1, ID3 and ID4?
Many thanks,
Pedro

valegagge, 2017-03-02:
hi @pedrombmachado ,
did you removed the cachefile is inside icub-main/build-pc104/CMakeCache.txt and than performed steps above?

marcoaccame, 2017-03-02:
Dear @pedrombmachado, maybe you misspelled. 
You meant `select the eobrd_foc board w/ address CAN1:21, did you not?

You must program eobrd_foc with 2FOC.hex, eobrd_strain with strain.hex etc.

About the other eobrd_foc boards at addresses CAN1:1, CAN1:3, and CAN1:4 .... They probably don't show up because maybe you programmed them w/ wrong FW and the boards crash.

You can recover them by catching them in the first five seconds after you switch the motors on. In such a way the 2FOC boards execute the bootloader and don't crash yet.

However, I would recommend to book a joint session w/ someone of the support team who will guide you through the operation.

pedrombmachado, 2017-03-02:
Hi @valegagge ,
   Yes, I done that a couple of times and it keeps complaining about the SVD.h. In fact, I have done everything from the scratch on the pc104. Compiled the yarp, icub-firmware-shared and then icub-main. Following all the steps that you suggested. Any idea why I am getting this error?
[  1%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/math.cpp.o
/usr/local/src/robot/icub-main/src/libraries/ctrlLib/src/math.cpp:23:27: fatal error: yarp/math/SVD.h: No such file or directory
 #include <yarp/math/SVD.h>
                           ^
compilation terminated.
src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/build.make:54: recipe for target 'src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/math.cpp.o' failed
make[2]: *** [src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/math.cpp.o] Error 1
CMakeFiles/Makefile2:736: recipe for target 'src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/all' failed
make[1]: *** [src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/all] Error 2
Makefile:107: recipe for target 'all' failed
make: *** [all] Error 2
icub@pc104:/usr/local/src/robot/icub-main/build-pc104$ cat CMakeCache.txt 
# This is the CMakeCache file.
# For build in directory: /usr/local/src/robot/icub-main/build-pc104
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

pedrombmachado, 2017-03-02:
Hi @marcoaccame  and @julijenv ,
@marcoaccame many thanks for the clarification. @julijenv  can you suggest someone to walk me through the process of upgrading the FW on the eobrd_foc boards at addresses CAN1:1, CAN1:3, and CAN1:4 as @marcoaccame suggested? 
Many thanks in advance,

julijenv, 2017-03-02:
Hi @pedrombmachado ,
polese add icub.support on you contact list on skype,
I'll try to give you my humble help to resolve the problem on which you struggle.

pedrombmachado, 2017-03-02:
Hi @valegagge,
  I happy to inform that the software problem is now sorted. I have basically installed a fresh installation of YARP, icub-firmware-shared and icub-main. I think the error was given because yarp needs also this options enabled:
on the icubsrv
CREATE_GUIS, set to ON
CREATE_LIB_MATH, set to ON
http://www.yarp.it/install_yarp_linux.html
and on pc104
 CREATE_LIB_MATH, set to ON
 CREATE_DEVICE_LIBRARY_MODULES:ON
 ENABLE_yarpmod_serial
 ENABLE_yarpmod_serialport
 ENABLE_yarpmod_portaudio
http://wiki.icub.org/wiki/Compilation_on_the_pc104
But many thanks for your support. It was really helpful.
Pedro

valegagge, 2017-03-03:
Perfect!
Can I close this issue?

pedrombmachado, 2017-03-03:
Hi @valegagge 
Not really,
 Because we are still solving the FW issue and then run the calibration.

pedrombmachado, 2017-03-03:
Hi @julijenv ,
   Can we chat in skype?

pedrombmachado, 2017-03-08:
Hi @julijenv and @valegagge 
  I am trying to run the yarprobotinterface after the yarp, icub-firmware-share and icub-main compilation and I am getting the following error:
  ||| did not find yarprobotinterface.ini
yarp: cannot read from yarprobotinterface.ini
[FATAL]Missing "config" argument 

Any idea how to solve this error? I found this issue https://github.com/robotology/icub-support/issues/289 that looks very similar to the error that I am getting.

======================== log ====================================================
icub@pc104:~$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/yarprobotinterface.ini] (pwd)
||| checking [/home/icub/.config/yarp/yarprobotinterface.ini] (YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/yarprobotinterface.ini] (YARP_DATA_HOME)
||| checking [/etc/yarp/yarprobotinterface.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| did not find yarprobotinterface.ini
yarp: cannot read from yarprobotinterface.ini
[FATAL]Missing "config" argument 
yarp: Port /log/pc104/yarprobotinterface/1047 active at tcp://10.0.0.2:10004/
LogForwarder error while connecting port /log/pc104/yarprobotinterface/1047
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(+0xab916) [0x7f373e8ec916]
yarprobotinterface() [0x406034]
yarprobotinterface() [0x40d06b]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE+0x42) [0x7f373e98d3a2]
yarprobotinterface() [0x4059cc]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f373d451b45]
yarprobotinterface() [0x405c77]
yarprobotinterface() [0x40620d]
yarprobotinterface() [0x40d06b]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE+0x42) [0x7f373e98d3a2]
yarprobotinterface() [0x4059cc]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f373d451b45]
yarprobotinterface() [0x405c77]

julijenv, 2017-03-08:
Hi @pedrombmachado ,
first try to launch yarprobotinterface from where you modify the file icub.xml (it would have been nicer to create a new one called like icub-no-left-arm.xml on which you should have comment any instance of the left arm). when inside this repository it should be able to find yarprobotinterface,ini (if you created icub-no-left-arm.xml please open yarprobotinterface.ini to point to this file).
If the robot launched it means that you need to update the path on the bash file of icub to append the path from where the robot was able to get launched.
let me know

marcoaccame, 2017-03-08:
Did you install robots-configuration?
See https://github.com/robotology/QA/issues/143.

pedrombmachado, 2017-03-08:
Hi @marcoaccame ,
   Problem solved with the support of @julijenv .
@julijenv I have now another problem. So I have changed the configuration of icub_no_left_arm.xml as you suggested (check bellow) and I am now getting the following error:

+++++++++++++++++++++ ERROR log +++++++++++++++++++++++++
icub@pc104:~$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/yarprobotinterface.ini] (pwd)
||| checking [/home/icub/.config/yarp/yarprobotinterface.ini] (YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/yarprobotinterface.ini] (YARP_DATA_HOME)
||| checking [/etc/yarp/yarprobotinterface.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/robots-configuration/iCubNottingham01/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/robots-configuration/iCubNottingham01/yarprobotinterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_no_left_arm.xml] (pwd)
||| checking [/home/icub/.config/yarp/./icub_no_left_arm.xml] (YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/./icub_no_left_arm.xml] (YARP_DATA_HOME)
||| checking [/etc/yarp/./icub_no_left_arm.xml] (YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/./icub_no_left_arm.xml] (YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/./icub_no_left_arm.xml] (YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/./icub_no_left_arm.xml] (YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/robots-configuration/iCubNottingham01/./icub_no_left_arm.xml] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/robots-configuration/iCubNottingham01/./icub_no_left_arm.xml
yarp: Port /log/pc104/yarprobotinterface/959 active at tcp://10.0.0.2:10033/
LogForwarder error while connecting port /log/pc104/yarprobotinterface/959
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./icub_no_left_arm.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./cartesian/right_arm-cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/motorControl/head-cfw2_can0-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/electronics/head-cfw2_can0-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/mechanicals/head-cfw2_can0-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./wrappers/motorControl/head-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/motorControl/torso-ems5-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/electronics/torso-ems5-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/mechanicals/torso-ems5-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/motorControl/torso-ems5-mc-service.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./wrappers/motorControl/torso-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/motorControl/right_upper_arm-ems3-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/electronics/right_upper_arm-ems3-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/mechanicals/right_upper_arm-ems3-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/motorControl/right_upper_arm-ems3-mc-service.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/motorControl/right_lower_arm-ems4-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/electronics/right_lower_arm-ems4-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/mechanicals/right_lower_arm-ems4-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/robots-configuration/iCubNottingham01/./hardware/motorControl/right_lower_arm-ems4-mc-service.xml 
[... cut ...]
[INFO]created wrapper <virtualAnalogServer>. See C++ class yarp::dev::VirtualAnalogWrapper for documentation.
[DEBUG](period 10) (networks (FirstSetOfJoints SecondSetOfJoints)) (channels 7) (robotName icub) (SecondSetOfJoints 4 6 0 2) (name "/icub/joint_vsens/right_arm:i") (device virtualAnalogServer) (FirstSetOfJoints 0 3 0 3) 
[DEBUG]Using VirtualAnalogServer
 
yarp: Port /icub/joint_vsens/right_arm:i active at tcp://10.0.0.2:10045/
[INFO]created wrapper <virtualAnalogServer>. See C++ class yarp::dev::VirtualAnalogWrapper for document

marcoaccame, 2017-03-09:
at first, I suggest to improve readibility of post by attaching long text files rather than cutting and paste them into the window.

then, let me explain some important prints of the log. The rule of thumb is to search for `[ERROR]` lines because tell you what is wrong. The `[INFO]` lines tells you something which can help you. 

1. The device canbusmotion control is not compiled. It is needed by the head. You need to compile it by enabling proper variable in cmake. Follow however, instructions on links already posted by Valentina.
```
[ERROR]Cannot find "canmotioncontrol" plugin (not built in, and no .ini file found for it)
[ERROR]Check that YARP_DATA_DIRS leads to at least one directory with 
plugins/canmotioncontrol.ini or share/yarp/plugins/canmotioncontrol.ini in it
yarpdev: ERROR could not find device 
```

2. the FW on the EMS board 10.0.1.5 is OK. The `[INFO]` line tells you that it is running version 2.57, which is the latest version. Do `git log icub-firmware-build/ETH/EMS/bin/application/ems.hex` to verify.
I assume also other EMS boards have the correct FW.
```
[INFO]EthResource::askBoardVersion() found BOARD EB5 @ IP 10.0.1.5 of type ems4 with 
FW version = ( 2 , 57 ) and build date 2016 Dec 06 13:33
```

3. The FW on 2FOC boards attached to 10.0.1.5 is incompatible. The .xml files specify a FW w/ CAN protocol 1.5 or later w/ any FW version. But yarprobotinterface detects boards w/ protocol 1.4. 
What you need to do is to update the FW on the 2FOC boards. 
```
[INFO] from BOARD 10.0.1.5 (EB5) @ 76s 157m 547u: CAN discovery has started for 3 eobrd_foc 
 boards on (can1map, can2map) = (0x001a, 0x0000) with target can protocol ver 1.5 and 
 application ver 0.0.0.
[WARNING] from BOARD 10.0.1.5 (EB5) @ 76s 158m 316u: CAN discovery has detected a eobrd_foc 
 board in CAN1 addr 1 with can protocol ver 1.4 and application ver 3.1.0 Search time was 0 ms
[WARNING] from BOARD 10.0.1.5 (EB5) @ 76s 158m 456u: CAN discovery has detected a eobrd_foc
 board in CAN1 addr 3 with can protocol ver 1.4 and application ver 3.1.0 Search time was 0 ms
[WARNING] from BOARD 10.0.1.5 (EB5) @ 76s 158m 595u: CAN discovery has detected a eobrd_foc
 board in CAN1 addr 4 with can protocol ver 1.4 and application ver 3.1.0 Search time was 0 ms
[ERROR] from BOARD 10.0.1.5 (EB5) @ 76s 158m 743u: CAN discovery has detected 3 invalid 
 eobrd_foc boards in CAN1:
[ERROR]1 of 3: wrong eobrd_foc BOARD 10.0.1.5:CAN1:1 because it has: WRONG PROTOCOL VERSION
[ERROR]2 of 3: wrong eobrd_foc BOARD 10.0.1.5:CAN1:3 because it has: WRONG PROTOCOL VERSION
[ERROR]3 of 3: wrong eobrd_foc BOARD 10.0.1.5:CAN1:4 because it has: WRONG PROTOCOL VERSION
[ERROR]embObjMotionControl::open() has an error in call of ethResources::serviceVerifyActivate() 
for BOARD EB5 IP 10.0.1.5
```

4. The device `embObjStrain` is not compiled. It is needed to manage the F/T sensors. See point 1 for how to solve.
```
[ERROR]Cannot find "embObjStrain" plugin (not built in, and no .ini file found for it)
```

5. The devices `embObjMais` and `canBusSkin` are not compiled. 
```
[ERROR]Cannot find "embObjMais" plugin (not built in, and no .ini file found for it)
[ERROR]Cannot find "canBusSkin" plugin (not built in, and no .ini file found for it)
```

What to do?
1. update 2FOC boards. It is not difficult if you follow the instructions on teh wiki and on read.me files in icub-firmware-build. However, **if it is your first update, maybe it is better to book a skype/netmeeting session w/ someone who can help you.**
2. compile missing devices.

pedrombmachado, 2017-03-09:
Hi @marcoaccame,
  Many thanks for the support. I would also like to thanks @julijenv  because his support was crucial. I made a short [video](https://www.dropbox.com/s/e5uqbwznz448uqn/icub.mp4?dl=0) that I would like to share. I hope you like the music.

pedrombmachado, 2017-03-23:
Hi @julijenv, @marcoaccame and @valegagge ,
   Many thanks for all the support, all of you were outstanding. The iCubNottingham01 is now up and running. I would like to leave special thanks to @julijenv because without your support it would be almost impossible!

closed 2017-03-23 by pedrombmachado