robotology/mesh-support #408: Wrong CAN boards of Skin Right Arm network after firmware upgrade

gsaponaro, 2017-02-22:
### Description of the failure

After a firmware upgrade, the CAN network 8 of our robot (Skin Right Arm) now shows this:

[image]
i.e. in the terminal:
```
[INFO]CONNECTED: 3 Boards
[DEBUG]  BUS:id   type     version
[DEBUG] CAN8:5       5     2.17.1
[DEBUG] CAN8:5       5     2.17.1
[DEBUG] CAN8:14       5     2.17.1
```

We notice two problems here:
1. repeated ID for two of the boards (we tried changing the ID of one of them, but it would change both);
2. some of the boards that were visible before are now disappeared.

The problems are consistent (after power-cycling motors and pc104).

We would like to bring the situation back to http://wiki.icub.org/wiki/Can_addresses_and_associated_firmware#CFW2CAN8:_Skin_Right_Arm_2 , any suggestion?

### Detailed conditions and logs of the failure

While debugging a problem with the skin, we decided to flash the firmware `skin.hex` on all the boards of CAN8 with the `canLoader` GUI, also using the 'Erase EEPROM' option. According to the messages during the download, everything went fine during the process. However, unfortunately the situation is now the one reported above.

yarp, icub-main, icub-firmware-shared are up to date (`master` branch) both on the pc104 and on the cluster. YARP version 2.3.68+194-20170221.1+gitaa2f78e

For reference, here is the correct configuration of the left arm skin (CAN7):

[image]

Thanks.

/cc @vicentepedro

julijenv, 2017-02-27:
Hi @gsaponaro,
Please try to change the id of the 2 boards 5 that u can see now into id 1 and 2. Update firmware for the new board id 1 and 2.
Then try to find which board are missing from the list 8to14. Check the vizualisation of skin and find the skin part not responding.
Afterwards change one board at a time the id 1 into one of the missing id and check if the visualization of the skin part come back. by deduction uyou should be able to rename correctly the boards htat have their id meesed up.
let me know

gsaponaro, 2017-02-27:
Thanks @julijenv , a couple of questions:
1) when we have two boards with the same ID and we try to change one, the other duplicate also changes, so we go from "5 5" to "1 1" etc. How to change just one? Perhaps we have an electrical problem in this skin (see next question);
2) do you have a diagram with the location of the boards and electrical cables for the right arm skin v2? We will check the continuity, as the `canLoader` is giving us inconsistent results after we power-cycle. For example, sometimes we see "duplicates" (5 5), other times we don't.

julijenv, 2017-02-28:
Hi @gsaponaro ,
 I guess that one precisely give you trouble or two and affect the possibility t see the other ones.
You will have to open both the upper arm cover as well as the cover from the forearms and un
plug every connectors, afterwards you will repeatedly reattach one after the other the connector until you understand which board(s) give u trouble.
[UPPERARM_1_4_REV1.pdf](https://github.com/robotology/icub-support/files/806151/UPPERARM_1_4_REV1.pdf)
[FOREARM_1_3_REV0.pdf](https://github.com/robotology/icub-support/files/806181/FOREARM_1_3_REV0.pdf)
Let me know

gsaponaro, 2017-03-14:
Thanks and sorry for the late update.

Good news:
* in the upper arm, we solved the duplicate ID issue by unplugging and replugging the boards one by one, assigning the correct ID again;
* we had to reflash the firmware in recovery mode on two of the MTB boards of the upper arm, otherwise they would not turn on correctly (blue light), now this is OK;
* 2B13 (RIGHT LOWER UPPERARM), board ID 8: resoldered a broken wire, now this board appears;
* in the lower arm (2B7 HAND, board ID 14) we also resoldered some broken wires, now all fingertips and palm sensors work.

Before closing this issue, another question:
* can you send us the cabling schema of our HAND, too? We need to check some connections related to the 2B7 right HAND network (ID 14).

julijenv, 2017-05-24:
Hi @,
sorry I didnt see the last part I thought I had already sent the information needed previously.
For sending the correct document could you please take a picture of the knuckles part of the hand to understand which version is it ( the electronic part of course).

gsaponaro, 2017-05-24:
No problem!

As for the pictures you requested, please let us know if these are sufficient: https://github.com/robotology/icub-support/issues/344 (the first picture shows the knuckles)

julijenv, 2017-05-24:
Your palm is different from the one in the file but the way it is connected is the same and for the fingertip it is perfectly similar
[HAND_CABLING_SCHEMA_V1_2_tactilesensor.pdf](https://github.com/robotology/icub-support/files/1025859/HAND_CABLING_SCHEMA_V1_2_tactilesensor.pdf)

closed 2017-05-24 by julijenv