robotology/mesh-support #410: iCub Basic Image Processing question

accateo, 2017-02-25:
Hi,
I'm trying to follow the "Basic Image Processing" tutorial of iCub:

http://wiki.icub.org/brain/icub_basic_image_processing.html

I have connected the image source, but I don't understand where I should see these messages:

We got an image of size 320x240
  We got an image of size 320x240
  We got an image of size 320x240
  ...
in what terminal? 
Second question: why in the iCub_Sim terminal I have always this warning?

`[WARNING]the simulation is too slow to run in real-time, you should increase the timestep in ode_params.ini (current value: xx, suggested value: xx)`

I have also changed the timestep value in ode_params.ini file, but the warning remains and change its value.

Thanks

vtikha, 2017-02-25:
Hi @accateo, first of all, please use the [Q&A](https://github.com/robotology/QA/issues) for any questions you have and not the icub-support which is a repository used to signal failures/repairs or ask for hardware/software questions regarding the physical robots. I will close this and reopen it in the proper repo.

vtikha, 2017-02-25:
This issue was moved to robotology/QA#189

closed 2017-02-25 by vtikha