robotology/mesh-support #413: Malfunctioning DR2-03S2C-EX-CS board & unresponsive motors

alelais, 2017-03-01:
Hi,
we have several issues with our iCub robot head and would like to ask kindly for support:
1) We don't know which version of the iCub robot head we have exactly.
2) Our Dragonfly2 board (DR2-03S2C-EX-CS) responsible for the left camera is malfunctioning.
3) We cannot control the motors of the robot.

How we proceeded up to now:
1) For being able to rule out the possibility that the OS we flashed on the flash drive of the robot is not compatible with our hardware we would like to find out which robot version we have exactly. From http://wiki.icub.org/wiki/ICub_versions we couldn't deduce it. You suggested to send you a picture of our robot:
[image]
[image]
[image]
[image]

2) i) We executed the following procedure to test both cameras. Different terminals are marked by **tx:**
```
t1: icub@icubsrv:~$ yarpserver
t2: icub@icubsrv:~$ ssh icub@pc104
t2: icub@pc104:~$ yarpdev --from ~/.local/share/yarp/robots/iCubFrankfurt01/camera/dragonfly2_config_right.ini --name /icub/right_cam
t3: icub@icubsrv:~$ yarpview --name /viewer_right
t4: icub@icubsrv:~$ yarp connect /icub/right_cam /viewer_right
```
At this point the yarpviewer viewer_right is showing the icub's right camera's image properly. When we continue, following happens:
```
t5: icub@icubsrv:~$ ssh icub@pc104
t5: icub@pc104:~$ yarpdev --from ~/.local/share/yarp/robots/iCubFrankfurt01/camera/dragonfly2_config_left.ini --name /icub/left_cam
[DEBUG]Format = 1
[WARNING]--d <unit_number> parameter is deprecated, use --guid <64_bit_global_unique_identifier> instead
libdc1394 error: Generic failure: in dc1394_video_get_iso_speed (control.c, line 747): Could not get ISO data

libdc1394 error: Generic failure: in dc1394_video_get_mode (control.c, line 663): Could not get video format

libdc1394 error: Generic failure: in dc1394_video_get_framerate (control.c, line 634): Could not get video framerate

libdc1394 error: Generic failure: in dc1394_camera_reset (control.c, line 500): Could not reset the camera

[ERROR]-1
[ERROR]LINE: 291
DragonflyDeviceDriver2: can't open camera
yarpdev: ***ERROR*** driver <dragonfly2> was found but could not open
libdc1394 error: Generic failure: in dc1394_video_set_transmission (control.c, line 953): Could not stop ISO transmission

yarpdev: ***ERROR*** driver <grabber> was found but could not open
yarpdev: ***ERROR*** device not available.
Suggestions:
+ Do "yarpdev --list" to see list of supported devices.
+ Or append "--verbose" option to get more information.
```
We could not continue the procedure. From here on, we investigated the hardware and repeated at each trial the described procedure. The hardware parts we are referring to in the following are marked by letters in the pictures.

2 ii) We switched the left firewire cable (D) with the right one (E). This resulted in the same behavior and we concluded that both cables are working. We switched the cables back again.

2 iii) We connected the left dragonfly2 board's (A) firewire cable (D) with the other port (P2) of the CFW card (G) and connected the right dragonfly2 board's (F) firewire cable (E) with port (P1). This resulted in the same behavior and we concluded that both ports (P1) and (P2) of the CFW card are working properly.

2 iv) We disconnected the left camera's band cable (B) from the left dragonfly2 board (A) and disconnected the band cable of the right counterpart. Then we connected (B) to (F). We tested just the first half of the testing procedure. This resulted in a working camera image transmission of the left camera (C). We concluded that the left camera itself (C) and the left band cable (B) are working fine, but the left dragonfly2 board (A) is malfunctioning.

Another indicator that (A) is not properly operating is the fact, that after connecting everything back as it was initially, powering up the robot and booting the OS, the right dragonfly2 board's (F) LED is constantly shining, whereas the LED of (A) is constantly blinking.

3 i) We followed the procedure described in http://robotcub.svn.sourceforge.net/svnroot/robotcub/trunk/iCubPlatform/doc/manuals/RC_DIST_100_D_14_01_HEAD_TORSO_CALIBRATION.doc, but could not connect to the CAN network, regardless which protocol we chose and which network number we wanted to connect to.

3 ii) By investigation of the hardware we noticed, that at one cable labeled N0 (H), which is leading from the motor control cards to the CFW card (G), the black wires is loose and not connected to the connector (I). Is that supposed to be like that or should we solder the black wire of (H) to the connector (I)? We refrained from doing that, because we couldn't find a pin-out diagram of (G) to verify that the black wire needs to be connected there.

[image]

alelais, 2017-03-02:
Thanks,
FLIR told us that the DR2-03S2C-EX-CS board is not produced anymore. Which hardware are you using in the new iCub head instead? Can we also use an alternative to the DR2-03S2C-EX-CS in our robot head? Is it maybe possible to repair the DR2-03S2C-EX-CS board?

julijenv, 2017-03-03:
Hi @alelais ,
we are still using this board so far, as long as we possess some spare parts, but some robots already used some better resolution board from (ex Point grey) DR2-08S2C-EX-C. I went to the website FLIR as u did, and it is not written the availability neither for this product nor for  DR2-03S2C-EX-CS.
I'll try to get in touch with our purchase office to know if they can contact FLIR to understand if the cameras are available.
In the meanwhile, i saw that it is possible (but i dont know what are the requisites) to ask for a repair of this board from the website.
https://eu.ptgrey.com/
in support you can find : contact support and returns and RMA and maybe they can repair it, it costs nothing to ask.
Ill let u know asap

alelais, 2017-03-06:
Hi,
these are great news. As you are using the DR2-08S2C-EX-CS boards already in some robots I wanted to ask whether those worked by plug and play or whether you needed to modify some hardware/software to run it.

julijenv, 2017-03-07:
Hi @alelais ,
good news, the purchase office sent a quotation for DR2-03S2C-EX-CS and receive a positive feedback from FLIR, it is still possible to get this board so please consider get in contact with them for either try to make them repairing it as I was explaining you howto on my last reply or place an order to them for the board.
Let me know,

alelais, 2017-05-22:
Hi @julijenv,
sorry for the long delay, I was busy at our software side until now. We decided to order two new DR2-08S2C-EX-CS boards and replace the DR2-03S2C-EX-CS boards by them. We connected the old cameras to the new boards and it seems to work for both cameras. There is still a resolution issue left, which we will deal with later.
For the motor control part I will open a new issue ticket but I consider the other two questions of this ticket resolved, thank you.

closed 2017-05-22 by alelais