robotology/mesh-support #415: Some left and right legs accelerometers not working properly (output all at 0xFF)

nunoguedelha, 2017-03-06:
### Description of the failure
All accelerometers (inertialMTB boards) on the right and left legs are activated. The data that we can read on ports `/icub/left_leg/inertialMTB` and `/icub/right_leg/inertialMTB` is showing all `0xFF` for the sensors: **10b10, 11b2, 11b6, 11b7, 11b8**.

### Detailed conditions and logs of the failure
The accelerometers are calibrated in the robot configuration files. A short sample of the log is attached.
The sensor 10b10 displays 0xFF measurements, but a valid timestamp.

nunoguedelha, 2017-03-07:
Further analysis on the MTB boards functionality, and more precisely: is the skin attached to those boards working?

- 10b10 (normal incrementing timestamp, but measurements = 0xFF) --> lower left leg front cover skin OK
- 11b2 (normal incrementing timestamp, but measurements = 0xFF) --> upper right leg front cover skin OK
- 11b6,11b7 (timestamp and measurements = 0xFF) --> upper right leg back cover skin **KO**
- 11b8 (timestamp and measurements now OK) --> lower right leg front cover skin OK

marcoaccame, 2017-03-08:
Hi @nunoguedelha, I had a look at the log file. 

You are right: the time-stamps are valid and regularly spaced but some values are invalid (-1).

That means that the mtb board regularly transmits but it does not acquire correctly and sends [-1, -1, -1].

**I suggest that the  icub-support guys gave a close look at those mtb boards.**

Moreover, ... if you write that the mtb board 11b8  which had incorrect measures now is OK I would say some cabling (or power supply?) problem.

nunoguedelha, 2017-03-08:
Thanks @marcoaccame, @julijenv has looked into it:

- actually boards 11b6 and 11b7 were disconnected under the cover
- @julijenv updated the firmware on boards 11b2 and 10b10 but it didn't change the behaviour. He then replaced 11b2

I'm about to retest.

julijenv, 2017-03-08:
Hi @nunoguedelha ,
it has been replaced the MTB3 (IITCODE 984.E) S/N:140 by S/N:362 - ID11B2 on the robot.
On hold to replace the 10B10.

nunoguedelha, 2017-03-08:
Thanks @julijenv , now 11B2, 11B6 and 11B7 are working fine.
I'll be testing the torso and arms accelerometers this afternoon.

nunoguedelha, 2017-03-10:
@julijenv the board you've just replaced (`10B10`) is working fine.
Thanks.
For the record, torso MTB boards are working fine also. I'll open a new issue if I find any problems with the arms.
Closing this issue now.

closed 2017-03-10 by nunoguedelha

julijenv, 2017-03-10:
it has been replaced the MTB3 (IITCODE 984.E) S/N:134 by S/N:372 - ID10B10 on the robot.