robotology/mesh-support #417: left little fingers joint is broken

valegagge, 2017-03-07:


julijenv, 2017-03-07:
Hi @valegagge ,
as requested by @pattacini and he is right about it, I think the template for opening na issue on icub-support it isrequested to fill up some fields in order to get more details about the failure.
In this case, is it mechanical (the word BROKEN makes me fear this), if not please reformulate as generical UNFUNCTIONNAL.
Another question would be, did you find it like that or after the use of the robot you make it that way.
At the end, it is just a way to find out if people are taking care of robot or not, if they do send an issue when they discover it or leave it to someone else to create the issue for them.
thx for the comprehension,

valegagge, 2017-03-07:
When I try to calibrated it, It does't calibrate. 
During calibration, the ring finger instead of opens, it closes, while pinkie remains open.

I have removed it from calibration file because it was broken, we talked about it some time ago and this morning I wanted only to check if it was ok and re-enable it in calibration, since some times the mechanical work is done, but files are not updated.

julijenv, 2017-03-10:
hi @valegagge ,
the last 2 fingers are ok, the PID may have been wrong (opposite sign).
Please try to change the sign and let me know

valegagge, 2017-03-13:
I had already checked the sign of PID: it has the same sign oh other robots with same electronics.
Did you re calibrate the joint?

valegagge, 2017-03-23:
I re calibrated the joint. Now it is ok.

closed 2017-03-23 by valegagge