robotology/mesh-support #418: motor encoder of right index distal joint doesn't read

valegagge, 2017-03-07:


valegagge, 2017-03-07:
I try to read the motor encoder by yarpmotorgui and I read only 0.

julijenv, 2017-03-10:
Hi @valegagge , I doucle check and the values are now ok, the only thing that may happen to induce the closing of the finger and the retroaction of the misplacement of the finger so to speak. why this misplacement, it is possible that the finger tendon is a bit worn so if you ask the finger to eturn to zero to fast the tendon make some pain to roll around some pulley and and some time to achieve the requsested position and so the motor start to accelerate.

julijenv, 2017-03-28:
Hi @valegagge ,
can we close it now?

valegagge, 2017-04-20:
Hi @julijenv ,
we cannot close this issue because the problem persists. I checked all motor encoders of other fingers and they read correctly. So I think there is a electrical issue. can you verify? thank you!

valegagge, 2017-07-04:
This problem was due to a wrong configuration of motor encoder: in particular the parameter "position" of group "ENCODER2" had value "none" insted of "atmotor".

see [commit](https://github.com/robotology/robots-configuration/commit/2127b207ed9fc3f813ef79d0323ea1cafb38c7d8)

closed 2017-07-04 by valegagge