robotology/mesh-support #421: iCub head joints not starting properly

AlexAntn, 2017-03-09:
### Description of the failure

We have an issue with the icub head, where the eyes and neck don't start up properly. All head joints start "idle"

### Detailed conditions and logs of the failure

aditionally, on yarpmotorgui, we can run the first 2 joints (pitch and roll), but yaw, and all eye joints, remain "idle".

upon checking the head, no cables appeared damaged/out of place.

output of yarprobotinterface: 
[output.txt](https://github.com/robotology/icub-support/files/831964/output.txt)

julijenv, 2017-04-06:
Hi @AlexAntn ,

I'm so sorry, I thought I had already replied to you on this issue.
I couldnt really found any problem on your logfile related to your failure so
let's make a list of possible problems:
1. first check if you can move those 4 joint with your hands
2. After having launched the yarprobotinterface, use yarpmotorgui to seee if the encoder of the motors from yaw to eyes joint move or give values and change if ever you move those particular joint
3. Check if there isnt any damaged grey flat cable which get out of every single motor
The last 4 joints or controlled by one board located inside the head
4. check where go all those grey flat cable - in the MC4 board - itself powered by a MCP board
5. . The power line of the control board if the 8ways rainbow connector, could you swap the 2 rainbow connector (see picture below)
[image]

The last but not the least, I saw this on your logfile:
_[[01;34mINFO[00m]Skin on can bus  7  uses NEW configuration version!!! 
yarp: Port /diagnostics/skin/errors:o active at tcp://10.0.0.2:10060
[[01;34mINFO[00m]created device <canBusSkin>. See C++ class CanBusSkin for documentation.
[[01;33mWARNING[00m]CanBusSkin id list contains more than one entry -> devices will be merged._
could you check with canLoader on can net 7 (check also net 8 to compare) if you dont have a missing board on that line.
use this to identificate the missing number
CFW2CAN7: Skin Left Arm from [here](http://wiki.icub.org/wiki/Can_addresses_and_associated_firmware)

Let me know,

AlexAntn, 2017-04-06:
hello @julijenv , no worries, in the meantime we figured out the problem on the head (and I also forgot the issue was open....)

The problem was in a small white wire on the back of the head (on the big bundle of wires coming from the back of the iCub). it was making poor contact on the head, and when I was checking every wire, it came off.

We have soldered it back, and everything seems to be running fine now! :)

As for the skin board, we haven't used the skin yet so I had not noticed such a problem. I will check into it later and send you the logs of the canLoader.

AlexAntn, 2017-04-11:
Hello @julijenv ,

I finally got to check on the boards on canLoader. I checked on net 7 and net 8, they are both equal. They report boards from 8 to 14 (so, 7 boards). 

From what I saw on the wiki page, this is indeed correct, so I detect no problems.

I will save the log of yarprobotinterface next time we run the robot, and send it to you for a final verification, but otherwise I think everything is running fine! :)

Thanks for the help!

AlexAntn, 2017-04-12:
Hello again @julijenv ,

We have used the robot so far without any issues. I tested again the canLoader, detects net 7 fine with all boards 8-14. 

I am sending the output of yarprobotinterface, so you can check it: 
[output.txt](https://github.com/robotology/icub-support/files/916823/output.txt)

Thanks for the help!

Alex

julijenv, 2017-04-12:
Hi @AlexAntn ,
if you have from 8 to 14 on each line 7 and 8, all is good!!!!.
I suppose that the message is just not perfectly defined, hopefully
C U,

closed 2017-04-12 by julijenv