robotology/mesh-support #423: Facial expression module doesn't work

claudia-lat, 2017-03-13:
The facial expression module has got a problem in reading from the serial port.

julijenv, 2017-03-24:
Hi @claudia-lat ,
went there to check other issues, and find out that depending on the connection with the power line we can see on dmesg ¦ tail the main face expression board respond but unfortunately the facial expression light and motor movement dont do anything.
Ill try next time to plug the USB to one of the 2 other USB plug to test the compatibility

julijenv, 2017-03-28:
Hi @claudia-lat ,
everything seems alright for now, I close it, if ever neeeded let's reopen it

S-Dafarra, 2017-04-20:
The software modules seem to open correctly, but I can't see any light and the eyes don't blink.

S-Dafarra, 2017-06-06:
Facial expressions seem working. It was probably related to missing flags.

closed 2017-06-06 by S-Dafarra