robotology/mesh-support #428: MC: exernal fault button pressed for the head joints error at startup, even if no external fault button is pressed

traversaro, 2017-03-20:
### Description of the failure
On startup, there is an error for the first two joints of the head relative to an "external fault button pressed", even if the fault button is not pressed at all. Interestingly, the joint seem to work fine when operated through the yarpmotorgui . 

### Detailed conditions and logs of the failure
~~~
3260	182,928161	ERROR	 from BOARD 10.0.1.20 (head-j0_1), src LOCAL, adr 0, time 179s 458m 717u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
3261	182,928253	ERROR	 from BOARD 10.0.1.20 (head-j0_1), src LOCAL, adr 0, time 179s 458m 807u: (code 0x02000000, par16 0x0001 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
3347	187,932562	ERROR	 from BOARD 10.0.1.20 (head-j0_1), src LOCAL, adr 0, time 184s 459m 716u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
3348	187,932595	ERROR	 from BOARD 10.0.1.20 (head-j0_1), src LOCAL, adr 0, time 184s 459m 807u: (code 0x02000000, par16 0x0001 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
~~~

ale-git, 2017-03-20:
Hi @traversaro, does the external fault persist after startup or it vanishes? In the second case it should be a problem of false contact in the wiring of the external fault.

traversaro, 2017-03-20:
It vanishes after those first two messages.

julijenv, 2017-03-20:
hi @ale-git, about this failure, would it be possible to correct the mispelling of exTernal once for all in the future release of the FW , thx in advance

ale-git, 2017-03-20:
Hi @julijenv, why "mispelling"? It signals that the external button is pressed, but there is no way to discriminate if the button is really pressed or if the external fault cable is detached.

francesco-romano, 2017-03-20:
There is a missing 't' in the error message
ex**t**ernal

ale-git, 2017-03-20:
Ah thanks! LOL

valegagge, 2017-03-21:
This issue has been fixed in master branch. In the future days i'm going to merge it in devel branch.

This issue is the same of https://github.com/robotology/icub-firmware/issues/79

traversaro, 2017-03-21:
Great!

valegagge, 2017-03-23:
I merged the patch in devel branch and I released new firmware on devel branch. 
MC4PLUS and MC2PLUS ver 3.1.

closed 2017-03-23 by valegagge