robotology/mesh-support #430: Left and right hand skins not working

claudiofantacci, 2017-03-23:
Left and right hand skins are unresponsive.

You can test them using the Skin_GUI_All applications.

julijenv, 2017-03-24:
Hi @claudiofantacci ,
To me nothing to declare for the left arm, everything looks alright,
for the right arm instead we found one MTB board with a different FW and not responding, and also 1 fingertip which gave some issue and on which I will tackle the issue soon. For the time being I just remove the switch board to prevent the fingertip to disturb the other patch of the hand skin.

claudiofantacci, 2017-03-24:
Thanks @julijenv ,
yesterday both hand skins were not working.
Did you tested the left hand skin?

julijenv, 2017-03-24:
I did either the forearm , palm and fingertip for the left arm, was one part of those not functionning during your test, yesterday?

claudiofantacci, 2017-03-24:
Yes, the left hand skins (including fingertips) were not working.

I just tested them again and they are working properly, but the middle fingertip.

julijenv, 2017-03-24:
you mean but the middle fingertip of the RIGHT hand I presume???
the middle left was working this morning?!?!

claudiofantacci, 2017-03-24:
Sorry Julien, my bad.

I'll start from the beginning 🙃 

The LEFT hand is working just fine.
The RIGHT hand is working fine, but for the middle finger.

julijenv, 2017-03-24:
i know i ll take care of it ;P

julijenv, 2017-03-27:
been changed 
IITCODe Qty Description
1466 x 1 : iCub, subassembly - Fingertips for right hand, middle finger, address 1, with lycra

closed 2017-03-27 by julijenv