robotology/mesh-support #432: left index finger trembling on grasp

tanismar, 2017-03-23:
Please **fill in the following two sections** and then **remove any comment and/or remark become unnecessary**:

### Description of the failure
I have noticed during the past days, that when doing a pinch grasp on the plush objects, the index finger of the left hand trembles. Its not a big deal, and indeed it does not prevent the grasp, but it looks weird, and it might be the symptom of a bigger problem

### Detailed conditions and logs of the failure
Have noticed it while testing the tool use demo during the last days.

---
**Remark**: if you are not a daily icub-support user, please read carefully the guidelines for contributing linked at the top of this page. In particular, the **name of your robot** must be always indicated: either using **labels** if you're already a contributor, as per standard, or by specifying it while detailing the issue.

julijenv, 2017-03-24:
Hi @tanismar ,
I didnt notice any real shaky movement coming from the left index finger, but more from the middle one, R u sure??? Can u let me see in which context, it happens?
thx in advance

julijenv, 2017-03-31:
it has been retensed all the proximla joints of the left and right hand.

closed 2017-03-31 by julijenv