robotology/mesh-support #433: Head - Eyes flat cables broken

amaroyo, 2017-03-24:
Hi @julijenv, 

The two flat cables which are connected to the eyes (left and right) are broken (see picture below). Could we have a new set for next Monday (27th) that I will pass by Morego to pick them up?

Thanks, 
Alex.
[image]

julijenv, 2017-03-27:
Hi @amaroyo  and @reafrancesco ,
both are broken????? quite unusual !?! anyhow, just pass and we will provide, just keep in moind that those ribbo flat cable are fragile, and HATE BEING BENT to 180 degrees so be aware of that when u will replace them. try just to copy the flat already in place to make sure that the trace that it will make wont damage it.

julijenv, 2017-03-27:
it has been given to @amaroyo ,
ITTCODE Qty Desciption
356 x 2 : ribbon flat cable 20vie

closed 2017-03-27 by julijenv