robotology/mesh-support #437: Replacing parts for the iCub fingers plastic protections

pedrombmachado, 2017-04-02:
### Description of the failure
Need replacing plastic parts for the iCub fingers.

### Detailed conditions and logs of the failure
This is not a critical problem because the there is none physical problem. Two of the plastic finger protections needs to be replaced. The wires are intact and there was no other damage.
Many thanks

julijenv, 2017-04-03:
Hi @pedrombmachado,
pictures please?

pedrombmachado, 2017-04-11:
Hi @julijenv ,
  Sorry for my late reply. I am on my annual leave and therefore I cannot upload the photos until the 19/04/2017.

julijenv, 2017-04-12:
HI @pedrombmachado ,
dont worry, have fun in holidays. I'm not in a hurry.

pedrombmachado, 2017-04-19:
Hi @julijenv ,
   I am back from my annual leave. Please find the photos below:
[image]
[image]
[image]
[image]
[image]

pedrombmachado, 2017-04-19:
Hi @julijenv ,
   Thanks for your help. The following tools and parts are required:
1) Cable Sleeve Expander [link](http://uk.rs-online.com/web/p/cable-sleeving-tools/3212436/)
2)  Silicone Flexible Tubing [link](http://uk.rs-online.com/web/p/flexible-tubes/4190062/?searchTerm=4190062&relevancy-data=636F3D3126696E3D4931384E525353746F636B4E756D6265724D504E266C753D656E266D6D3D6D61746368616C6C26706D3D5E5C647B367D247C5E5C647B377D247C5E5C647B31307D2426706F3D313426736E3D592673743D52535F53544F434B5F4E554D4245522677633D4E4F4E45267573743D34313930303632267374613D3431393030363226)

closed 2017-04-19 by pedrombmachado