robotology/mesh-support #438: Yarprobotinterface segfault after firmware update

dcam0050, 2017-04-04:
### Description of the failure

I have updated `yarp`, `icub-main`, `icub-firmware-shared` and `icub-firmware-build` to their latest version on pc104. Then I updated the firmware for all ETH and all the CAN boards,  

Then after I power cycled the motors and pc104 I ran yarprobotinterface and I got a Segmentation fault error late in the booting process.

### Detailed conditions and logs of the failure
I've attached here the log for yarprobotinterface.

[yarprobotinterface_log.txt](https://github.com/robotology/icub-support/files/893780/yarprobotinterface_log.txt)

I've tried compiling yarp with libmath using eigen3 and libmath using gsl to check if it makes a difference but it still fails in the same spot.

Any help would be greatly appreciated! Thanks

pattacini, 2017-04-04:
Hi @dcam0050, as per our [guidelines](https://github.com/robotology/icub-support/blob/master/.github/CONTRIBUTING.md), this repo is only for tracking requests for robot maintenance. Thus, I'll move this issue to yarp.

pattacini, 2017-04-04:
Issue moved to [robotology/yarp #1165](https://github.com/robotology/yarp/issues/1165) via [**ZenHub**](https://www.zenhub.com/)

closed 2017-04-04 by pattacini