robotology/mesh-support #442: Broken head tendon

fvannu, 2017-04-13:
### Description of the failure

During the calibration at the startup of the robot, one of the tendons in the head broke.
Particularly is Cable A as indicated in the Head_V2_cabling.pdf

[image]

julijenv, 2017-04-14:
Hi @fvannu , @reafrancesco ,
the first question is : Do u have spare parts for replacing it??
If not, Please head up to IIT Main Headquarter to get cable A.

fvannu, 2017-04-18:
We don't have any, so we will get one as soon as possible.

closed 2017-04-21 by amaroyo

julijenv, 2017-04-21:
it has been restored,
IITCODE Qta DEsceiption
5965 x 1 : head neck cable A