robotology/mesh-support #444: missing eobrd_mais board

pi-q, 2017-04-18:
When trying to run yarprobotinterface on iCubGenova02, it fails with the following error showing up in yarplogger:

586	13,250663	ERROR	embObjMais::open() has an error in call of ethResources::serviceVerifyActivate() for BOARD EB2 IP 10.0.1.2
652	14,020145	ERROR	FAILED writing  PAGE_ID
585	12,248711	ERROR	1 of 1: missing eobrd_mais BOARD 10.0.1.2:CAN1:14
584	12,248704	ERROR	 from BOARD 10.0.1.2 (EB2) @ 446s 758m 560u: CAN discovery after 2999 ms has detected 1 missing eobrd_mais boards in CAN1:
583	12,248695	ERROR	 from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 446s 758m 414u: (code 0x0500000b, par16 0x000e par64 0x0000000001000000) -> CFG: EOtheMAIS cannot be configured. can discovery fails. can address in p16, prot and vers in p64 lower 32 bits + .
582	12,248683	ERROR	 from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 446s 758m 300u: (code 0x0500000b, par16 0x000e par64 0x0000000001000000) -> CFG: EOtheMAIS cannot be configured. can discovery fails. can address in p16, prot and vers in p64 lower 32 bits + .
581	12,248673	ERROR	1 of 1: missing eobrd_mais BOARD 10.0.1.2:CAN1:14
580	12,248635	ERROR	 from BOARD 10.0.1.2 (EB2) @ 446s 758m 185u: CAN discovery after 2999 ms has detected 1 missing eobrd_mais boards in CAN1:
342	1,901606	ERROR	embObjMotionControl::open() fails to start mc service for board EB2 IP 10.0.1.2 : cannot continue
851	17,257495	ERROR	One or more devices failed opening... see previous log messages for more info

In effect, the red light in the left hand also fails to turn on. Is there any known way to deal with this problem?

DanielePucci, 2017-04-18:
@spada1978 @julijenv

iron76, 2017-04-19:
@spada1978 @julijenv tomorrow (11am) there will be a demo and therefore we need to know if the robot will be functional

spada1978, 2017-04-19:
Now the robot it's ok.

ale

closed 2017-04-21 by julijenv