robotology/mesh-support #445: driver <canmotioncontrol> was found but could not open

mpatacchiola, 2017-04-18:
### Description of the failure

We had this problem without any apparent reason on iCubPlymouth01. After shutting down the robot we could not manage to start again the yarprobotinterface module from the manager. We tried to lunch it directly from the pc104 internal board and we saved the log.

### Detailed conditions and logs of the failure
The main error seems to be:

`[ERROR]checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
`

It does not make sense to me, we did not touch the firmaware and we did not update any component. I will attach the full log output. Do you have any idea?

Massimiliano

[log.txt](https://github.com/robotology/icub-support/files/930123/log.txt)

mpatacchiola, 2017-04-18:
Ok I carefully read the log and I probably found the problem, but I do not know how to solve it. I think the problem is in the control board of the left hand. I deduced this from the lines before the error in the log cascade. In `sharedcan [5]` joints: 0, 1, 2, 3 the version of the board and the `CAN_protocol` seems to be corrupted. Here I will report one of this line:

`
[INFO]sharedcan [5] joint: 0 can_address:  5 board type: 3 (4DC) version:ffffffff.ffffffff build: -1 CAN_protocol:-1.-1
`

I noticed that this line is different from all the other which report a valid number of version and protocol, like this:

`[INFO]sharedcan [5] joint: 4 can_address:  7 board type: 3 (4DC) version: 2.30 build: 87 CAN_protocol:1.6`

At this point in the log the error message that I showed in the previous post appears and on the bottom there is a message about the left hand motor controller:

`[WARNING]Cannot open device left_hand_mc `

My hypothesis is that one or more board is damaged. Do you have any insight? There is a way to disable the left arm (we are not using it in this moment and we need the robot for an important demo).

Any suggestion is appreciated, Massimiliano

julijenv, 2017-04-19:
Hi @mpatacchiola ,
sorry to answer one day later than I should, I was quite busy yesterday, my bad.
so you are absolutely right , you problem is on tjhe Line 5, but the line 

> [INFO]sharedcan [5] joint: 0 can_address:  5 board type: 3 (4DC) version:ffffffff.ffffffff build: -1 CAN_protocol:-1.-1

is not you solely problem there is also

> [DEBUG]sharedcan [5] msg from board 5: ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ 

So here what you're gonna do
Turn OFF and On the motor switch
open on a shell window, the software canLoader
select NET6 - cfw2can and click CONNECT
make a printscreen, this will help you to debug NET5 later on, because those 2 net should be identical.
So now DISCONNECT, and select NET5 and click CONNECT.
from the printscreen of net6 make a comparaison.
Chances are that you can see the 3 boards (2 board 4DC and 1 MAIS) - Do u infirm???
The only problem should be that the firmware version of 4DC board ID5 is somehow corrupted.
So select the board ID5 (tick the left case- next to ID5) click on open folder on the right and go into the repository icub-firmware-build that you should have downloaded normally - make sure to update it before proceeding. Now select the firmware with the same version that you can read for the board ID5 for net6. Once selected, download it and that should be it.
I fever you dont understand my explainations, please send me some pictures of the net 5 and net6, find me on skype , my ID is icub.support, and make sure you have teamviewer on your server side to give you an hand.
Let me know,
good luck

mpatacchiola, 2017-04-19:
Dear @julijenv this morning I uploaded the new firmware using the canLoader. Following your suggestion I was able to fix the issue. During the day I tested the robot and it seems there are no problems now.

Thanks for your support, 

Massimiliano

closed 2017-04-19 by mpatacchiola