robotology/mesh-support #446: Cameras/eyes cover do not open on the new iCub face

pedrombmachado, 2017-04-19:
Please **fill in the following two sections** and then **remove any comment and/or remark become unnecessary**:

### Description of the failure
Cameras/eyes cover do not open on the new iCub face

### Detailed conditions and logs of the failure
The iCub eyes cover do not open after replacing the iCub face. This only happens with the new version, because the situation is solved when I install the old iCub face.

[image]
[image]
[image]

julijenv, 2017-04-19:
Hi @pedrombmachado ,
Could you try tpo launch some command of facial expression to see if the eyes reopen???
thx in advance or else C U on skype

pedrombmachado, 2017-04-19:
Hi @julijenv ,
  Sorted with your support.

closed 2017-04-19 by pedrombmachado