robotology/mesh-support #448: The motors have to be rebooted before relaunching yarprobotinterface

S-Dafarra, 2017-04-20:
If the yarprobotinterface is launched, then closed and launched again without rebooting the motors it (usually) fails to startup. For example in the Green I got the following errors:
```
116972	2117,226082	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD head-j0_1 with IP 10.0.1.20  even after  2  attempts and 0.101272 seconds: CANNOT PROCEED ANY FURTHER
116994	2118,579934	ERROR	hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.20. Expected: 559764, Received: 559765, Missing: 1, Prev Frame TX at 5458629426 us, This Frame TX at 5459165964 us
116998	2118,680616	ERROR	  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD head-j0_1 with IP 10.0.1.20 : cannot proceed any further
117000	2118,680628	ERROR	embObjMotionControl: failed verifyEPprotocol. Cannot continue!
117003	2119,426841	ERROR	hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.20. Expected: 559766, Received: 559768, Missing: 2, Prev Frame TX at 5459165964 us, This Frame TX at 5460016136 us
117037	2119,872045	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD head-j2_5 with IP 10.0.1.21  even after  2  attempts and 0.10099 seconds: CANNOT PROCEED ANY FURTHER
117055	2121,662239	ERROR	hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.21. Expected: 563145, Received: 563146, Missing: 1, Prev Frame TX at 5461304418 us, This Frame TX at 5462282448 us
117060	2121,763236	ERROR	  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD head-j2_5 with IP 10.0.1.21 : cannot proceed any further
117062	2121,763249	ERROR	embObjMotionControl: failed verifyEPprotocol. Cannot continue!
117066	2122,511922	ERROR	hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.21. Expected: 563147, Received: 563149, Missing: 2, Prev Frame TX at 5462282448 us, This Frame TX at 5463141960 us
117106	2122,967014	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD EB5 with IP 10.0.1.5  even after  2  attempts and 0.101026 seconds: CANNOT PROCEED ANY FURTHER
117123	2124,273310	ERROR	  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD EB5 with IP 10.0.1.5 : cannot proceed any further
117125	2124,273367	ERROR	embObjMotionControl: failed verifyEPprotocol. Cannot continue!
117129	2124,932319	ERROR	hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.5. Expected: 558208, Received: 558209, Missing: 1, Prev Frame TX at 5464678523 us, This Frame TX at 5465447823 us
117131	2124,932376	ERROR	eoprot_fun_UPDT_mn_comm_cmmnds_command_replyarray() has received an unexpected message
117198	2125,499820	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD leftArm-j0_3 with IP 10.0.1.1  even after  2  attempts and 0.101097 seconds: CANNOT PROCEED ANY FURTHER
117266	2126,725216	ERROR	  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD leftArm-j0_3 with IP 10.0.1.1 : cannot proceed any further
117268	2126,725273	ERROR	embObjMotionControl: failed verifyEPprotocol. Cannot continue!
117272	2127,483328	ERROR	hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.1. Expected: 558556, Received: 558558, Missing: 2, Prev Frame TX at 5467133770 us, This Frame TX at 5468009850 us
117344	2127,907988	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD leftArm-j4_7 with IP 10.0.1.24  even after  2  attempts and 0.100863 seconds: CANNOT PROCEED ANY FURTHER
117374	2128,019104	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD leftArm-j4_7 with IP 10.0.1.24  even after  2  attempts and 0.10092 seconds: CANNOT PROCEED ANY FURTHER
117500	2129,336825	ERROR	hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.24. Expected: 930559, Received: 930560, Missing: 1, Prev Frame TX at 5469413016 us, This Frame TX at 5469918537 us
117503	2129,438046	ERROR	  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD leftArm-j4_7 with IP 10.0.1.24 : cannot proceed any further
117506	2129,438112	ERROR	embObjMotionControl: failed verifyEPprotocol. Cannot continue!
117510	2130,097361	ERROR	hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.24. Expected: 930561, Received: 930562, Missing: 1, Prev Frame TX at 5469918537 us, This Frame TX at 5470694274 us
117512	2130,107488	ERROR	eoprot_fun_UPDT_mn_comm_cmmnds_command_replyarray() has received an unexpected message
117559	2132,190011	ERROR	  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD leftArm-j8_11 with IP 10.0.1.25 : cannot proceed any further
117562	2132,190062	ERROR	embObjMotionControl: failed verifyEPprotocol. Cannot continue!
117566	2132,958921	ERROR	hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.25. Expected: 561736, Received: 561738, Missing: 2, Prev Frame TX at 5472958995 us, This Frame TX at 5473564905 us
117658	2134,002093	ERROR	  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD leftArm-j12_15 with IP 10.0.1.26 : cannot proceed any further
117660	2134,002131	ERROR	embObjMotionControl: failed verifyEPprotocol. Cannot continue!
117665	2134,659811	ERROR	eoprot_fun_UPDT_mn_comm_cmmnds_command_replyarray() has received an unexpected message
117763	2135,750618	ERROR	  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD rightArm-j0_3 with IP 10.0.1.3 : cannot proceed any further
117766	2135,750668	ERROR	embObjMotionControl: failed verifyEPprotocol. Cannot continue!
117770	2136,205928	ERROR	eoprot_fun_UPDT_mn_comm_cmmnds_command_replyarray() has received an unexpected message
117876	2137,187090	ERROR	  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD rightArm-j4_7 with IP 10.0.1.27 : cannot proceed any further
117878	2137,187106	ERROR	embObjMotionControl: failed verifyEPprotocol. Cannot continue!
117880	2137,399261	ERROR	eoprot_fun_UPDT_mn_comm_cmmnds_command_replyarray() has received an unexpected message
117912	2138,167681	ERROR	  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD rightArm-j8_11 with IP 10.0.1.28 : cannot proceed any further
117915	2138,167732	ERROR	embObjMotionControl: failed verifyEPprotocol. Cannot continue!
117917	2138,208112	ERROR	eoprot_fun_UPDT_mn_comm_cmmnds_command_replyarray() has received an unexpected message
118012	2139,248444	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD rightArm-j12_15 with IP 10.0.1.29  even after  2  attempts and 0.101271 seconds: CANNOT PROCEED ANY FURTHER
118028	2139,449814	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD rightArm-j12_15 with IP 10.0.1.29  even after  2  attempts and 0.101098 seconds: CANNOT PROCEED ANY FURTHER
118040	2139,560568	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD rightArm-j12_15 with IP 10.0.1.29  even after  2  attempts and 0.100832 seconds: CANNOT PROCEED ANY FURTHER
118052	2139,711683	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD rightArm-j12_15 with IP 10.0.1.29  even after  2  attempts and 0.101183 seconds: CANNOT PROCEED ANY FURTHER
118063	2139,833003	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD rightArm-j12_15 with IP 10.0.1.29  even after  2  attempts and 0.101275 seconds: CANNOT PROCEED ANY FURTHER
118074	2139,944154	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD rightArm-j12_15 with IP 10.0.1.29  even after  2  attempts and 0.101293 seconds: CANNOT PROCEED ANY FURTHER
118091	2140,125748	ERROR	  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD rightArm-j12_15 with IP 10.0.1.29  even after  2  attempts and 0.101061 seconds: CANNOT PROCEED ANY FURTHER
[... cut ...]
119210	2146,404215	ERROR	         EthResource::verifyBoardTransceiver() asks: can you ping the board? if so, is the MN protocol version of BOARD equal to ( 2 10 )? if not, perform FW upgrade. if so, was the ropframe transmitted in time?
119214	2146,404276	ERROR	EthResource::verifyEPprotocol() cannot verify BOARD EB6 with IP 10.0.1.6 : cannot proceed any further
121068	2155,257519	ERROR	One or more devices failed opening... see previous log messages for more info
```

By rebooting motors, everything works fine.
@DanielePucci @traversaro  @francesco-romano  @gabrielenava  @valegagge @marcoaccame

randaz81, 2017-04-20:
Did the yarprobotinterface close successfully or with SEGFAULT?  :smirk:
In this latter case, you cannot expect the robot to restart smoothly. we need to fix this issue first...

S-Dafarra, 2017-04-20:
I should have seen [this](https://github.com/robotology/yarp/issues/1168) issue before :grin:.

S-Dafarra, 2017-04-20:
Indeed it seems that by running the yarprobotinterface through terminal this problem does not occur. I mean, the SEGFAULT happens every time, but I don't need to reboot the motors.

marcoaccame, 2017-06-01:
Hi @S-Dafarra and @randaz81, I believe this ussue can be closed. 
See https://github.com/robotology/icub-main/issues/452 where I wrote:
```
See the bugfix in https://github.com/robotology/icub-main/issues/452. 
Now after a crash of yarprobotinterface the ETH boards do no need to be restarted again.
```

marcoaccame, 2017-06-05:
Hi, 

I had a look also at the systematic crash of yarprobotinterface when it is stopped with a CTRL-C. 
Analysis w/ gdb has show that the responsible was the cartesian controller.

The crash with a single CTRL-C does not show anymore after changes in https://github.com/robotology/icub-main/commit/cd31d37e540f8d9d7720b2d8765db656e397e2f1.

However, I have seen crashes with multiple CTRL-C in quick succession. Maybe there are other unsolved bugs. But they are not relevant to this issue, hence I close it.

closed 2017-06-05 by marcoaccame