robotology/mesh-support #451: yarprobotinterface is not starting due to fullscalePWM

amaroyo, 2017-04-21:
Hello, 

It's written in the error of yarprobotinterface to update the robots-configuration or to contact icub-support. The first thing didn't work :)

error:
```
[ERROR]fullscalePWM not found
[ERROR]fullscalePWM param not found in config file. Please update robot configuration files or contact https://github.com/robotology/icub-support
[ERROR]One or more errors found while parsing device configuration 
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
[WARNING]Cannot open device left_arm_mc 
[WARNING]Cannot open device left_arm_mc 
[ERROR]fullscalePWM not found
[ERROR]fullscalePWM param not found in config file. Please update robot configuration files or contact https://github.com/robotology/icub-support
[ERROR]One or more errors found while parsing device configuration 
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
[WARNING]Cannot open device left_hand_mc 
[WARNING]Cannot open device left_hand_mc 
[ERROR]fullscalePWM not found
[ERROR]fullscalePWM param not found in config file. Please update robot configuration files or contact https://github.com/robotology/icub-support
[ERROR]One or more errors found while parsing device configuration 
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
[WARNING]Cannot open device right_arm_mc 
[WARNING]Cannot open device right_arm_mc 
[ERROR]fullscalePWM not found
[ERROR]fullscalePWM param not found in config file. Please update robot configuration files or contact https://github.com/robotology/icub-support
[ERROR]One or more errors found while parsing device configuration 
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
[WARNING]Cannot open device right_hand_mc 
[WARNING]Cannot open device right_hand_mc 
[ERROR]fullscalePWM not found
[ERROR]fullscalePWM param not found in config file. Please update robot configuration files or contact https://github.com/robotology/icub-support
[ERROR]One or more errors found while parsing device configuration 
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
[WARNING]Cannot open device head_mc 
[WARNING]Cannot open device head_mc 
[ERROR]fullscalePWM not found
[ERROR]fullscalePWM param not found in config file. Please update robot configuration files or contact https://github.com/robotology/icub-support
[ERROR]One or more errors found while parsing device configuration 
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
[WARNING]Cannot open device torso_mc 
[WARNING]Cannot open device torso_mc 
```

P.S.: I checked all the boards with the canLoader and they are present. I suppose the error of canmotioncontrol is because of the error of PWM.

Thanks, 
Alex.

julijenv, 2017-04-21:
please @amaroyo  add the full logfile of the session, i guess the error is previously mentionnes BUT not inside your few lines of log,
thx in advance

amaroyo, 2017-04-21:
@julijenv, sorry, here you are:

```
icub@pc104:~$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/yarprobotinterface.ini] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubGenova03/yarprobotinterface.ini] (robot)
||| found /home/icub/.local/share/yarp/robots/iCubGenova03/yarprobotinterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubGenova03/./icub_all.xml] (robot)
||| found /home/icub/.local/share/yarp/robots/iCubGenova03/./icub_all.xml
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./icub_all.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./icub_all.xml at line 2 . "robot" element should contain the "build" attribute [unsigned int]. Assuming 0 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./icub_all.xml at line 2 . "robot" element should contain the "portprefix" attribute. Using "name" attribute 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml at line 2 . Trying to import a file for the wrong robot. Found  instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml at line 2 . Trying to import a file for the wrong robot. Found  instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/head_mc_wrapper.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/head_mc_wrapper.xml at line 2 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/head_mc_wrapper.xml at line 2 . Trying to import a file for the wrong robot. Found  instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/head_mc_wrapper.xml at line 2 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/torso_mc_wrapper.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/torso_mc_wrapper.xml at line 2 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/torso_mc_wrapper.xml at line 2 . Trying to import a file for the wrong robot. Found  instead of icub 
[... cut ...]
[INFO]created device <parametricCalibrator>. See C++ class yarp::dev::parametricCalibrator for documentation.
[INFO]created device <parametricCalibrator>. See C++ class yarp::dev::parametricCalibrator for documentation.
[WARNING]There was some problem opening one or more devices. Please check the log and your configuration 
[ERROR]One or more devices failed opening... see previous log messages for more info 
[INFO]Closing Server Inertial...
[INFO]Server Intertial thread finished
Segmentation fault
icub@pc104:~$ 
```

traversaro, 2017-04-21:
Hi @amaroyo , 
  from the error message it seems that you are using the `devel` branch of `yarp`, `icub-main`, etc etc (see https://github.com/robotology/robots-configuration/pull/25). 
At the moment, the only robots for which the configuration files are maintained to work on devel are `iCubGenova01` , `iCubGenova02` and `iCubGenova04`. 
The actual policy regarding the `devel`/`master` branches and robots configuration is going to be decide soon (see  https://github.com/robotology/robots-configuration/issues/26). 

I think that everything should work fine if you use the `master` branch of all the repos.

amaroyo, 2017-04-21:
Hi @traversaro, 

We have been in the branch devel since quite some months ago, as @randaz81 asked us to be. 
It has been working well till yesterday when I updated the setup, I have seen that you have merged [Pwm and current control](https://github.com/robotology/yarp/pull/1145) and I thought that could be the problem.

traversaro, 2017-04-21:
Ok, I did not know you were supposed to be on the `devel` branch! 
iCubGenova03 is a CAN robot (if I remember correctly) so the modification necessary should be similar to the one done in https://github.com/robotology/robots-configuration/pull/25 for `iCubGenova01`.
However **these parameters are strictly related to quite low level behaviour of the robot and touching them could result in permanent damage of the robot** , so waiting for the expert @randaz81 and @davidetome should be the best strategy.

amaroyo, 2017-04-23:
Thanks @traversaro!

So, @randaz81, @davidetome should I replicate the parameters suggested in [robotology/robots-configuration#25](https://github.com/robotology/robots-configuration/pull/25#issue-220594582) in the same way you did in the commit [**```c7ba256```**](https://github.com/robotology/robots-configuration/commit/c7ba256fc36c0cb6912c6d7b81fc9223d1d3ed75) for ```iCubGenova01```, i.e., the same addition of new parameters in all the files?

randaz81, 2017-04-23:
Yes, you can use in `iCubGenova03` the same values copied from `iCubGenova01`, devel branch. Sorry for forgetting this robot during the files update! It's super easy to do the diff, but please tell me if you have problems, I can do it for you. Otherwise, if everything is ok, please open a PR on the devel branch of robot-configuration. Thank you!

randaz81, 2017-04-23:
Ah! the same modifications apply to ikart configuration! I'll take care of them.

randaz81, 2017-04-24:
ikart fixed by https://github.com/robotology/ikart/commit/a44b3fc8a34424dd307129938582522679085743 on `devel `branch

amaroyo, 2017-04-24:
@randaz81, thanks for the iKart! I may contact you for doing the proper changes for the iCub, just in case. Maybe we can skype so to see how to properly update the ikart as well! This Wednesday will be ok for you?

cognitiveinteraction, 2017-04-26:
Hello!

I did the changes, and it does the calibration ok. However, I have the following errors in the terminal (see below full trace). They keep appearing without stopping.

``` 
[ERROR]run()-pushReadMsg() failed on CAN bus 1 
[ERROR]recv buffer overrun ( 500 >  500) 
[ERROR]getDutyCycleRaw not yet implemented for CanBusMotionControl
[ERROR]getDutyCycleRaw not yet implemented for CanBusMotionControl
[ERROR]getDutyCycleRaw not yet implemented for CanBusMotionControl
``` 

```
icub@pc104:~$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/yarprobotinterface.ini] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubGenova03/yarprobotinterface.ini] (robot)
||| found /home/icub/.local/share/yarp/robots/iCubGenova03/yarprobotinterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubGenova03/./icub_all.xml] (robot)
||| found /home/icub/.local/share/yarp/robots/iCubGenova03/./icub_all.xml
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./icub_all.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./icub_all.xml at line 2 . "robot" element should contain the "build" attribute [unsigned int]. Assuming 0 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./icub_all.xml at line 2 . "robot" element should contain the "portprefix" attribute. Using "name" attribute 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml at line 2 . Trying to import a file for the wrong robot. Found  instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml at line 2 . Trying to import a file for the wrong robot. Found  instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/head_mc_wrapper.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/head_mc_wrapper.xml at line 2 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/head_mc_wrapper.xml at line 2 . Trying to import a file for the wrong robot. Found  instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/head_mc_wrapper.xml at line 2 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubGenova03/./wrappers/motorControl/torso_mc_wrapper.xml 
[... cut ...]
[ERROR]getDutyCycleRaw not yet implemented for CanBusMotionControl
[ERROR]getDutyCycleRaw not yet implemented for CanBusMotionControl
[ERROR]getDutyCycleRaw not yet implemented for CanBusMotionControl
[ERROR]getDutyCycleRaw not yet implemented for CanBusMotionControl
[ERROR]getDutyCycleRaw not yet implemented for CanBusMotionControl
[DEBUG]sharedcan [0] thread ran 595 times, req.dT:10[ms], av.dT:10.08[ms] av.loopT :0.03[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]sharedcan [0] printing boards infos:
None

```

amaroyo, 2017-04-28:
@randaz81, can I omit that errors?

randaz81, 2017-05-03:
Indeed.
Fixed by https://github.com/robotology/icub-main/commit/bf3c0ff3edc058b6a3aeae82b6a4836eb3d0a9f6

amaroyo, 2017-05-04:
Updated and fixed. Thank you very much!

closed 2017-05-04 by amaroyo

randaz81, 2017-05-04:
@amaroyo Have you opened a PR on `robots-configuration`? Thank you!

amaroyo, 2017-05-04:
@randaz81, I modified the files in local for now. I wanted to make the robot fully operating before opening a PR. Is that ok?

amaroyo, 2017-05-12:
@randaz81, We are still having problems with the encoders of the right hand. We are waiting for Julien to try to help us with that, if we solve the problem I will update the robots-configuration file with the right values. Thanks for the patience.