robotology/mesh-support #452: Disturbing vibrations in the legs

serena-ivaldi, 2017-04-21:
Please **fill in the following two sections** and then **remove any comment and/or remark become unnecessary**:

### Description of the failure

Dear icub-support

we are having disturbing problems with the legs. When we turn on the iCub (fixed on the pole), and we launch the yarprobotinterface, the legs start vibrating a lot, to the point where the entire robot is vibrating. Basically, we have to put the legs in idle to work on the upper-body.

We logged with the dataDumper the joints of the legs:

[image]
[image]

It is visible that in some joints the encoder values are changing too much (for a robot that is not moving). This also impacts the torque estimation by the way (I don't have the plot, but I can produce it).

Could this be solved simply by lowering the PID gains?

serena-ivaldi, 2017-04-21:
Currently, the PID values for the legs are (in hardware/motorControl):

Right upper leg:
<param name="kp">           -1280      1280         711.11   1066.66 </param>       
<param name="kd">              0         250        0.00       0.00 </param>       
<param name="ki">           -6250      25000        7111.09   1066.64 </param>

Right lower leg:
<param name="kp">        -2105.00     -2310.00         </param>       
<param name="kd">            0.00         0.00         </param>       
<param name="ki">           -0.09        -0.09         </param>

Left lower leg:
<param name="kp">            1280      -1280    -711.11      -1066.66 </param>       
<param name="kd">               0       -250       0.00         0.00 </param>       
 <param name="ki">            6250     -25000   -7111.09     -1066.64</param>

Left lower leg:
<param name="kp">          2200.00     2200.00         </param>       
<param name="kd">            0.00         0.00         </param>       
 <param name="ki">            0.09         0.09         </param>

julijenv, 2017-04-21:
Hi @serena-ivaldi ,
I dont know for the PID, maybe it could work, BUT maybe it is a HArdware problem, could you try to see by disabling one joint at a time which one start creating the shakes on the robot.
let us know

serena-ivaldi, 2017-04-24:
Hi Julien, 
if you look at the plots it is quite explicit: I have to disable almost all joints (all those that you see oscillating).

randaz81, 2017-04-24:
Hi @serena-ivaldi, are you sure you the attached plots refer to the problem you mentioned? I can only see vibrations in the ~0.1 deg range which are comparable with the resolution of encoder (and so very very small...)

serena-ivaldi, 2017-04-24:
Yes Randaz, it is quite weird actually: the encoders' values do not change a lot as you point out, but the legs really vibrate a lot, to the point that we need to idle the legs to work with the upper body. You can see the hip joint moving: my guess is that the encoder says 0.1deg but it is 1deg at least. I do not know how to explain that though.

randaz81, 2017-04-24:
Was the robot ok in the last weeks? when this problem started to happen? Can you check that the power supply voltage has not been accidentally raised?

julijenv, 2017-04-26:
Hi @serena-ivaldi ,
as advised by @randaz81 , to me the only possible way that all the motors shake is that the power supply setup get raised.
Please check.

serena-ivaldi, 2017-05-01:
Hi @julijenv @randaz81 
this problem has been there since the robot arrived. Since we have been mostly working with the upperbody, we have been disabling since the lower body putting the legs on idle.
But of course it is not a viable solution.

Regarding the power supply, I don't think we changed that, but it is possible we made a mistake. Which values should we have (to double check)?

julijenv, 2017-05-02:
Hi @serena-ivaldi ,
A good wiki is a useful tool only if people find what they are looking for inside it.
So here is the link
http://wiki.icub.org/wiki/Power_supply
give us feedback,
thx in advance

serena-ivaldi, 2017-05-05:
Hi Julien,

when we turn on the power supplies, the voltages and currents are:
13.48V - 3.1A (pc104)
39.10V - 1.83A (motors)

Is it ok? Should we lower the voltage of the motors?

gabrielenava, 2017-06-14:
were the legs torque controlled or position controlled? if they were position controlled, can you try to put them in torque control and see if they still vibrate? if not, I strongly suspect it is a PID gains problem.

If they were torque controlled, is wholebodydynamics running on the external pc? We experienced a similar vibration in this case, it was due to communication delays. Then, try to run wholebodydynamics on the pc104.

closed 2018-01-16 by julijenv