robotology/mesh-support #453: Skin, Mais and Xsensmtx problems

amaroyo, 2017-04-28:
Dear @julijenv, 

The skin of the left arm is not working at all. I tried connecting with the canLoader and the boards don't give any answer. 

The repair we did in the right arm (biceps) where you changed the cable, which was giving shortcuts, is still giving  some problems. Not sure if for the same reason. But from time to time, when I start the robot, the mais turns off (I see it in the right hand of the robot, and the hardware fault). However, it doesn't turn off all the iCub as before. 

The connector on the back of the head of iCub - Xsensmtx is giving more problems, almost every time I start it up. Lately I had to disconnect it and connect it several times to solve the problem. Is there a cleaner way to fix this issue?

Thank you a lot, 
Alex.

julijenv, 2017-05-02:
Hi @amaroyo ,
If the left arm skin is not responding at all, it means that either the white or the green wire is broken - NET7, do the MTB board (skin boards) get unlighted when you start the robot?
About the MAIS board - right arm-, I fear that something else got partially powered on the line (poor connection - which prevents the correct functionment of the MAIS) - ask edl the recrimp the entire line NET6.
About Xsensmtx, unfortunately you are dealing with  brand new hardware, and I dont know if the firmware, software could be responsible for this misbehaviour. Please unconnect the xsensmtrx usb and chack on shell - dmesg tail to see if the xsensmtx appear to get unconnected, reconnect and redo - dmesg tail and check that some usb port got opened.
let me know

amaroyo, 2017-05-02:
Thanks @julijenv, I got in contact with EDL to try to solve the problem. 

The problem of Xsensmtx is the connection to the board, I will try to ask EDL also to try to change the connection because it is very lose, even touching the cable a little big, makes it disconnect. 

A question for you, do you think that the problem with the MAIS board can generate problems in the calibration of the hand fingers? Because in the moment I applied the changes in the [robot configuration](https://github.com/robotology/icub-support/issues/451) I experienced an offset of about 30º per each one of the joints of the right hand (the one that we have the MAIS problem). This was not happening before. What do you think?
Thx.

julijenv, 2017-05-02:
Hi @amaroyo ,
I have never experienced such a behaviour until now with the MAIS, so I dont think it is related, especially if you modify some festures on the robot configuration files.
If the MAIS doesnt work, there is no way that the right arm got calibrated. when working properly everything should go smooothly.
let me know, ASA they will have recrimp the line NET6.

reafrancesco, 2017-05-03:
Hi @amaroyo, 
- left arm: are the green/white cables broken?
- MAIS right arm: do you asked EDLers to recrimp the entire live. Should I ask it?
- Xsensmtx: nothing has changed for Xsensmtx in the new hardware of the red iCub anyway we should look into the dmesg output as Julien is asking

Thanks

amaroyo, 2017-05-03:
@reafrancesco, 

EDL has a new procedure to ask for reparation, and the webpage didn't work. In any case, EDL was busy and maybe tomorrow morning will pass by to check the mais and the cables of the skin. 

The Xsensmtx connection to the board is like a tooth about to fall out, it moves very easily and it doesn't make good connection.

amaroyo, 2017-05-04:
@reafrancesco, @julijenv, 

Today with EDL we changed the connection, it's quite strong now and it's not moving. It worked for quite a lot of times, however, I spotted again the same issue. I hit a bit the head and it reconnected. I will keep checking it again to spot the issue. 

Mais right arm is totally dead now. Tomorrow we will check the wiring of the cables, the same as the skin. 

I will keep you posted. 

Thanks, 
Alex.

amaroyo, 2017-05-05:
Skin and Mais working fine now. 

Xsensmtx is not connected and the yarprobotinterface doesn't start up. Possible problem could be the round connector. We don't have a spare one here.

amaroyo, 2017-05-10:
@julijenv, summing up:

One of the cables of the round connector of the Xsensmtx was not soldered properly and was giving problems. Now it seems to be working fine. 

1) Left arm thumb distal broken.
2) We still have the same problems with the offset of the right hand. Some times when you start the iCub the offset if the hand joints is around 30º, other times is around 5º.

julijenv, 2017-05-11:
Hi @amaroyo ,
1. I think I explain you how to repair it, and actually it is quite a simple joint - I think you may need to give it a try, or else I could drop by to show you again but this time I will make you do it.
2. Never seen this kind of behaviour, are u sure  u didnt change any values in the config files?!?! Or may be some compoents on MAIS are dead.

amaroyo, 2017-05-12:
Hi @julijenv, 

1. I will give it a try! :)
2. The right hand joints are again in Hardware fault. I think this behaviour is due to some loose cable, which is transmitting bad the values and is making them enter in hardware fault. 

Please check the photos below to see the desoldered components of the Mais. Could you please bring us a left hand Mais? 

In any case, if you can pass by Monday, we can try to check the cables of the right hand, with the help of EDL. 

[image]
[image]
[image]

amaroyo, 2017-05-17:
Dear @julijenv, it seems that is still working fine. I will close the issue for now, and check it from time to time. 

Thanks a lot, 
Alex.

closed 2017-05-17 by amaroyo