robotology/mesh-support #454: Hardware fault on left arm

serena-ivaldi, 2017-05-09:
Dear all

we just had an improbable hardware fault on the left arm. The first 4 joints went off with " hardware fault", while the arm was torque controlled (with wholeBodyDynamics + torque control launched by demoForceControl).

Here' s the error from yarplogger:

```
3204		1528.342573		WARNING		 from BOARD 10.0.1.1 (EB1), src LOCAL, adr 0, time 2036s 349m 57u: (code 0x0000000d, par16 0x0195 par64 0x0007002200a400a5) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
3180		1488.937817		ERROR		 from BOARD 10.0.1.1 (EB1), src LOCAL, adr 0, time 1996s 945m 559u: (code 0x02000007, par16 0x0003 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
3179		1488.937807		ERROR		 from BOARD 10.0.1.1 (EB1), src LOCAL, adr 0, time 1996s 945m 517u: (code 0x02000007, par16 0x0002 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
3178		1488.937797		ERROR		 from BOARD 10.0.1.1 (EB1), src LOCAL, adr 0, time 1996s 945m 482u: (code 0x02000007, par16 0x0001 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
3177		1488.937766		ERROR		 from BOARD 10.0.1.1 (EB1), src LOCAL, adr 0, time 1996s 945m 419u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
3176		1488.837193		ERROR		Virtual analog sensor timeout!! No new value received for more than  0.0808389  secs. 
3175		1488.837186		ERROR		Virtual analog sensor timeout!! No new value received for more than  0.0807109  secs. 
3174		1488.837178		ERROR		Virtual analog sensor timeout!! No new value received for more than  0.0804529  secs. 
3173		1488.837169		ERROR		Virtual analog sensor timeout!! No new value received for more than  0.080755  secs. 
3172		1488.837140		ERROR		Virtual analog sensor timeout!! No new value received for more than  0.0805779  secs. 
3089		1340.186328		WARNING		embObjMotionControl::updateMeasure() torque measure saturated from -2.0114 to +2.0000 on BOARD EB2 IP 10.0.1.2 joint 0, count: 1
2469		974.842631		ERROR		Virtual analog sensor timeout!! No new value received for more than  0.0813868  secs. 
2468		974.842623		ERROR		Virtual analog sensor timeout!! No new value received for more than  0.0807462  secs. 
2467		974.842613		ERROR		Virtual analog sensor timeout!! No new value received for more than  0.080301  secs. 
2466		974.842602		ERROR		Virtual analog sensor timeout!! No new value received for more than  0.080483  secs. 
2465		974.842557		ERROR		Virtual analog sensor timeout!! No new value received for more than  0.0802181  secs. 
2435		962.500885		WARNING		 from BOARD 10.0.1.4 (EB4), src LOCAL, adr 0, time 1470s 522m 944u: (code 0x0000000b, par16 0x01e5 par64 0x0018002500070125) -> SYS: the RX phase of the control loop has last more than wanted. In par16: RX execution time [usec]. In par64: latest previous execution times of TX, RX, DO, TX [usec] + .

```

### Detailed conditions and logs of the failure

---
**Remark**: if you are not a daily icub-support user, please read carefully the guidelines for contributing linked at the top of this page. In particular, the **name of your robot** must be always indicated: either using **labels** if you're already a contributor, as per standard, or by specifying it while detailing the issue.

julijenv, 2017-05-09:
Hi @serena-ivaldi ,
ASA I can see the problem is that the FT sensor stop stransmiting or did not sufficiently sent messages to the control board.FT sensor communicate with the board IP 10.0.1.1.
If not responding anymore, it means that one cable is broken inside the left upper arm cover from the connector of the FT sensor.
let me know

serena-ivaldi, 2017-05-10:
Hi Julien, can you tell us where to look precisely? With the nice figures as in the other torso issue, where we able to find easily the right board :)
I am away today, but maybe @dogoepp can have a look!

DanielePucci, 2017-05-10:
This issue looks like #449

serena-ivaldi, 2017-05-10:
indeed @DanielePucci 
By the way, when moving the left arm in torque control mode (activated by demoForceControl) the robot arm is very stiff and not fluid when you physically move it. iCubParis01 and 02 were much better when we were interacting physically. How is that possible? I also noticed that in zero-torque control it seems to have gravity compensation by default, but I didn't activate gravityCompensation (the module).
By the way, we are using wholeBodyDynamics, should we do it with wholeBodyDynamicstree? In that case which is the right command to use if the robot is on the pole?

julijenv, 2017-05-24:
Hi @serena-ivaldi ,
do u still have this issue when laucnching the robot or when launching some modules??
let me know

serena-ivaldi, 2017-06-15:
Didn't have recently... however when we do the demo with zero torque control the robot arm is very stiff, too much, and difficult to move. iCubParis and iCubDarmstadt were more fluid and easier to move.

closed 2018-01-16 by julijenv