robotology/mesh-support #457: Left shoulder tendon broken

ChiaraBartolozzi, 2017-05-11:
### Description of the failure
Left shoulder tendon broken

### Detailed conditions and logs of the failure
We don't know how this happened, as we found the part broken during a workshop, when the robot was off. The cable is stressed and damaged in other parts as well.

arrenglover, 2017-05-11:
Looking at the robot it is already fixed. However, @massimoregoli  said that something is still not working with the left arm.

ChiaraBartolozzi, 2017-05-11:
@massimoregoli could you please explain what do you mean by "something is still not working"?

spada1978, 2017-05-11:
sostituiti cavi n.7 e 8 codice IIt 4764.1 e 300 4765.2 perche rotti.
effettuato nuovamente il cablaggio meccanico della spalla

closed 2017-05-11 by spada1978