robotology/mesh-support #458: Very stiff behaviour of right shoulder pitch with motors off

gabrielenava, 2017-05-11:
### Description of the failure

with the motors off, the right shoulder pitch seems much more stiffer than the left. It may be a 
mechanical problem related to the bearing.

DanielePucci, 2017-05-18:
@julijenv @spada1978

julijenv, 2017-05-18:
Hi @gabrielenava  and @DanielePucci ,
You are absolutely right guys about the stiffness in between both arm, BUT unfortunately the right shoulder is as stiff as the right and left of the purple robot so I cant really notice a real problem inside it.
AS explain to @gabrielenava , there is no specific noise and the shoulder dont get bloack in some position.
For reminder, the robot is supposed to be moved (controlled) by powering it NOT moving it by hand, so it is quite delicate issue the one you raised, but I agree with you. BUT I repat as long as you can control it correctly, there is not so much we can do.

closed 2017-08-08 by julijenv