robotology/mesh-support #461: Hand joints and fingertips not working

massimoregoli, 2017-05-17:
I noticed the following problems on the purple iCub hands:

Left hand
- joints 10 and 13 not working
Right hand
- fingertips of the thumb, index and middle fingers not working

julijenv, 2017-05-18:
Updates:
Right Hand
fingertips problems have been successfully diagnosed
Left Hand
joint 15 was not mentionned but was broken totally and has been repaired
joint 10 is now working properly (one wire got snapped)
joint 13 is still under maintenance - broken motor - we need to spin the gearbox shaft to couple and restore it.

julijenv, 2017-05-19:
The motor for the proximal middle finger has been successfully replaced (new SN , old SN 0331), the cable of the proximal joint as well.
IITCODE Qta ALIAS Version Description
3542 x 1 RC_TLR_007_P_029 - A2 - FINGER MOTOR CAPSTAIN
4469 x 1 1224N012S_30B20_1-256	 - 1224M012S-K380-30B20+10/1-256:1+K262, Faulhaber - Brush motor, gear box, encoder with 450 mm cable length
1815 x 1m  carlstahl cable 0.63mm uncoated

closed 2017-05-22 by julijenv