robotology/mesh-support #462: Mais Energy Problem

amaroyo, 2017-05-17:
Ciao @julijenv, 

Following the [previous issue](https://github.com/robotology/icub-support/issues/453), I am creating a new one to show you the evolution of the problem. 

1. This morning (and yesterday with you) we checked the encoder values of the right hand and they were perfect (zero). It was working pretty well. 

2. I turned iCub on again to check some things, and I saw that the values changed (with an offset around 5 degrees), as you can see in this image:
[image]

3. Turn off yarprobotinterface, turn on. The values were a bit higher:
[image]

4. Same thing, off and on. The offset is around 15-20 degrees:
[image]

5. Off and On: This time we have the 30s!
[image]

6. Off and On again, even turning the power off. Calibration done, after a bit Mais OFF.
[image]

7. Turn off the power. Turn on, the Mais was NOT on. 

8. Turn off the power, moved the arm and some of the connectors, on again. the Mais started, the values were around 5 degrees. After a couple of minutes, mais off again. 

9. Moved a bit the arm, turn it on. The degrees were 0! The mais was on.

10. Restarted again, the degrees are still 0. And after a while, it was still working.

So, I need your advice. I really think that there is a broken cable which is not giving enough power to the Mais. You recrimped the NET6 in the hand, so the problem may be in the arm. If not, where does the NET6 go after the shoulder? Can be that the problem is there?

julijenv, 2017-05-18:
Hi @amaroyo ,
when you say turn off and on yarprobotInterface, you mean

- close the yarprobotInterface ONLY BUT NOT shutting down the motor switch or do you mean
- close the power line when yarprobotInterface finally close and reboot ALSO motors OFF and ON

the line 6 is just in between the board 2B4 and the hand, you have however a black filter in between thenm at the very beginning of the 2B4, make them checked by Davide Dellepiane.

good luck,
btw if you test it in the first way that I assume you did, please try the other way and let me know, tht could also be related to firmware issue and in that case I would also suggest you to reupload teh firmware of MAIS inside the board.
Let me know,

amaroyo, 2017-05-18:
Hi @julijenv, 

The first couple of times we did it only via software. When the values were getting strange, we did turn off the powerline and then we turned it on. Each time we tested it. This morning, the values were 30 again, and it entered in hardware fault.

I saw yesterday that there is a filter touching the motor of the shoulder, I also experienced that touching the cables between the shoulder and the back of the iCub, the mais turns on and off. Also, the encoder 1 of the shoulder got disconnected, and reconnected again while touching the cables. I do believe that there is something loose there. 

I will check them with Delle.

Thanks for the help. 

[image]
So, is this photo correct? Then which is the line that connects the board 2B4 to the back of the iCub?

julijenv, 2017-05-18:
Hi @amaroyo ,
you are right, the net 6 goes up to the back of the head of iCub from the filter I was mentionning in my previous mail. BUT the power of the line for the MAIS is provided by 2B4, that why I dont think data cant be  altered from the filter to the back of the head but a little check is always welcome.
Let me know.

amaroyo, 2017-05-22:
collateral damage: right pinky fingers are broken.

julijenv, 2017-05-24:
It has been restored the all line net6, the skin wires as well.
We performed the rewiring of the pinky and ring finger motr of the right  hand.
IITCODE Qta descrition
2391 x 2m : cable carlstahl 0.45-0.61mm coated

closed 2017-05-24 by julijenv