robotology/mesh-support #465: Head malfunction

pedrombmachado, 2017-05-19:
### Description of the failure
Hi Julien,
   I have an issue to report. Theleft motor, joint 0 starts starts to work full-speed a few seconds after the calibration process. Motor belt looks to be partially damaged. Please note that the head does not moves while this is happening.

[image]
[image]
[image]
[image]
[image]
[image]

julijenv, 2017-05-19:
Hi @pedrombmachado ,
please come on skype because the pictures R far from being useful.

pedrombmachado, 2017-05-19:
Hi @julijenv ,
   Can we schedule a skype call for Monday? Let me know the best time for you.

julijenv, 2017-05-22:
Hi @pedrombmachado ,
I'm right behind my desktop, so let me know when U r available, all morning is good for me.

pedrombmachado, 2017-05-22:
Hi @julijenv,
  I am now ready. I have a 30 min window. Are you available or should we postpone to the afternoon?

pedrombmachado, 2017-05-22:
Sorted with @julijenv support. Many thanks.

closed 2017-05-22 by pedrombmachado